rauc/rauc #1137: Add systemd service file and dbus configuration to Debian package

martin-petzold, 2023-05-19:
**Is your feature request related to a problem? Please describe.**

Currently the Debian package only contains the RAUC binary [1]. This may be suitable for build environment, where RAUC bundles are created, but not for Debian embedded environments, where RAUC is used as daemon for upgrades.

[1] https://packages.debian.org/bookworm/arm64/rauc/filelist

**Describe the solution you'd like**

I would suggest to add the 'rauc.service' and the 'de.pengutronix.rauc.conf' to the Debian package. The RAUC service should also have a 'WantedBy=default.target' declaration in the '[Install]' section. Maybe it would make sense to place the service file in some 'share' location because it is not used for build environment. Then the documentation should explain to move this file to '/usr/lib/systemd/system'.

ejoerns, 2023-05-20:
The `rauc` package is for host tool usage, which is the most common scenario for the Debian package. Service-related configuration should not be included in this package.

In cases where Debian will be used for a target system, there is a distinct `rauc-service` package that includes the files you suggested:

https://packages.debian.org/bookworm/arm64/rauc-service/filelist

closed 2023-05-20 by ejoerns

martin-petzold, 2023-05-21:
Okay, I was not aware of the rauc-service package. However, BlueZ, IWD, a lot of tools for Linux use *d and *ctl separate binaries. IMHO this is cleaner than having one binary for two different purposes. But with the separate Debian package at least it is separated there.