rauc/rauc #1138: RAUC should be separated into a daemon executable 'raucd' (the service) and a control executable 'raucctl'

martin-petzold, 2023-05-19:
**Is your feature request related to a problem? Please describe.**

I have seen several issues related to miss-understanding about RAUC to be used on build and host machine. When using Debian as build and host platform it can also be confusing. Because on the embedded platform there is a daemon AND a control use case.

**Describe the solution you'd like**

This tool should be separated into a daemon executable 'raucd' (the service) and a control executable 'raucctl'.

martin-petzold, 2023-05-21:
Okay, there is already a separate RAUC service package for Debian as mentioned in https://github.com/rauc/rauc/issues/1137: https://packages.debian.org/bookworm/rauc-service

However, I will open a new ticket to add this to the documentation.

martin-petzold, 2023-05-21:
Also it seems that RAUC is not meant to be a daemon at the moment, rather only a tool with dbus frontend.

closed 2023-05-21 by martin-petzold

jluebbe, 2023-05-22:
This is incorrect. The `rauc` binary provides both the client and D-Bus service, depending on build-time configuration. Accordingly, the `rauc-service` Debian package provides only the necessary configuration files.

Combining both aspects into a single binary is done done to save some space, which is relevant for resource constrained systems.