rauc/rauc #1139: Add some information about the 'rauc' and 'rauc-service' Debian packages to the documentation and maybe GitHub README page

martin-petzold, 2023-05-21:
**Is your feature request related to a problem? Please describe.**

I was not aware of the `rauc-service` Debian package. It would have helped a lot to know about this one. Also at the moment there is no information about RAUC on Debian platforms in the documentation

**Describe the solution you'd like**

Maybe add a section under `6. Integration`: https://rauc.readthedocs.io/en/latest/integration.html#

Maybe also add some information to the GitHub `README`: [README.rst](https://github.com/rauc/rauc/blob/master/README.rst?plain=1)

jluebbe, 2023-05-22:
Note that the Debian packaging is not part of the RAUC project, but maintained by the Debian project. Perhaps the https://salsa.debian.org/debian/rauc/-/blob/debian/master/debian/README.Debian is a better place? @ukleinek

ukleinek, 2023-05-22:
A note in the Debian package of rauc doesn't help people reading the upstream docs to find the Debian package however. So I doubt adding anything there helps to resolve the original issue.

jluebbe, 2023-05-22:
> A note in the Debian package of rauc doesn't help people reading the upstream docs to find the Debian package however. So I doubt adding anything there helps to resolve the original issue.

Perhaps a pointer in the integration docs to the Debian package's README.Debian, where the packaging can be explained?

PR #1141 (2023-05-22): point to distribution packages in the integration chapter and in the README
Fixes: #1139