rauc/rauc #1140: Invalid confirmation of slot marked as good

martin-petzold, 2023-05-21:
**Describe the bug**

I set the status of a slot as `good` and this is confirmed with `rauc status: marked slot rootfs.0 as good`, but if I check the status this is not the case. The status is still `bad`.

**Background information**

Debian 11
Kernel 5.10
RAUC 1.9

**To Reproduce**

Steps to reproduce the behavior:

1. Boot system
2. Mark slot as good
3. Check status

**Expected behavior**

Either the command `sudo rauc status mark-good` should fail and return an error or the post-command status should actually be `good`.

**Logs**

Shell:

```
$ sudo rauc status mark-good

(rauc:667): rauc-WARNING **: 19:20:28.946: Failed getting primary slot: uboot backend: fw_printenv failed with exit code: 1

rauc status: marked slot rootfs.0 as good
$ sudo rauc status

(rauc:682): rauc-WARNING **: 19:20:32.276: Failed getting primary slot: uboot backend: fw_printenv failed with exit code: 1

=== System Info ===
Compatible:  xyz
Variant:     
Booted from: rootfs.0 (/dev/mmcblk2p1)

=== Bootloader ===
Activated: (null) ((null))

=== Slot States ===
○ [rootfs.1] (/dev/mmcblk2p2, ext4, inactive)
	bootname: B
	boot status: bad

○ [rootfs.0] (/dev/mmcblk2p1, ext4, booted)
	bootname: A
	mounted: /
	boot status: bad
```

Journal:

```
Mai 21 19:20:28 tavla sudo[666]:    tavla : TTY=pts/0 ; PWD=/home/tavla ; USER=root ; COMMAND=/usr/bin/rauc status mark-good
Mai 21 19:20:28 tavla sudo[666]: pam_unix(sudo:session): session opened for user root(uid=0) by tavla(uid=5280)
Mai 21 19:20:28 tavla rauc[670]: Warning: Bad CRC, using default environment
Mai 21 19:20:28 tavla rauc[670]: ## Error: "BOOT_ORDER" not defined
Mai 21 19:20:28 tavla rauc[381]: Failed getting primary slot: uboot backend: fw_printenv failed with exit code: 1
Mai 21 19:20:28 tavla rauc[672]: Warning: Bad CRC, using default environment
Mai 21 19:20:28 tavla rauc[672]: ## Error: "BOOT_ORDER" not defined
Mai 21 19:20:28 tavla rauc[381]: Failed to get boot state of rootfs.1: uboot backend: fw_printenv failed with exit code: 1
Mai 21 19:20:28 tavla rauc[673]: Warning: Bad CRC, using default environment
Mai 21 19:20:28 tavla rauc[673]: ## Error: "BOOT_ORDER" not defined
Mai 21 19:20:28 tavla rauc[381]: Failed to get boot state of rootfs.0: uboot backend: fw_printenv failed with exit code: 1
Mai 21 19:20:28 tavla rauc[381]: Failed to determine boot states: Could not determine all boot states
Mai 21 19:20:28 tavla rauc[674]: Warning: Bad CRC, using default environment
Mai 21 19:20:28 tavla rauc[381]: rauc mark: marked slot rootfs.0 as good
Mai 21 19:20:28 tavla sudo[666]: pam_unix(sudo:session): session closed for user root
Mai 21 19:20:32 tavla sudo[681]:    tavla : TTY=pts/0 ; PWD=/home/tavla ; USER=root ; COMMAND=/usr/bin/rauc status
Mai 21 19:20:32 tavla sudo[681]: pam_unix(sudo:session): session opened for user root(uid=0) by tavla(uid=5280)
Mai 21 19:20:32 tavla rauc[685]: ## Error: "BOOT_ORDER" not defined
Mai 21 19:20:32 tavla rauc[381]: Failed getting primary slot: uboot backend: fw_printenv failed with exit code: 1
Mai 21 19:20:32 tavla rauc[686]: ## Error: "BOOT_ORDER" not defined
Mai 21 19:20:32 tavla rauc[381]: Failed to get boot state of rootfs.1: uboot backend: fw_printenv failed with exit code: 1
Mai 21 19:20:32 tavla rauc[687]: ## Error: "BOOT_ORDER" not defined
Mai 21 19:20:32 tavla rauc[381]: Failed to get boot state of rootfs.0: uboot backend: fw_printenv failed with exit code: 1
Mai 21 19:20:32 tavla rauc[381]: Failed to determine boot states: Could not determine all boot states
Mai 21 19:20:32 tavla sudo[681]: pam_unix(sudo:session): session closed for user root
```

**Additional context**

The u-boot script is not included in this test case. I am testing basic commands in case the u-boot environment is not pre-configured.

ejoerns, 2023-08-29:
@martin-petzold In recent releases `/var/lib/rauc` will be written on the first modification of the system status.
So having this empty on startup is valid.

Is the above-mentioned error still present? Or did you find a solution?

ejoerns, 2023-10-24:
I will close this now, feel free to reopen if the issue still exists.

closed 2023-10-24 by ejoerns