rauc/rauc #1149: Missing documentation: Boot after install fails because of (wrong) partition UUIDs in /etc/fstab

martin-petzold, 2023-06-04:
**Describe the bug**

After installation of a RAUC bundle the boot fails to mount some partitions (all other except of the current rootfs). The reason is the UUID-based entries in the fstab. As far as I understand either "LABEL=" or device names should be used.

**Background information**

RAUC 1.9
debos (debootstrap)

**To Reproduce**
Steps to reproduce the behavior:
1. Build debos image (or any other using UUIDs as default in `/etc/fstab`)
2. Build RAUC bundle
3. Install RAUC bundle

Search for `fstab`: https://rauc.readthedocs.io/en/latest/search.html?q=fstab&check_keywords=yes&area=default

**Expected behavior**

Some documentation about how to setup `/etc/fstab` for usage with RAUC.

martin-petzold, 2023-06-04:
Opened also related issue in debos: https://github.com/go-debos/debos/issues/409

ejoerns, 2023-06-04:
It is not the direct responsibility of RAUC to tell users how to create an fstab for their system. And using UUID-based entries I guess is a little uncommon in areas where RAUC is normally used.

Thus yes, it will be good to have documentation, but I don't see it as a bug.
A good matching section should be https://rauc.readthedocs.io/en/latest/integration.html#partitioning-your-device

About the UUID, yes since RAUC is a tool for image-based updates it can be expected that the file system changes with each update, and thus does the UUID. So it is improper for being used in the `/etc/fstab`.
As you mentioned, the `LABEL=` or device names would be an alternative. Also the `PARTUUID=` will remain consistent across updates.

jluebbe, 2023-06-05:
For embedded devices, which run on fixed hardware, the links in `/dev/disk/by-path` (which are generated by udev) are often a good choice. They are independent of any data on disk and cannot be confused by removable storage devices (which could be at least a denial-of-service attach vector).

martin-petzold, 2023-06-05:
Okay sorry, it is rather a feature request then.

obbardc, 2023-06-08:
Yeah this really is the responsibility of the rootfs - perhaps something could be mentioned in the rauc documentation about this to make users really sure or we just close this ticket?

jluebbe, 2023-10-24:
@martin-petzold @obbardc I've created a PR to add a note about this in #1271.

PR #1271 (2023-10-25): docs/integration: recommend using stable paths in fstab and system.conf
Fixes: #1149
commit 47bed290 (2023-10-25): docs/integration: recommend using stable paths in fstab and system.conf

Fixes: #1149
closed 2023-10-25 by ejoerns