rauc/rauc #1165: meson build system installs /usr/libexec/rauc-service.sh without execute permissions

BubuOT, 2023-06-16:
**Describe the bug**
When building rauc with dbus support with meson, running it on a non-systemd system every call to rauc fails with the following error:

```
Error retrieving slot status via D-Bus: error creating proxy: Error calling StartServiceByName for de.pengutronix.rauc: Cannot launch daemon, file not found or permissions invalid
```

It seems that `/usr/libexec/rauc-service.sh` has no execute permissions. Comparing against a system with rauc 1.8 built with autotools, this use-case works correctly and this file has execute permissions.

When starting the service manually (with `rauc service &`) everything works afterwards.

**Background information**
rauc version v1.9

**To Reproduce**

S. above, then run i.e.` `rauc status` on the system.

**Expected behavior**
`rauc status` works without having to manually start the service.

**Logs**
-

**Additional context**
-

ejoerns, 2023-06-16:
Confirmed. Using `install_data()` does not seem to be the right option for executables (as its name already suggests).

ejoerns, 2023-06-16:
> Confirmed. Using `install_data()` does not seem to be the right option for executables (as its name already suggests).

Ok, seems `install_data()` can be used, just lacks a proper `install_mode`

PR #1166 (2023-06-16): data: meson.build: install rauc-service.sh with executable bit set
The script needs to be executable. This got lost during meson transition.

Fixes #1165
commit 5697f1b2 (2023-06-16): data: meson.build: install rauc-service.sh with executable bit set

The script needs to be executable. This got lost during meson
transition.

Fixes #1165
closed 2023-06-16 by ejoerns