rauc/rauc #1187: Rauc will only accidentally build against OpenSSL

fhars, 2023-07-19:
The `Makefile.am` references the variables `OPENSSL_LDFLAGS`and `OPENSSL_INCLUDES` in the definitions of  `AM_LDFLAGS` and `AM_CPPFLAGS`, and nothing ssl-related in `AM_CFLAGS`, while `configure` adds definitions for `OPENSSL_LIBS` and `OPENSSL_CFLAGS`. As a result, rauc will only build if OpenSSL happens to be installed in the same directory as some other library it depends on.
A similar thing is probably also true for the other libraries where the definition of `AM_LDFLAGS` uses different names that the one for `librauc_la_LIBADD`.

```patch
From d22046902a9e14ee0cbcaf9bab95c9441912f22f Wed Jul 19 00:00:00 2023
From: Florian Hars <florian@hars.de>
Date: Wed, 19 Jul 2023 15:01:39 +0200
Subject: [PATCH] actually link against the found libssl

---
 Makefile.am | 6 +++---
 1 file changed, 3 insertions(+), 3 deletions(-)

diff --git a/Makefile.am b/Makefile.am
index 45d5994..24d880a 100644
--- a/Makefile.am
+++ b/Makefile.am
@@ -15,9 +15,9 @@ CODE_COVERAGE_IGNORE_PATTERN = "*-generated.c"

 GC_CFLAGS = -fdata-sections -ffunction-sections
 GC_LDFLAGS = -Wl,--gc-sections -Wl,-Map,$@.map
-AM_CFLAGS = -DG_LOG_DOMAIN=\"rauc\" $(GC_CFLAGS) $(WARN_CFLAGS) $(GLIB_CFLAGS) $(CURL_CFLAGS) $(NL3_CFLAGS)
-AM_LDFLAGS = $(GC_LDFLAGS) $(WARN_LDFLAGS) $(GLIB_LDFLAGS) $(CURL_LDFLAGS) $(OPENSSL_LDFLAGS)
-AM_CPPFLAGS = -I${top_srcdir}/include -include ${top_builddir}/config.h $(OPENSSL_INCLUDES)
+AM_CFLAGS = -DG_LOG_DOMAIN=\"rauc\" $(GC_CFLAGS) $(WARN_CFLAGS) $(GLIB_CFLAGS) $(CURL_CFLAGS) $(NL3_CFLAGS) $(OPENSSL_CFLAGS)
+AM_LDFLAGS = $(GC_LDFLAGS) $(WARN_LDFLAGS) $(GLIB_LDFLAGS) $(CURL_LDFLAGS) $(OPENSSL_LIBS)
+AM_CPPFLAGS = -I${top_srcdir}/include -include ${top_builddir}/config.h $(OPENSSL_CFLAGS)

 gdbus_installer_generated = \
        rauc-installer-generated.c \
-- 
2.25.1
```

fhars, 2023-07-19:
It looks like the `meson setup` step finds the libraries, but if they er used correctly or not I cannot tell, as the meson build instructions from the readme fail on Ubuntu 20.04 LTS:
```
$ meson compile -C build
usage: meson [-h] {setup,configure,dist,install,introspect,init,test,wrap,subprojects,help,rewrite} ...
meson: error: unrecognized arguments: -C build
```

jluebbe, 2023-07-20:
> It looks like the `meson setup` step finds the libraries, but if they er used correctly or not I cannot tell, as the meson build instructions from the readme fail on Ubuntu 20.04 LTS:
>
> ```
> $ meson [...]

The `meson compile` command was only introduced in 0.54.0, while Ubuntu 20.04 LTS has 0.53.2. `ninja -C build` is the alternative.

jluebbe, 2023-07-20:
We'll update the README via https://github.com/rauc/rauc/pull/1188.

fhars, 2023-07-20:
I saw the issue on Ubuntu 20.04 while experimenting with non-system OpenSSL versions installed in non-default locations.

Anyway, the ninja-build can find the non-standard openssl location.

jluebbe, 2023-07-20:
@fhars I've created #1189 to fix this issue. Could you verify that it also fixes your issue?

closed 2023-07-31 by jluebbe