rauc/rauc #1190: Heads up regarding change in et2fstools ext4 generation

hcl-dr, 2023-07-20:
Mickledore introduces e2fstools 1.47.x which by default adds the _orphan_file_ feature on ext4 file systems.
When performing an update from an earlier version (Kirkstone and downwards < 1.47.0 for e2fstools), the update will fail since the running version of e2fstools does not accept this feature.

While this is not a Rauc problem as such, a recommendation may be in place to avoid this scenario.
For Yocto, this is avoided by adding
`EXTRA_IMAGECMD:ext4:append = " -O ^orphan_file"`
to local.conf (or to the image recipe)

jluebbe, 2023-07-21:
At which point will this fail? When the old system on the target tries to mount the newly written image?

What do you mean by 'the running version of e2fstools'?

hcl-dr, 2023-07-21:
Fails on _resize2fs_ (e2fsprog) after the raw image copy into the partition.
When running on Yocto Kirkstone, e2fsprogs is at 1.46.5 while on Mickledore it is raised to 1.47.

Let say you are currently running Kirkstone and doing a rauc install to Mickledore:
The image to install will by default have the _orphan_file_ feature set and the 1.46.x version of _resize2fs_ (the running one) will complain on an unknown feature (FATURE_C12) and bail out.

jluebbe, 2023-07-21:
Ah, right. Now I'm wondering where we should document that.

Perhaps a FAQ entry for now? Could you post the exact error message as seen in the journal? Including that would make it easier to google.

hcl-dr, 2023-07-21:
```
root@xxxxx:~# rauc install /service/rauc/dr-qt5-bundle-dr-imx8mp.raucb 
installing
  0% Installing
  0% Determining slot states
 10% Determining slot states done.
 10% Checking bundle
 10% Verifying signature
 20% Verifying signature done.
 20% Checking bundle done.
 20% Checking manifest contents
 30% Checking manifest contents done.
 30% Determining target install group
 40% Determining target install group done.
 40% Updating slots
 40% Checking slot rootfs.1
 46% Checking slot rootfs.1 done.
....
 99% Copying image to rootfs.1
 99% Copying image to rootfs.1 failed.
 99% Updating slots failed.
100% Installing failed.
LastError: Installation error: Failed updating slot rootfs.1: Failed to run resize2fs: Child process exited with code 1
Installing `/service/rauc/dr-qt5-bundle-dr-imx8mp.raucb` failed
```

journal snippet:

```
Jul 21 08:01:04 linkbox2p-CD410632 rauc[409]: Configured loop device '/dev/loop0' for 267120640 bytes
Jul 21 08:01:04 linkbox2p-CD410632 rauc[409]: Checking image type for slot type: ext4
Jul 21 08:01:04 linkbox2p-CD410632 rauc[409]: Image detected as type: *.ext4
Jul 21 08:01:04 linkbox2p-CD410632 rauc[409]: Marking target slot rootfs.1 as non-bootable...
Jul 21 08:01:04 linkbox2p-CD410632 rauc[409]: installing /service/rauc/dr-qt5-bundle-dr-imx8mp.raucb: Updating slots...
Jul 21 08:01:04 linkbox2p-CD410632 rauc[409]: installing /service/rauc/dr-qt5-bundle-dr-imx8mp.raucb: Checking slot rootfs.1
Jul 21 08:01:04 linkbox2p-CD410632 rauc[409]: Updating /dev/mmcblk0p2 with /run/rauc/bundle/dr-qtimage-dr-imx8mp.ext4
Jul 21 08:01:04 linkbox2p-CD410632 rauc[409]: installing /service/rauc/dr-qt5-bundle-dr-imx8mp.raucb: Updating slot rootfs.1
Jul 21 08:01:04 linkbox2p-CD410632 rauc[409]: opening slot device /dev/mmcblk0p2
Jul 21 08:01:04 linkbox2p-CD410632 rauc[409]: writing data to device /dev/mmcblk0p2
Jul 21 08:01:26 linkbox2p-CD410632 rauc[409]: Resizing /dev/mmcblk0p2
Jul 21 08:01:26 linkbox2p-CD410632 systemd-journald[180]: Forwarding to syslog missed 25 messages.
Jul 21 08:01:27 linkbox2p-CD410632 rauc[551]: resize2fs 1.46.5 (30-Dec-2021)
Jul 21 08:01:27 linkbox2p-CD410632 rauc[551]: resize2fs: Filesystem has unsupported feature(s) (/dev/mmcblk0p2)
Jul 21 08:01:27 linkbox2p-CD410632 rauc[409]: Installation error: Failed updating slot rootfs.1: Failed to run resize2fs: Child proc>
Jul 21 08:01:27 linkbox2p-CD410632 rauc[409]: installing /service/rauc/dr-qt5-bundle-dr-imx8mp.raucb: Installation error: Failed upd>
Jul 21 08:01:27 linkbox2p-CD410632 rauc[409]: installing /service/rauc/dr-qt5-bundle-dr-imx8mp.raucb: finished
Jul 21 08:01:27 linkbox2p-CD410632 rauc[409]: installing `/service/rauc/dr-qt5-bundle-dr-imx8mp.raucb` failed: 1
```

closed 2023-07-25 by jluebbe

jluebbe, 2023-09-12:
There is an related issue filed for e2fsprogs: https://github.com/tytso/e2fsprogs/issues/147

Bastian-Krause, 2023-09-12:
More context: This causes issues when downgrading from a rootfs containing e2fsprogs 1.47 to a previous version when using tarballs in the bundle and mkfs.ext4. This scenario results in fsck being unhappy:

```
/dev/mmcblk1p1 has unsupported feature(s): FEATURE_C12
```

A possible solution is excluding `orphan_file` in `/etc/mk2fs.conf` in the target rootfs containing e2fsprogs 1.47.

PR #1192 (2023-07-25): docs/faq: document possible filesystem incompatibility with ext4
Fixes: #1190