rauc/rauc #1200: Casync update on NAND mtd partition

claudels, 2023-08-01:
Hello,

**Describe the bug**
I use RAUC as update system on custom board. All works fine but I am trying to set up casync updates to save Mo on remote updates.  I can't get a casync update to work on a nand partition. Here's the command I use and its output : 

```
# G_MESSAGES_DEBUG="all" rauc install /run/media/sda1/casync-guinault-dev-bundle-gmc160-hdmi-20230731144653.raucb -d
rauc-Message: 09:34:30.734: Debug log domains: 'all'
(rauc:4015): rauc-DEBUG: 09:34:30.741: install started
(rauc:4015): rauc-DEBUG: 09:34:30.741: input bundle: /run/media/sda1/casync-guinault-dev-bundle-gmc160-hdmi-20230731144653.raucb
(rauc:4015): rauc-DEBUG: 09:34:30.774: Trying to contact rauc service
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
 40% Checking manifest contents
 60% Checking manifest contents done.
 60% Determining target install group
 80% Determining target install group done.
 80% Updating slots
 80% Checking slot dtb.1
 83% Checking slot dtb.1 done.
 83% Copying image to dtb.1
 86% Copying image to dtb.1 failed.
 99% Updating slots failed.
100% Installing failed.
LastError: Installation error: Failed updating slot dtb.1: failed to run casync extract: Child process exited with code 1
Installing `/run/media/sda1/casync-guinault-dev-bundle-gmc160-hdmi-20230731144653.raucb` failed
```

I also noticed these messages in the logs : 
```
[ 1908.584153] loop0: detected capacity change from 0 to 285
[ 1908.627459] loop_set_status: loop0 () has still dirty pages (nrpages=4)
[ 1908.667460] loop0: detected capacity change from 285 to 280
[ 1908.712448] device-mapper: verity: sha256 using implementation "sha256-caam"
[ 1909.212847] nand: nand_do_write_ops: attempt to write non page aligned data
```

I can't get more log about the subprocess error. We use sysvinit instead of systemd in our distribution because we have boot time constraints.

**Background information**
I am using RAUC V1.9 built with a custom Yocto distribution (langdale). I have well applied required patches to use casync on char devices. The casync rauc bundle and its chunk store are located on a usb stick but I have the same problem if I place file on network location.

Here is the rauc configuration : 
```
[system]
compatible=guinault-gmc
bootloader=uboot
mountprefix=/mnt/rauc
bundle-formats=-plain
boot-attempts=6
boot-attempts-primary=2

[keyring]
path=guinault-rauc.cert.pem

[slot.kernel.0]
parent=rootfs.0
device=/dev/mtd4
type=nand

[slot.dtb.0]
parent=rootfs.0
device=/dev/mtd3
type=nand

[slot.rootfs.0]
device=/dev/ubi0_0
type=ubifs
bootname=A

[slot.kernel.1]
parent=rootfs.1
device=/dev/mtd6
type=nand

[slot.dtb.1]
parent=rootfs.1
device=/dev/mtd5
type=nand

[slot.rootfs.1]
device=/dev/ubi0_1
type=ubifs
bootname=B

```

Do you have any idea where the problem is ?

Thank you

ejoerns, 2023-08-01:
@claudels Thanks for the detailed report!

Yes, the dtb partitions are raw NAND instead of UBI which casync in RAUC does not support. I would assume that you have converted all images to casync images in your bundle and thus it fails for dtb, right? Please verify with 'rauc info'.

What is missing for this or similar cases is an option for `rauc convert` (or for the manifest itself) to skip specific images during conversion. As far as I remember we came across this a couple of times but I cannot remember that this was ever implemented (neither find an issue for this).

Maybe someone using casync more often has an existing workaround for this?

jluebbe, 2023-08-01:
You could store kernel & DTB in UBI as well. That way, it would also benefit from UBI's wear levelling.

claudels, 2023-08-01:
Thank you for your prompt replies

> @claudels Thanks for the detailed report!
>
> Yes, the dtb partitions are raw NAND instead of UBI which casync in RAUC does not support. I would assume that you have converted all images to casync [...]

You're right, I did convert all the images to casync images : 
```
3 Images:
  [dtb]
        Filename:  imx6dl-gmc160.dtb.img.caibx
        Checksum:  ff7a089a603fe2d22bc07e76fe0d47a4af10a23d19b0caeb21b85b5d0f824262
        Size:      45011
        Hooks:
  [kernel]
        Filename:  zImage.img.caibx
        Checksum:  c73243f5655711b00ae183792fffeb4280c8f32a5951d7a00d21e996810aedf3
        Size:      5699872
        Hooks:
  [rootfs]
        Filename:  guinault-dev-image-gmc160-hdmi.tar.caidx
        Checksum:  a00490caa8655f879bcedddbcb3f87e6b2b826972606c3d901348804418855a0
        Size:      356526080
        Hooks:
```
Just converting the ubi image to a casync image would be an excellent solution for us. This would save us a good part of the bundle size. I am interested if anyone knows how to do that with actual version of RAUC.

> You could store kernel & DTB in UBI as well. That way, it would also benefit from UBI's wear levelling.

Yes, it can work, but the drawback is that loading the kernel and dtb from the ubi partition is slow and takes more space than loading from raw memory.

jluebbe, 2023-08-01:
> > You could store kernel & DTB in UBI as well. That way, it would also benefit from UBI's wear levelling.
> 
> Yes, it can work, but the drawback is that loading the kernel and dtb from the ubi partition is slow and takes more space than loading from raw memory.

It should actually need less space, as with separate MTD partitions, you need to reserve space for bad erase blocks in each partition separately (as there is no guarantee, that bad block are spread out).

claudels, 2023-08-01:
Yes, I could try to make one ubi device on whole nand and create multiple ubi volumes. Is it the only solution to use casync for you ?

claudels, 2023-08-01:
I changed my rauc config as well as my mtd partitionning. Now I have one ubi device on the whole nand and multiples ubi volumes for RAUC slots. Here is the config file : 
```
[system]
compatible=guinault-gmc
bootloader=uboot
mountprefix=/mnt/rauc
bundle-formats=-plain
boot-attempts=6
boot-attempts-primary=2

[keyring]
path=guinault-rauc.cert.pem

[slot.dtb.0]
parent=rootfs.0
device=/dev/ubi0_0
type=ubivol

[slot.kernel.0]
parent=rootfs.0
device=/dev/ubi0_2
type=ubivol

[slot.rootfs.0]
device=/dev/ubi0_4
type=ubifs
bootname=A

[slot.dtb.1]
parent=rootfs.1
device=/dev/ubi0_1
type=ubivol

[slot.kernel.1]
parent=rootfs.1
device=/dev/ubi0_3
type=ubivol

[slot.rootfs.1]
device=/dev/ubi0_5
type=ubifs
bootname=B

```

I managed to update my system with a classic RAUC bundle but I have an error when i try to update my system with the casync bundle : 
```

```

Did i make a mistake somewhere ?

ejoerns, 2023-08-01:
@claudels at least not an obvious one that I see.

The RAUC service (and the kernel) log for this would be interesting. Could you start the service in a way that it logs to a file or to a shell?

claudels, 2023-08-02:
A more detailled log 
```
60% Determining target install group
(rauc:3030): rauc-DEBUG: 09:48:19.422: Checking slot: dtb.1
(rauc:3030): rauc-DEBUG: 09:48:19.423:  Adding mapping: dtb -> dtb.1
(rauc:3030): rauc-DEBUG: 09:48:19.423: Checking slot: kernel.0
(rauc:3030): rauc-DEBUG: 09:48:19.423:  No mapping found
(rauc:3030): rauc-DEBUG: 09:48:19.423: Checking slot: rootfs.1
(rauc:3030): rauc-DEBUG: 09:48:19.423:  Adding mapping: rootfs -> rootfs.1
(rauc:3030): rauc-DEBUG: 09:48:19.423: Checking slot: dtb.0
(rauc:3030): rauc-DEBUG: 09:48:19.424:  No mapping found
(rauc:3030): rauc-DEBUG: 09:48:19.424: Checking slot: rootfs.0
(rauc:3030): rauc-DEBUG: 09:48:19.424:  No mapping found
(rauc:3030): rauc-DEBUG: 09:48:19.424: Checking slot: kernel.1
(rauc:3030): rauc-DEBUG: 09:48:19.424:  Adding mapping: kernel -> kernel.1
 80% Determining target install group done.
(rauc:3030): rauc-DEBUG: 09:48:19.424: Using default installation handler
(rauc:3030): rauc-DEBUG: 09:48:19.424: Found image mapping: imx6dl-gmc160.dtb.img.caibx -> dtb
(rauc:3030): rauc-DEBUG: 09:48:19.425: Found image mapping: zImage.img.caibx -> kernel
(rauc:3030): rauc-DEBUG: 09:48:19.425: Found image mapping: guinault-dev-image-gmc160-hdmi.tar.caidx -> rootfs
rauc-Message: 09:48:19.431: Marking target slot rootfs.1 as non-bootable...
(rauc:3030): rauc-subprocess-DEBUG: 09:48:19.431: launching subprocess: fw_printenv BOOT_ORDER
(rauc:3030): rauc-subprocess-DEBUG: 09:48:19.607: launching subprocess: fw_setenv BOOT_ORDER A
(rauc:3030): rauc-subprocess-DEBUG: 09:48:19.737: launching subprocess: fw_setenv BOOT_B_LEFT 0
 80% Updating slots
Updating slots...
rauc-Message: 09:48:19.882: Checking image type for slot type: ubivol
rauc-Message: 09:48:19.882: Image detected as type: *.img.caibx
 80% Checking slot dtb.1
Checking slot dtb.1
 83% Checking slot dtb.1 done.
Updating slot dtb.1
rauc-Message: 09:48:19.882: Updating /dev/ubi0_1 with /mnt/rauc/bundle/imx6dl-gmc160.dtb.img.caibx
 83% Copying image to dtb.1
rauc-Message: 09:48:19.882: opening slot device /dev/ubi0_1
rauc-Message: 09:48:19.895: Extracting /mnt/rauc/bundle/imx6dl-gmc160.dtb.img.caibx to /dev/ubi0_1
rauc-Message: 09:48:19.895: Cannot use /dev/ubi0_0 as seed device (non-existing or char device)
(rauc:3030): rauc-DEBUG: 09:48:19.895: Using casync store path: '/run/media/sda1/casync-guinault-dev-bundle-gmc160-hdmi-20230801142020.castr'
(rauc:3030): rauc-subprocess-DEBUG: 09:48:19.895: launching subprocess: casync extract --store /run/media/sda1/casync-guinault-dev-bundle-gmc160-hdmi-20230801142020.castr --seed-output=no /mnt/rauc/bundle/imx6dl-gmc160.dtb.img.caibx -
 86% Copying image to dtb.1 failed.
 99% Updating slots failed.
100% Installing failed.

(rauc:3030): rauc-WARNING **: 09:48:19.932: Installation error: Failed updating slot dtb.1: Error closing file descriptor: Bad file descriptor
Installation error: Failed updating slot dtb.1: Error closing file descriptor: Bad file descriptor
(rauc:3030): rauc-DEBUG: 09:48:19.933: thread finished for /run/media/sda1/casync-guinault-dev-bundle-gmc160-hdmi-20230801142020.raucb
finished
Installing `/run/media/sda1/casync-guinault-dev-bundle-gmc160-hdmi-20230801142020.raucb` failed
```

When I manually run the casync extract, it seems to work.

claudels, 2023-08-03:
I dug a little deeper the problem by running rauc step-by-step with GDB. I identified that the line `res = g_output_stream_close(G_OUTPUT_STREAM(outstream), NULL, &ierror);` in the function `img_to_ubivol_handler` returns an error.

Here is the GDB output : 
```
 80% Checking slot dtb.1
 83% Checking slot dtb.1 done.
rauc-Message: 10:46:49.119: Updating /dev/ubi0_1 with /mnt/rauc/bundle/imx6dl-gmc160.dtb.img.caibx
 83% Copying image to dtb.1
Updating slots...
Checking slot dtb.1
Updating slot dtb.1
[Switching to Thread 0x763af400 (LWP 4586)]

Thread 2 "installer" hit Breakpoint 1, img_to_ubivol_handler (image=0x75a1a150, dest_slot=0x468f10, hook_name=0x0, error=0x763aed70) at /usr/src/debug/rauc/1.9-r0/src/update_handler.c:1385
1385    /usr/src/debug/rauc/1.9-r0/src/update_handler.c: No such file or directory.
(gdb) next
1387    in /usr/src/debug/rauc/1.9-r0/src/update_handler.c
(gdb) next
1392    in /usr/src/debug/rauc/1.9-r0/src/update_handler.c
(gdb) next
1401    in /usr/src/debug/rauc/1.9-r0/src/update_handler.c
(gdb) next
rauc-Message: 10:47:00.740: opening slot device /dev/ubi0_1
1402    in /usr/src/debug/rauc/1.9-r0/src/update_handler.c
(gdb) next
1403    in /usr/src/debug/rauc/1.9-r0/src/update_handler.c
(gdb) next
1410    in /usr/src/debug/rauc/1.9-r0/src/update_handler.c
(gdb) next
1417    in /usr/src/debug/rauc/1.9-r0/src/update_handler.c
(gdb) next
1418    in /usr/src/debug/rauc/1.9-r0/src/update_handler.c
(gdb) next
rauc-Message: 10:47:03.672: Extracting /mnt/rauc/bundle/imx6dl-gmc160.dtb.img.caibx to /dev/ubi0_1
1420    in /usr/src/debug/rauc/1.9-r0/src/update_handler.c
(gdb) next
rauc-Message: 10:47:04.430: Cannot use /dev/ubi0_0 as seed device (non-existing or char device)
[Detaching after fork from child process 4630]
1421    in /usr/src/debug/rauc/1.9-r0/src/update_handler.c
(gdb) next
1434    in /usr/src/debug/rauc/1.9-r0/src/update_handler.c
(gdb) next
[Thread 0x74fff400 (LWP 4606) exited]
1435    in /usr/src/debug/rauc/1.9-r0/src/update_handler.c
(gdb) next
1444    in /usr/src/debug/rauc/1.9-r0/src/update_handler.c
(gdb) print res
$1 = 0
(gdb)
```

claudels, 2023-08-03:
I removed this block from the `img_to_ubivol_handler` function : 
```
res = g_output_stream_close(G_OUTPUT_STREAM(outstream), NULL, &ierror);
	if (!res) {
		g_propagate_error(error, ierror);
		return FALSE;
	}
```

The installation works fine with this patch but I guess it's not a good solution. Could you try to make a commit that fixes the problem correctly?

ejoerns, 2023-08-08:
@claudels Thank you for investigating deeper on this!

I now also had another look at your logs and findings. What catches my attention is the line

> rauc-Message: 10:47:04.430: Cannot use /dev/ubi0_0 as seed device (non-existing or char device)

which is present in both logs.

It originates from `casync_extract_image()` which notes above the line that emits the error message:

``` c
/* For the moment do not utilize UBI volumes as seed because they are
 * character devices - additional logic is needed to (temporarily) map
 * them to UBIBLOCK devices which are suitable for that purpose */
```

Thus I fear with blob seeds as used for the kernel and dtb, you will not get very far anyway.

I am not quite sure why `g_output_stream_close()` failed then in the following. It might be the case that the underlying file descriptor is closed by casync or somewhere else. But before digging deeper into this I would assume adding a manifest option to skip casync conversion is a more future-proof solution.

Also, note that the future of casync in RAUC is limited anyway and we are more focussing on adaptive updates as a concept for allowing delta-like updates. However, since there are a few key users and advantages of the casync feature, it will also not disappear for any currently foreseeable timescale.

@jluebbe What's your opinion on this?

claudels, 2023-08-08:
> @claudels Thank you for investigating deeper on this!
>
> I now also had another look at your logs and findings. What catches my attention is the line
>
>
> which is present in both logs.
>
> It originates [...]

I would have liked to use adaptative updates but in my system I guess that is not possible to do that with the ubifs slot right ?

ejoerns, 2023-08-08:
> I would have liked to use adaptative updates but in my system I guess that is not possible to do that with the ubifs slot right ?

Unfortunately not with the existing implemented method(s) I guess. For now, we have only `block-hash-index` which works for block devices only. I know there is mtdblock, but, I don't think it is the best idea to use this (neither know if it actually would work).

ejoerns, 2023-08-08:
@claudels I have drafted an option-based solution to ignore images during conversion here: https://github.com/rauc/rauc/pull/1210

claudels, 2023-08-09:
I applied your draft and it seems work :)
```

3 Images:
  [dtb]
        Filename:  imx6dl-gmc160.dtb.img
        Checksum:  38364de5e6b92dd1bc1bcbfffb75538fdc9edb88f270eed2c930f0fbda33f183
        Size:      44591
        Hooks:
  [kernel]
        Filename:  zImage.img
        Checksum:  8fe87b968b24dac1233ce97781076703830ad693e185a4684c316429836337e6
        Size:      5698944
        Hooks:
  [rootfs]
        Filename:  guinault-dev-image-gmc160.tar.caidx
        Checksum:  2ccd9ff70f8e93a6c234426637dca082d2eb5ab5ca18e30184aa9e03dd135176
        Size:      356454400
        Hooks:

```

ejoerns, 2023-08-09:
@claudels Sounds great, thanks for testing.

Since @jluebbe now agreed that it is a simple solution to the problem, I will then try to make an actual PR out of my draft as soon as I have time.

closed 2023-08-11 by ejoerns

jluebbe, 2023-08-14:
> > I would have liked to use adaptive updates but in my system I guess that is not possible to do that with the ubifs slot right ?
> 
> Unfortunately not with the existing implemented method(s) I guess. For now, we have only `block-hash-index` which works for block devices only. I know there is mtdblock, but, I don't think it is the best idea to use this (neither know if it actually would work).

In principle, `block-hash-index` could also work with UBI devices. Reading is not much different from block devices, while writing should be LEB (erase-block) aligned.

If anyone wants to implement that, they should first measure how many identical 4k blocks are in UBIFS of different software releases, to see how much download size could be saved.

A better approach would likely be to implement an adaptive method for directory trees.

PR #1210 (2023-08-11): src: implement --ignore-image option for rauc convert (casync)
This option allows providing the slotclass of an image to ignore during conversion. Multiple images can be ignored by using the argument multiple times.

This can be useful to generate 'mixed' casync bundles that contain both .caidx/.caibx images as well as conventional ones.

Usage:

    rauc convert [..] --ignore-image=rootfs input-bundle.raucb mixed-bundle.raucb

Fixes: #1200

Note: This is a proof-of-concept and a base for discussion only for now.