rauc/rauc #1202: Status doesn't show correct data when both partitions

Paiusco, 2023-08-01:
**Describe the bug**
Whenever 

**Background information**
Describe which RAUC version and build system you are using.

**To Reproduce**
Steps to reproduce the behavior:
1. …

**Expected behavior**
A clear and concise description of what you expected to happen.

**Logs**
If applicable, add console output or logs from the RAUC service to help explain your problem.

**Additional context**
Add any other context about the problem here.

ejoerns, 2023-08-01:
@Paiusco I am not sure if I understand the issue correctly.

> Would expect to have a clear overview of what bootloader partition was booted. (instead of (null) ((null)))

The information what bootloader partition is booted is in your above-mentioned log:

    Booted from: rootfs.0 (A)

The information shown in `Activated: ` reflects the boot target that would be booted on the *next* boot. If you disable both, none will be booted (let's exclude possible reset mechanisms in the bootloader here that RAUC cannot know about).

Thus I would assume the information shown in your logs looks quite consistent. Am I wrong?

> I think rauc only considers valid a partition that's bigger than 0, however if we max of tries is 1 or if it failed a few times up until it reached 0, booting from 0 left is also a valid one, until it restarts again before being marked as good.

If no mark-good was executed yet, the next boot will fail because of 0 attempts. If a mark-good was already executed, the information shown in 'Activated: ' should actually say 'good'. Thus the result depends on if you run `rauc status` before or after marking good.

Paiusco, 2023-08-02:
@ejoerns It was not clear for me that the `bootloader active:` would give me the next boot. Maybe I missed that on the documentation, thank you! (or can it be improved?)

However, I feel it doesn't look nice that the message becomes `(null) ((null))` instead of something a bit clearer like `Unable to find next boot partition`. What do you think?

ejoerns, 2023-08-02:
> @ejoerns It was not clear for me that the bootloader active: would give me the next boot.

Indirectly. It shows how the bootloader is currently configured. And this determines what will be booted

> Maybe I missed that on the documentation, thank you! (or can it be improved?)

Not sure where but suggestions are always welcome! Also for the user interface.

> However, I feel it doesn't look nice that the message becomes (null) ((null)) instead of something a bit clearer like Unable to find next boot partition. What do you think?

I'd agree that `(null) ((null))` is not the most user-friendly representation. If we see null, we could say:

    Activated: none

The warning emitted in your example already tells the verbose background: 

    Failed getting primary slot: uboot backend: Unable to find primary boot slot

> > If a mark-good was already executed, the information shown in 'Activated: ' should actually say 'good'
>
> That's not the case, right? ...

I actually meant it should say print the activated slot (not 'good') :wink:

ejoerns, 2023-08-29:
> I'd agree that (null) ((null)) is not the most user-friendly representation. If we see null, we could say:
> Activated: none

I have addressed this in #1239 now to gain a bit of enhancement out of this issue. 

@Paiusco Anything left from your site or could we close the issue then?

Paiusco, 2023-09-02:
@ejoerns I planned to send a patch, you were faster tho :)
I saw it and it looks good, that's all then. Thanks.

ejoerns, 2023-09-03:
@Paiusco Oh sorry, I didn't have in mind anymore that you actually promised to make a PR for it.
I bet there are even more options like this in RAUC :wink:

Thanks for your feedback, I will close it then.

closed 2023-09-03 by ejoerns