rauc/rauc #1216: rauc mark-active other failes on a grub / x86 system

davidhuwyler, 2023-08-11:
**Describe the bug**
"rauc status mark-active other" failes with the following error:
rauc mark: failed to activate slot rootfs.1: Failed marking 'rootfs.1' as primary: Failed to run grub-editenv: Child process exited with code 1

**Background information**
I use a custom board (x86 with grub bootloader).

system.conf file:
```
[slot.rootfs.0]

device=/dev/sda4

type=raw

bootname=A

 

[slot.rootfs.1]

device=/dev/sda6

type=raw

bootname=B

 

[slot.fitimage.0]

device=/dev/sda3

parent=rootfs.0

type=raw

 

[slot.fitimage.1]

device=/dev/sda5

parent=rootfs.1

type=raw

 

[slot.rescue.0]

device=/dev/sda2

type=raw

bootname=R

 

[slot.bootloader.0]

device=/dev/sda

type=boot-gpt-switch

region-start=4M

region-size=200M
```

rauc install failes with the same error as "rauc status mark-active other"

```
  0% Installing

  0% Determining slot states

20% Determining slot states done.

20% Checking bundle

20% Verifying signature

40% Verifying signature done.

40% Checking bundle done.

40% Checking manifest contents

60% Checking manifest contents done.

60% Determining target install group

80% Determining target install group done.

80% Updating slots

80% Checking slot rootfs.1

83% Checking slot rootfs.1 done.

83% Copying image to rootfs.1

86% Copying image to rootfs.1 done.

86% Checking slot fitimage.1

90% Checking slot fitimage.1 done.

90% Copying image to fitimage.1

93% Copying image to fitimage.1 done.

93% Checking slot bootloader.0

96% Checking slot bootloader.0 done.

96% Copying image to bootloader.0

99% Copying image to bootloader.0 done.

99% Updating slots failed.

100% Installing failed.

LastError: Installation error: Failed marking slot rootfs.1 bootable: failed to activate slot rootfs.1: Failed marking 'rootfs.1' as primary: Failed to run grub-editenv: Child process exited with code 1

Installing `/tmp/update_cache/payload.plain.img` failed
```

rauc status and rauc status mark-good work as expected:
```
#rauc status

=== System Info ===

Compatible:  TestDevice

Variant:

Booted from: rootfs.0 (A)

 

=== Bootloader ===

Activated: rootfs.1 (B)

 

=== Slot States ===

  [bootloader.0] (/dev/sda, boot-gpt-switch, inactive)

 

x [rootfs.1] (/dev/sda6, raw, inactive)

        bootname: B

        boot status: good

    [fitimage.1] (/dev/sda5, raw, inactive)

 

o [rootfs.0] (/dev/sda4, raw, booted)

        bootname: A

        boot status: good

    [fitimage.0] (/dev/sda3, raw, active)

        mounted: /var/volatile/tmp/fitimagePart-mnt

 

o [rescue.0] (/dev/sda2, raw, inactive)

        bootname: R

        boot status: bad

 

#rauc status mark-good other

rauc status: marked slot rootfs.1 as good

# rauc status mark-active other

rauc mark: failed to activate slot rootfs.1: Failed marking 'rootfs.1' as primary: Failed to run grub-editenv: Child process exited with code 1
```

**Additional context**
grub-editenv works (I can set a variable and list it afterwards)

I have no clue what the problem is...
Has someone encountered a similar problem or can help me out?

ejoerns, 2023-08-11:
Could you post the output of the RAUC service here?

If using systemd, you can get it from the journal:

    journalctl -u rauc

davidhuwyler, 2023-08-12:
No its a yocto system using SysVinit.
Is there a way to get the journal out with SysVinit?
Or can i kill the service, restart it and pipe the output into a file or something?

ejoerns, 2023-08-12:
@davidhuwyler restarting and viewing the log live on a second terminal or piping it into a file sounds like a good way, yes.

Anyway, a good SysVinit service configuration should already have a log file configured I'd say but that actually depends on your integration.

davidhuwyler, 2023-08-29:
Sorry I am on vacation... Ill start getting into the problem again soon...

davidhuwyler, 2023-08-30:
I was able to get the information now:

The rauc service has the following error:

```
rauc-Message: 21:43:11.615: Using central status file /mnt/config/ct-product-config/central-status.raucs
rauc-Message: 21:43:11.615: Using central status file /mnt/config/ct-product-config/central-status.raucs
grub-editenv: error: environment block too small.
rauc-Message: 21:43:15.550: rauc mark: failed to activate slot rootfs.1: Failed marking 'rootfs.1' as primary: Failed to run grub-editenv: Child process exited with code 1
```

Hmm do you know whats meant by "environment block"? 
`grub-editenv /boot/EFI/BOOT/grubenv list` shows the grub environment with no problem and I can change the variables by the set command...

ejoerns, 2023-08-30:
> I was able to get the information now:
>
> The rauc service has the following error:
>
> ```
> rauc-Message: 21:43:11.615: Using central status file /mnt/config/ct-product-config/central-status.raucs [...]

As far as I know it means that the content cannot cannot be stored in this file.

Listing should be possible as no content is added.

Try checking which commands are actually executed and check if invoking them manually also fails. Then it's a problem of your environment block/file (size).

You can let rauc print the subcommand calls by invoking the service manually with:

    G_MESSAGES_DEBUG="rauc rauc-subprocess" rauc service

davidhuwyler, 2023-08-30:
Nice you are correct.
My grubenv file got auto formated by clan-format --> it had a newline at the end.
but the grubenv file needs to be exactli 1024 bytes long with '#' characters filled...
Thanks for the support and sorry for the non rauc issue!

closed 2023-08-30 by davidhuwyler

ejoerns, 2023-08-30:
> Nice you are correct. My grubenv file got auto formated by clan-format --> it had a newline at the end. but the grubenv file needs to be exactli 1024 bytes long with '#' characters filled... Thanks for the support and sorry for the non rauc issue!

Glad you solved it! And thanks for the feedback!