rauc/rauc #1236: Attempt to use boot_raw_fallback on an mtd partition results in an assertion crash

esden, 2023-08-25:
**Describe the bug**
When using the boot_raw_fallback handler on `/dev/mtd*` the following error is produced by rauc:
```
Nov 08 14:04:30 helix rauc[739]: Updating /dev/mtd0 with /run/rauc/bundle/boot.bin.img
Nov 08 14:04:30 helix rauc[739]: Updating fallback partition at 3670016 on /dev/mtd0
Nov 08 14:04:40 helix rauc[739]: save_slot_status: assertion 'error == NULL || *error == NULL' failed
Nov 08 14:04:40 helix systemd[1]: rauc.service: Main process exited, code=killed, status=5/TRAP
Nov 08 14:04:40 helix systemd[1]: rauc.service: Failed with result 'signal'.
```
I would assume that it should actually print an error instead of crashing with an assertion violation.

**Background information**
I have checked where the error occurs and the handler fails to write to the partition.
https://github.com/rauc/rauc/blob/52dfd8ecedad9aa60e013fe71d2794de252babfa/src/update_handler.c#L2613

**To Reproduce**
Steps to reproduce the behavior:
1. setup a slot as `type=boot-raw-fallback`
2. setup that slot to use `device=/dev/mtd0`
3. attempt to update using `rauc install bundle.raucb`
4. see rauc service crash

**Expected behavior**
`rauc service` should not crash but rather report an update write error.

**Additional context**
This crash does not accur when writing the image manually using `rauc write-slot`. But it also silently fails without reporting of the write error.

ejoerns, 2023-08-29:
> rauc service should not crash but rather report an update write error.

I would assume that this is fixed by #1214. Do you have the ability to verify this?

The fix is in `master` but not in any release, yet.

ejoerns, 2023-08-29:
> This crash does not accur when writing the image manually using rauc write-slot. But it also silently fails without reporting of the write error.

This should be addresses and fixed by #1238 now.

ejoerns, 2023-08-29:
@esden Thanks for the detailed report and investigation!

The actual issue covered by the previously noted bugs is that `boot-raw-fallback` cannot be used for mtd devices since `fsync()` does not work here. The error you should have gotten is:

> Syncing content to disk failed: Invalid argument

But this is actually what you have found out in #1235 already.

esden, 2023-08-29:
Thanks, I will test it out and report if rauc is reporting the correct error.

esden, 2023-09-01:
Ok, I have tested it and it nicely errors out without crashing:
```
Sep 01 02:31:12 xxxx rauc[1049]: Updating fallback partition at 3670016 on /dev/mtd0
Sep 01 02:31:27 xxxx rauc[1049]: Updating slot bootloader.0 status
Sep 01 02:31:27 xxxx rauc[1049]: Installation error: Failed updating slot bootloader.0: Syncing content to disk failed: Invalid argument
Sep 01 02:31:27 xxxx rauc[1049]: installing /home/xxxx/data/home/xxxx/boot-update-bundle-zynq-generic-7z010.raucb: Installation error: Failed updating slot bootloader.0: Syncing content to disk failed: Invalid argument
Sep 01 02:31:27 xxxx rauc[1049]: installing /home/xxxx/data/home/xxxx/boot-update-bundle-zynq-generic-7z010.raucb: finished
Sep 01 02:31:27 xxxx rauc[1049]: installing `/home/xxxx/data/home/xxxx/boot-update-bundle-zynq-generic-7z010.raucb` failed: 1
```

esden, 2023-09-01:
Ok, I have tested it and it nicely errors out without crashing:
```
Sep 01 02:31:12 xxxx rauc[1049]: Updating fallback partition at 3670016 on /dev/mtd0
Sep 01 02:31:27 xxxx rauc[1049]: Updating slot bootloader.0 status
Sep 01 02:31:27 xxxx rauc[1049]: Installation error: Failed updating slot bootloader.0: Syncing content to disk failed: Invalid argument
Sep 01 02:31:27 xxxx rauc[1049]: installing /home/xxxx/data/home/xxxx/boot-update-bundle-zynq-generic-7z010.raucb: Installation error: Failed updating slot bootloader.0: Syncing content to disk failed: Invalid argument
Sep 01 02:31:27 xxxx rauc[1049]: installing /home/xxxx/data/home/xxxx/boot-update-bundle-zynq-generic-7z010.raucb: finished
Sep 01 02:31:27 xxxx rauc[1049]: installing `/home/xxxx/data/home/xxxx/boot-update-bundle-zynq-generic-7z010.raucb` failed: 1
```

I allow myself to close this issue. :)

closed 2023-09-01 by esden