rauc/rauc #1250: Question: Are multiple tries per partition possible in GRUB?

davidhuwyler, 2023-09-21:
Hi

I was not able to find an example of a GRUB configration to allow more than one try per rauc slot.
Maybe with "test" and "eval" or something?
Or is only one try per slot possible with GRUB?

Regards
Dave

jluebbe, 2023-09-21:
If I remember correctly, the main issue was decrementing the TRY counters. If that's possible in new grub versions, we'd be happy to take a PR implementing that (and adding a second example script to contrib).

mbessler, 2023-09-26:
I'm using the following patch for grub that implements a increment/decrement command:
0253-Reimplement-boot_counter.patch from https://src.fedoraproject.org/rpms/grub2/c/db4a99687c8306760702abda9c01a2e74d19a3f3
Direct link: https://src.fedoraproject.org/rpms/grub2/blob/db4a99687c8306760702abda9c01a2e74d19a3f3/f/0253-Reimplement-boot_counter.patch

I also have a patch for rauc that creates a boot loader type "grubalt" to work with this feature. The grub bootscript is basically a port of the rauc u-boot boot selection script. 

If there is any interest, I can share both.

davidhuwyler, 2023-09-27:
thanks mbessler, very interesting!
I'll try the patch if it works for me.

Ive got a double for loop working now in the grub bootscript to implement a 3 try a b system.
Not very pretty but functional...

mbessler, 2023-09-28:
Before I found this patch, I had used some "not very pretty" code to increment/decrement using an if/then/else construct that handled numbers from around 0-10. However should the value be something outside that range, it could not handle it well. 
Having the proper increment and decrement functions really helps.

jluebbe, 2023-10-24:
> I'm using the following patch for grub that implements a increment/decrement command: 0253-Reimplement-boot_counter.patch from [...]

@mbessler Do you know it they plan to upstream this change? I've not found it in https://git.savannah.gnu.org/cgit/grub.git/log/ so far.

> I also have a patch for rauc that creates a boot loader type "grubalt" to work with this feature. The grub bootscript is basically a port of the rauc u-boot boot selection script.
> 
> If there is any interest, I can share both.

I'd be interested! We could at least take a look if it could be integrated into RAUC without too much overhead. As long as a patched GRUB is needed, I'd like to avoid unnecessary extra code.

jluebbe, 2023-10-24:
A related discussion can be found at https://pagure.io/fedora-iot/issue/12.

mbessler, 2023-10-25:
@jluebbe no idea, I don't run or follow Fedora/RH, just found the patch useful. It would seem that the increment/decrement would not be a controversial change, but I have not looked into Grub upstream (other than checking for the existence of the feature).

Grubalt bootchooser pull request opened: https://github.com/rauc/rauc/pull/1275