rauc/rauc #1277: Detailed progress reporting not working with adaptive update bundles

BubuOT, 2023-10-30:
**Describe the bug**
The more detailed progress reporting currently seems not to be working with adaptive update bundles:

```
 70% Checking slot rootfs.1
 73% Checking slot rootfs.1 done.
 73% Copying image to rootfs.1
 99% Copying image to rootfs.1 done.
```

**Background information**
Rauc 1.10.1
System image/update bundles built with buildroot.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a bundle with `adaptive=block-hash-index`
2. Install to a target

**Expected behavior**
Rauc produces intermediate progress reporting while updating a slot. Instead rauc only prints the next progress line after updating a single slot is done.

**Logs**
See above.

**Additional context**
-

jluebbe, 2024-11-19:
See also https://github.com/rauc/rauc/discussions/1564#discussioncomment-11301626

ejoerns, 2024-11-22:
FTR: We don't consider this a real bug, but a missing implementation.
(There are also other handlers in RAUC that do not support fine-grained progress, yet.)
However, having support for this would be great.

PR #1619 (2025-02-07): src/update_handler: add detailed progress when installing block hash index images
Use the number of chunks in the source image for tracking installation progress.

Fixes: #1277
closed 2025-02-07 by ejoerns