rauc/rauc #1287: Installation error: Failed updating slot boot.0: failed to splice data to tar: Broken pipe

gambels, 2023-11-24:
**Describe the bug**
After updating rauc it is not possible anymore to install an update bundle. The installation fails during updating the boot slot. The boot file is compressed.

**Background information**
The following rauc version and build system is used.
- rauc: 1.9 -> 1.10.1
- yocto: kirkstone
  - poky: b1ddd4178d469cf0c14e8c31d0fd2ba1b53bf293
  - meta-rauc: 804386f04826ea7fe74a1b4e0a20991425c06d9c

Busybox was built with the following configs:
- CONFIG_FEATURE_TAR_LONG_OPTIONS=y
- CONFIG_FEATURE_TAR_AUTODETECT=y

**To Reproduce**
Steps to reproduce the behavior:
1. Create update bundle with compressed boot file "boot.tar.gz"
2. Install update bundle with rauc 1.9 (updates rauc to 1.10.1)
3. Re-Install update bundle with rauc 1.10.1

**Expected behavior**
My expectation would be that installing a compressed boot file still works after updating rauc from 1.9 to 1.10.1.

**Logs**
 20:54:06 phyboard-polis-imx8mm-5 rauc[531]: installing /tmp/bundle-phyboard-polis-imx8mm-5.raucb: Checking slot boot.0
Nov 23 20:54:06 phyboard-polis-imx8mm-5 rauc[531]: Updating /dev/mmcblk2p1 with /run/rauc/bundle/boot.tar.gz.img
Nov 23 20:54:06 phyboard-polis-imx8mm-5 rauc[531]: installing /tmp/bundle-phyboard-polis-imx8mm-5.raucb: Updating slot boot.0
Nov 23 20:54:06 phyboard-polis-imx8mm-5 rauc[531]: Formatting vfat slot /dev/mmcblk2p1                     
Nov 23 20:54:06 phyboard-polis-imx8mm-5 rauc[710]: mkfs.fat 4.2 (2021-01-31)                                                  
Nov 23 20:54:06 phyboard-polis-imx8mm-5 rauc[531]: Mounting vfat slot /dev/mmcblk2p1                                          
Nov 23 20:54:06 phyboard-polis-imx8mm-5 rauc[531]: Extracting /run/rauc/bundle/boot.tar.gz.img to /run/rauc/boot.0      
Nov 23 20:54:06 phyboard-polis-imx8mm-5 rauc[712]: tar: invalid tar magic                             
Nov 23 20:54:06 phyboard-polis-imx8mm-5 rauc[531]: Unmounting vfat slot /dev/mmcblk2p1                                          
Nov 23 20:54:06 phyboard-polis-imx8mm-5 rauc[531]: Installation error: Failed updating slot boot.0: failed to splice data to tar: Broken pipe 

**Additional context**
I already figured out that the rauc function untar_image() has changed from 1.9 to 1.10.1. Instead of setting the boot image file by the tar option '-f' it is now read over stdin.

The behaviour could be also reproduced by calling the busybox tar directly
- rauc-1.9: "tar -xf boot.tar.gz.img -C <dest> --numeric-owner"
- rauc-1.10.1: "cat boot.tar.gz.img | tar -xf - -C <dest> --numeric-owner" (tar: invalid tar magic)

I am not sure, if this is really a bug in rauc, or maybe compressed boot files are not supported anymore while reading from stdin.

jluebbe, 2023-11-24:
The `.img` suffix is breaking the compression detection in RAUC, which had to be added to support fine grained progress in 23b96a1f272510f70b524a5eae2d067567bf246f (see https://github.com/rauc/rauc/pull/1093). I suspect this suffix is added erroneously by the `bundle.bbclass` in `meta-rauc`. @ejoerns?

ejoerns, 2023-11-24:
@jluebbe @gambels Yes, I would expect the `.img` suffix is a result of `bundle.bbclass` handling.

For specific image types, it adds the `.img` suffix for convenience reasons, because RAUC accepts only a very limited set of image suffixes.

A `.img` suffix is added for `[type]`s `kernel` and `file`. What `[type]` did you specify for the slot image?

gambels, 2023-11-24:
@ejoerns I specified the type `file` for the slot image.

Slot image:
`RAUC_SLOT_boot[file] = "boot.tar.gz"`

ejoerns, 2023-11-24:
Btw. this is a duplicate of https://github.com/rauc/rauc/discussions/1203. Since it is not really a RAUC bug I will close this thread now and point to this GitHub Discussion for more.

The workaround to use `[rename]` there should work for you, too.

On the other hand, we should probably add a better convenience there if this use case tends to repeat itself.
Suggestions are welcome in #1203 or in a new meta-rauc issue/discussion.

closed 2023-11-24 by ejoerns

gambels, 2023-11-24:
Thank you very much for the quick feedback and for the hint regarding #1203.