rauc/rauc #1297: Failed mounting bundle: failed to resolve 'nbd'

intera91, 2023-12-18:
rauc install http://10.0.0.10/bundles/update.raucb
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
100% Installing failed.
LastError: Failed mounting bundle: failed to resolve 'nbd' netlink family - BLK_DEV_NBD not enabled in kernel?

Obviously all necessary CONFIGs are in the kernel, so what gives?

ONFIG_BLK_DEV_LOOP=y
CONFIG_SQUASHFS=y
CONFIG_MD=y
CONFIG_BLK_DEV_DM=y
CONFIG_BLK_DEV_NBD=y
CONFIG_DM_VERITY=y
CONFIG_CRYPTO_SHA256=y

Thanks

jluebbe, 2023-12-18:
Which kernel version are you using? Could your attach your `.config`?

intera91, 2023-12-18:
Kernel is 4.9.253
```
#
# Automatically generated file; DO NOT EDIT.
# Linux/arm64 4.9.253 Kernel Configuration
#
CONFIG_ARM64=y
CONFIG_64BIT=y
CONFIG_ARCH_PHYS_ADDR_T_64BIT=y
CONFIG_MMU=y
CONFIG_DEBUG_RODATA=y
CONFIG_ARM64_PAGE_SHIFT=12
CONFIG_ARM64_CONT_SHIFT=4
CONFIG_ARCH_MMAP_RND_BITS_MIN=18
CONFIG_ARCH_MMAP_RND_BITS_MAX=24
CONFIG_ARCH_MMAP_RND_COMPAT_BITS_MIN=11
CONFIG_ARCH_MMAP_RND_COMPAT_BITS_MAX=16
CONFIG_STACKTRACE_SUPPORT=y
CONFIG_ILLEGAL_POINTER_VALUE=0xdead000000000000
CONFIG_LOCKDEP_SUPPORT=y
CONFIG_TRACE_IRQFLAGS_SUPPORT=y
CONFIG_RWSEM_XCHGADD_ALGORITHM=y
CONFIG_GENERIC_BUG=y
CONFIG_GENERIC_BUG_RELATIVE_POINTERS=y
CONFIG_GENERIC_HWEIGHT=y
CONFIG_GENERIC_CSUM=y
CONFIG_GENERIC_CALIBRATE_DELAY=y
CONFIG_ZONE_DMA=y
CONFIG_HAVE_GENERIC_RCU_GUP=y
CONFIG_ARCH_DMA_ADDR_T_64BIT=y
CONFIG_NEED_DMA_MAP_STATE=y
CONFIG_NEED_SG_DMA_LENGTH=y
CONFIG_SMP=y
# CONFIG_IOMMU_HELPER is not set
CONFIG_KERNEL_MODE_NEON=y
CONFIG_FIX_EARLYCON_MEM=y
CONFIG_PGTABLE_LEVELS=3
# CONFIG_ARM64_NON_SHARED_TLBI is not set
CONFIG_DEFCONFIG_LIST="/lib/modules/$UNAME_RELEASE/.config"
CONFIG_IRQ_WORK=y
CONFIG_BUILDTIME_EXTABLE_SORT=y
CONFIG_THREAD_INFO_IN_TASK=y

#
# General setup
#
CONFIG_INIT_ENV_ARG_LIMIT=32
CONFIG_CROSS_COMPILE=""
# CONFIG_COMPILE_TEST is not set
CONFIG_LOCALVERSION="Guidance-Yocto-0.1.0.75""
# CONFIG_LOCALVERSION_AUTO is not set
CONFIG_DEFAULT_HOSTNAME="(none)"
CONFIG_SWAP=y
CONFIG_SYSVIPC=y
CONFIG_SYSVIPC_SYSCTL=y
CONFIG_POSIX_MQUEUE=y
CONFIG_POSIX_MQUEUE_SYSCTL=y
CONFIG_CROSS_MEMORY_ATTACH=y
CONFIG_FHANDLE=y
# CONFIG_USELIB is not set
CONFIG_AUDIT=y
CONFIG_HAVE_ARCH_AUDITSYSCALL=y
CONFIG_AUDITSYSCALL=y
CONFIG_AUDIT_WATCH=y
CONFIG_AUDIT_TREE=y

#
# IRQ subsystem
#
CONFIG_GENERIC_IRQ_PROBE=y
[... 5591 lines cut ...]
CONFIG_GLOB=y
# CONFIG_GLOB_SELFTEST is not set
CONFIG_NLATTR=y
CONFIG_CLZ_TAB=y
# CONFIG_CORDIC is not set
# CONFIG_DDR is not set
CONFIG_IRQ_POLL=y
CONFIG_MPILIB=y
CONFIG_LIBFDT=y
CONFIG_FONT_SUPPORT=y
# CONFIG_FONTS is not set
CONFIG_FONT_8x8=y
CONFIG_FONT_8x16=y
# CONFIG_SG_SPLIT is not set
CONFIG_SG_POOL=y
CONFIG_ARCH_HAS_SG_CHAIN=y
CONFIG_SBITMAP=y

#
# Tegra BPMP Driver
#
CONFIG_ARCH_TEGRA_18x_SOC=y
CONFIG_ARCH_TEGRA_19x_SOC=y
CONFIG_ARCH_TEGRA_23x_SOC=y
```

ejoerns, 2023-12-18:
> Kernel is 4.9.253

This Kernel is from end 2016 and reached EOL earlier this year. I'd strongly suggest updating to a recent one.
Is this a new development or an ongoing project that should be equipped with OTA functionality, now?

ejoerns, 2023-12-18:
FTR: netlink API for NBD was only added with kernel v4.12 in 2017.

intera91, 2023-12-19:
Yes, I inherited the project when I joined the company. Didn’t make any of the decisions initially.
Should I upgrade the kernel to 4.12? In meta-tegra?
And yes are need OTA now

intera91, 2023-12-19:
> When we added streaming, we expected that any new integration of RAUC would start with a relatively recent kernel, as otherwise a kernel update would be required anyway (due to security). And it's easier to update the kernel first before building a system on top rather than the other way around. :)
> So, streaming is not possible on this old kernel, but you should be able to download the bundle to a temporary partition (outside of RAUC) and install it from there, as NBD is not used in that case. That bundle could then contain a still supported kernel and all of those support NBD via netlink, so the next update after that could use streaming.

Thank you for that reply.My understanding of the partitions scheme of the Xavier NX means that the kernel and boot partitions are separate from the userspace partitions that we are updating (these partitions only contain user programs and data) . Henceforth the rescue partition (which contains RAUC) is using the same kernel as our main partition (4.9.253) which comes from meta-tegra, dunfell branch. In other words I have to rebuild the whole system from scratch using the master branch of meta-tegra which should give me a 5.10 kernel. OE branch had to be updated as well obviouslySo while this ambitious program is ongoing we will have to download the bundle to the SSD inside the Xavier and update from there.At the moment am struggling with the following error: rauc install /mnt/nvme/update.raucbinstalling  0% Installing  0% Determining slot states 20% Determining slot states done. 20% Checking bundle 20% Verifying signature 40% Verifying signature done. 40% Checking bundle done. 40% Checking manifest contents 60% Checking manifest contents done.100% Installing failed.LastError: Failed mounting bundle: Failed to load dm table: Invalid argumentInstalling `/mnt/nvme/update.raucb` failedAny pointer appreciated :-)
    	Sent from Outlook for iOS

jluebbe, 2023-12-19:
> My understanding of the partitions scheme of the Xavier NX means that the kernel and boot partitions are separate from the userspace partitions that we are updating (these partitions only contain [...]

Sharing kernel between two partitions is problematic anyway, because that means that during a kernel update, the rescue partition is not bootable, which means that any update is not atomic.

> So while this ambitious program is ongoing we will have to download the bundle to the SSD inside the Xavier and update from there.
>
> At the moment am struggling with the following error:
>
> rauc [...]

Did you follow https://rauc.readthedocs.io/en/latest/integration.html#kernel-configuration? It looks like `CONFIG_DM_VERITY=y` is missing from the config you posted.

jluebbe, 2023-12-19:
I've created a PR to make this requirement move obvious: https://github.com/rauc/rauc/pull/1298

intera91, 2023-12-19:
Still getting the error:

LastError: Failed mounting bundle: Failed to load dm table: Invalid argument

despite having  CONFIG_DM_VERITY=y set

any pointer appreciated

Thanks

jluebbe, 2023-12-19:
Try setting up a dm-verity target with `veritysetup` from https://gitlab.com/cryptsetup/cryptsetup to make sure that your kernel works. Unless that works, RAUC can't work either.

ejoerns, 2024-02-02:
@intera91 Any progress on this? Otherwise, I would like to close it.

intera91, 2024-02-05:
Issues can be closed thanks

closed 2024-02-05 by intera91