rauc/rauc #1311: 1.11: Tests related to service and bundle fail

dvzrv, 2023-12-23:
**Describe the bug**
Hi! :wave: 
When trying to package 1.11 for Arch Linux I ran the tests (`meson test -C build`).
Unfortunately the tests for `service` and `bundle` fail:

```
ninja: Entering directory `/build/rauc/src/build'
[1/1] Generating version.h with a custom command
 1/23 boot_raw_fallback OK               0.01s
 2/23 bootchooser       OK               1.79s
 3/23 checksum          OK               0.01s
 4/23 config_file       OK               0.16s
 5/23 context           OK               0.05s
 6/23 dm                OK               0.01s
 7/23 event_log         OK               0.06s
 8/23 hash_index        OK               0.01s
 9/23 manifest          OK               0.01s
10/23 signature         OK               0.03s
11/23 update_handler    OK               1.05s
12/23 utils             OK               0.01s
13/23 install           OK               0.53s
14/23 service           FAIL             0.14s   killed by signal 6 SIGABRT
>>> ASAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1 UBSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 MALLOC_PERTURB_=62 /build/rauc/src/build/test/servic
e-test

15/23 status_file       OK               0.09s
16/23 bundle            FAIL             1.03s   killed by signal 6 SIGABRT
>>> MALLOC_PERTURB_=20 ASAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1 UBSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 /build/rauc/src/build/test/bundle
-test

17/23 progress          OK               0.01s
18/23 slot              OK               0.01s
19/23 stats             OK               0.01s
20/23 network           OK               0.82s
21/23 nbd               OK               0.02s
22/23 boot_switch       OK               0.01s
23/23 rauc.t            OK               1.70s

Ok:                 21
Expected Fail:      0
Fail:               2
Unexpected Pass:    0
Skipped:            0
Timeout:            0

Full log written to /build/rauc/src/build/meson-logs/testlog.txt
```

**Background information**
1.11 from signed source tarball

**To Reproduce**
Steps to reproduce the behavior:
1. `arch-meson $pkgname-$pkgver build` (see https://gitlab.archlinux.org/archlinux/packaging/packages/meson/-/blob/97f7805dd7f5c9421e9b222d22a22a195c7a3c6d/arch-meson)
2. `meson compile -C build`
3. `meson test -C build`

**Expected behavior**
All tests pass.

**Logs**
Full meson test log: 
[testlog.txt](https://github.com/rauc/rauc/files/13758848/testlog.txt)

pkgctl build logs:
[rauc-1.11-1-x86_64-check.log](https://github.com/rauc/rauc/files/13758869/rauc-1.11-1-x86_64-check.log)
[rauc-1.11-1-x86_64-build.log](https://github.com/rauc/rauc/files/13758870/rauc-1.11-1-x86_64-build.log)

**Additional context**
n/a

dvzrv, 2023-12-23:
FWIW, the service test suite seems to fail because it can not start the systemd service via dbus (which I guess is to be expected when running in an systemd-nspawn container, that does not have that dbus/systemd service installed).

The bundle test failing seems to be due to a malformed signature:
```
Verify error: unsuitable certificate purpose (r_signature_error_quark, 16)
```

jluebbe, 2023-12-23:
For the service check, please take a look at https://github.com/rauc/rauc/pull/1310.

Also, please note that the recommended way to run the tests is via ./qemu-test: https://github.com/rauc/rauc?tab=readme-ov-file#running-the-test-suite
Otherwise, large parts of the test suite cannot run due to missing privileges or emulated devices.

jluebbe, 2024-01-02:
@dvzrv I can't reproduce the certificate purpose error on Debian, but have no experience with arch. Do you have a Dockerfile to prepare a build environment for RAUC somewhere? Or just the commands I'd need to install the build dependencies?

BubuOT, 2024-01-05:
I ran into the `Verify error: unsuitable certificate purpose` error now. (In an actual test system, not the rauc tests) 

This seems to be caused by upgrading openssl to 3.2.0, which fails the same check even after downgrading back to rauc 1.10.1

ejoerns, 2024-01-07:
The OpenSSL error may be caused by the fact that OpenSSL now defines 'codesign' itself: https://github.com/openssl/openssl/commit/178696d6020878361a088086243d56203e0beaa9

I guess this conflicts with our definition.

jluebbe, 2024-01-10:
For now, we are going to rename our purpose check internally to `codesign-rauc`. That should keep the existing behavior and avoid the conflict with the OpenSSL definition.

PR #1319 (2024-01-11): Resolve Conflict With OpenSSL's built-in Check Purpose 'codesign'
OpenSSL so far defined a number of certificate check purposes but none for 'codesign'. Thus RAUC added its own purpose check for this.

Since v3.2.0, OpenSSL now defines a 'codesign' check purpose itself which not only conflicts with the RAUC variant in naming, but also has some differences compared to the RAUC check:

* OpenSSL enforces `Key Usage` existence, while RAUC only checks that it allows digital signatures if it exists
* RAUC enforces that `Extended Key Usage` on CA certs allows code signing if it exists

While RAUC is [... cut]
closed 2024-01-11 by jluebbe