rauc/rauc #1312: [PATCH] Double-free crash when calling InstallBundle twice (successfully or otherwise)

ars3niy, 2024-01-03:
**Describe the bug**

```
From 70c048d56c2d97204178ff85a2ef8f79ae85af17 Mon Sep 17 00:00:00 2001
From: Arseniy Lartsev <arseniy.lartsev@qodes.se>
Date: Tue, 2 Jan 2024 17:53:40 +0100
Subject: [PATCH] Fix double free

---
 src/bundle.c | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/src/bundle.c b/src/bundle.c
index 8a3f199..f7050ad 100644
--- a/src/bundle.c
+++ b/src/bundle.c
@@ -1955,7 +1955,7 @@ gboolean check_bundle(const gchar *bundlename, RaucBundle **bundle, CheckBundleP
 
 	/* Determine store path for casync, defaults to bundle */
 	if (r_context()->config->store_path) {
-		ibundle->storepath = r_context()->config->store_path;
+		ibundle->storepath = g_strdup(r_context()->config->store_path);
 	} else {
 		gchar *path = ibundle->origpath ?: ibundle->path;
 
-- 
2.35.3

```

ejoerns, 2024-01-03:
@ars3niy Looks valid, thank you! :+1:

Would you like to make a pull-request for this?

jluebbe, 2024-01-10:
Alternatively, could you provide a SoB line? (`Signed-off-by: Arseniy Lartsev <arseniy.lartsev@qodes.se>`)

ars3niy, 2024-01-10:
```
From a8b0250b3c26bbd4ebafb2ddc0e6e8614fc257ee Mon Sep 17 00:00:00 2001
From: Arseniy Lartsev <arseniy.lartsev@qodes.se>
Date: Tue, 2 Jan 2024 17:53:40 +0100
Subject: [PATCH] Fix double free

---
 src/bundle.c | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/src/bundle.c b/src/bundle.c
index b2f0703..05ec358 100644
--- a/src/bundle.c
+++ b/src/bundle.c
@@ -1949,7 +1949,7 @@ gboolean check_bundle(const gchar *bundlename, RaucBundle **bundle, CheckBundleP
 
 	/* Determine store path for casync, defaults to bundle */
 	if (r_context()->config->store_path) {
-		ibundle->storepath = r_context()->config->store_path;
+		ibundle->storepath = g_strdup(r_context()->config->store_path);
 	} else {
 		gchar *path = ibundle->origpath ?: ibundle->path;
 
-- 
2.35.3

```

ars3niy, 2024-01-10:
I'll try to get away with not making pull request for now

ejoerns, 2024-01-10:
> I'll try to get away with not making pull request for now

@ars3niy Thank you for the Signed-off-by, I've created a PR for you in #1320

PR #1320 (2024-01-10): src/bundle: fix double free
Closes #1312
closed 2024-01-10 by jluebbe