rauc/rauc #1313: Rauc service core dump on 1.11

hcl-dr, 2024-01-05:
**Describe the bug**
Rauc service core dump running from external media (USB) (first install scenario)

**Background information**
1.11

**To Reproduce**
Boot

**Expected behavior**
Service should run (ok on 1.10.1)

**Logs**
```
Jan 05 07:09:21 dr-imx8mp systemd[1]: Starting RAUC Update Service...
Jan 05 07:09:21 dr-imx8mp rauc[698]: Using central status file /service/rauc/status
Jan 05 07:09:21 dr-imx8mp rauc[698]: Failed to load system status: No such file or directory
Jan 05 07:09:21 dr-imx8mp rauc[698]: Getting Systeminfo: /usr/lib/rauc/info-provider.sh
Jan 05 07:09:21 dr-imx8mp rauc[698]: Detected explicit external boot, ignoring missing active slot
Jan 05 07:09:21 dr-imx8mp rauc[698]: r_slot_status_load: assertion 'dest_slot' failed
Jan 05 07:09:21 dr-imx8mp systemd[1]: rauc.service: Main process exited, code=dumped, status=5/TRAP
Jan 05 07:09:21 dr-imx8mp systemd[1]: rauc.service: Failed with result 'core-dump'.
Jan 05 07:09:21 dr-imx8mp systemd[1]: Failed to start RAUC Update Service.
```

**Additional context**
Boots from USB stick targeting install on eMMC

jluebbe, 2024-01-05:
Please provide your `system.conf`, `rauc status` and `rauc info <bundle>` output.

hcl-dr, 2024-01-05:
Rauc core dumps, so no status/info available.

```
[system]
compatible=dr-imx8mp
bootloader=barebox
variant-dtb=true
statusfile=/service/rauc/status

[keyring]
path=/etc/rauc/keyring.pem
check-purpose=codesign

[handlers]
system-info=/usr/lib/rauc/info-provider.sh

[slot.rootfs.0]
device=/dev/mmcblk0p1
type=ext4
bootname=system0
resize=true

[slot.rootfs.1]
device=/dev/mmcblk0p2
type=ext4
bootname=system1
resize=true

[slot.bootloader.0]
device=/dev/mmcblk0
type=boot-emmc

```

ejoerns, 2024-01-05:
@hcl-dr Thank you for the report!

It's easy to reproduce here when starting rauc with `rauc.external` in cmdline. It seems to be caused by the boot detection code at: https://github.com/rauc/rauc/blob/cc03d35f9876ce55706c08472daaf3179a18e701/src/main.c#L2084

ejoerns, 2024-01-09:
FTR: I have prepared a fix with #1314.

PR #1314 (2024-01-12): src/main: fix and add support for boot detection with rauc.external
The current boot detection handling did not take into account that there is no booted slot when using `rauc.external` in cmdline and called `r_slot_status_load()` with `NULL` argument causing a

    (rauc:676): rauc-CRITICAL **: 16:16:16.336: r_slot_status_load: assertion 'dest_slot' failed

Fix this by explicitly checking for external boot and also emit a dedicated log message since this might be relevant for the log, too.

Fixes #1313
Fixes b9e4cdce0879862126aeb786bdd7f6672e392ca9 ("src/main: log boot and service restart")
commit 2928ac06 (2024-01-12): src/main: fix and add support for boot detection with rauc.external

The current boot detection handling did not take into account that there
is no booted slot when using 'rauc.external' in cmdline and called
r_slot_status_load() with NULL argument causing a

| (rauc:676): rauc-CRITICAL **: 16:16:16.336: r_slot_status_load: assertion 'dest_slot' failed

Fix this by explicitly checking for external boot and also emit a
dedicated log message since this might be relevant for the log, too.

Fixes #1313
Fixes b9e4cdce ("src/main: log boot and service restart")
closed 2024-01-12 by jluebbe