rauc/rauc #1339: no root device or rauc slot info in /proc/cmdline

intera91, 2024-02-02:
meta-tegra 35-4-1
rauc 1-11

flash ok, then any rauc command returns

nvidia@jetson-xavier-nx-devkit-emmc:~$ rauc status
Error retrieving slot status via D-Bus: error creating proxy: Error calling StartServiceByName for de.pengutronix.rauc: Timeout was reached
Any hint appreciated

patch applied:

diff --git a/src/context.c b/src/context.c
index 0de5b8b..430a993 100644
--- a/src/context.c
+++ b/src/context.c
@@ -62,12 +62,22 @@ static gchar* get_cmdline_bootname(void)
        g_autofree gchar *contents = NULL;
        g_autofree gchar *realdev = NULL;
        gchar *bootname = NULL;
+       const char *suffix = NULL;

        if (context->mock.proc_cmdline)
                contents = g_strdup(context->mock.proc_cmdline);
        else if (!g_file_get_contents("/proc/cmdline", &contents, NULL, NULL))
                return NULL;

+       suffix = regex_match("boot\\.slot_suffix=((\\s+|\\S+))", contents);
+       if(suffix) {
+               if(g_strcmp0(suffix, " ") == 0) {
+                       return "APP";
+               }else if(g_strcmp0(suffix, "_b") == 0) {
+                       return "APP_b";
+               }
+       }
+
        if (strstr(contents, "rauc.external") != NULL)
                return g_strdup("_external_");

jluebbe, 2024-02-02:
What is the contents of your kernel commandline (`cat /proc/cmdline`)? It should use one of the alternatives documented here: https://rauc.readthedocs.io/en/latest/integration.html#identification-via-kernel-commandline

jluebbe, 2024-02-02:
> Error calling StartServiceByName for de.pengutronix.rauc: Timeout was reached

This looks like the service doesn't start correctly. What does the service log during startup? (see https://rauc.readthedocs.io/en/latest/using.html#debugging-rauc)

intera91, 2024-02-02:
cat /proc/cmdline
console=ttyTCU0,115200 console=tty0 fbcon=map:0 video=efifb:off nospectre_bhb

intera91, 2024-02-02:
G_MESSAGES_DEBUG="rauc rauc-subprocess" rauc service
(rauc:889): rauc-DEBUG: 14:06:24.709: No value for key "max-bundle-download-size" in [system] defined - using default value of 8388608 bytes.
rauc-Message: 14:06:24.709: No data directory or status file set, falling back to per-slot status.
Consider setting 'data-directory=<path>' or 'statusfile=<path>/per-slot' explicitly.
rauc-Message: 14:06:24.710: Using per-slot statusfile. System status information not supported!
(rauc:889): rauc-DEBUG: 14:06:24.710: Obtained system boot ID: 'edc76c63-bdfc-4c0a-80d7-17c5f7da4b72'
(rauc:889): rauc-DEBUG: 14:06:24.710: Obtained system machine ID: '1a12709bd9aa4682960b3fcbed498982'
(rauc:889): rauc-DEBUG: 14:06:24.710: service start
(rauc:889): rauc-DEBUG: 14:06:24.710: Found booted slot: recovery.0 on /dev/mmcblk0p2
rauc-Message: 14:06:24.710: Started RAUC service
rauc-event-Message: 14:06:24.710: Service started
Failed to obtain name de.pengutronix.rauc on system bus

intera91, 2024-02-02:
Another patch applied

diff --git a/src/context.c b/src/context.c
index 430a993..1762fb1 100644
--- a/src/context.c
+++ b/src/context.c
@@ -428,7 +428,7 @@ static gboolean r_context_configure_target(GError **error)
                g_warning("Ignoring surrounding whitespace in system variant: %s", context->config->system_variant);

        if (context->bootslot == NULL) {
-               context->bootslot = get_cmdline_bootname();
+               context->bootslot = "/dev/mmcblk0p2"; //get_cmdline_bootname();
        }

        if (context->bootslot == NULL && (g_strcmp0(context->config->system_bootloader, "custom") == 0)) {
        
Now getting rauc status to answer correctly

ejoerns, 2024-02-20:
@intera91 Any update on this? If this is about the tegra bootloader, maybe https://github.com/rauc/rauc/pull/1215 solves your problem?

intera91, 2024-02-20:
Problem solved :-)

closed 2024-02-20 by intera91

ejoerns, 2024-02-20:
Ok, thanks for the update! If you would like to share with others how you solved it, just feel free. :wink: