rauc/rauc #1347: Bundle creation problem with key

amorirReverse, 2024-02-07:
Hello everyone,

I have a problem when I want to create a bundle with the command : 
rauc --cert demo.cert.pem --key demo.key.pem bundle ./test/temp-dir/ update-2019.01-1.rauc

I get this problem: 
[image]

Failed to create bundle: Key file contains line "EOF" which is not a key-value pair, group, or comment

amorirReverse, 2024-02-07:
@ejoerns no my key is different from my manifest

amorirReverse, 2024-02-07:
the problem is solved, I've deleted EOF from my manifest, thanks for your help.

ejoerns, 2024-02-07:
I would tend to agree that it is not entirely clear from the error message that loading the manifest failed.

Maybe https://github.com/rauc/rauc/pull/1348 can improve the error message here.

"Key file" might still be a bit misleading in context of RAUC where we primarily have *keys* for cryptography, but this is the terminology used for the key-value-file handling in glib, thus not something that we would or could change :wink: