rauc/rauc #1351: Problème rauc status

amorirReverse, 2024-02-20:
Bonjours,

J'ai un soucis, j'utilise rauc avec un bootloader custom. Lorsque je fais rauc status. 
Le terminale m'affiche cela : 
`maintenance@rescue2:~ $ rauc status
rauc-Message: 14:48:52.100: Config option 'statusfile=<path>/per-slot' unset, falling back to per-slot status
rauc-Message: 14:48:52.101: Using per-slot statusfile
Failed to determine slot states: Could not find any root device or rauc slot information in /proc/cmdline`

jluebbe, 2024-02-20:
Please use English. You can edit your comment using the 'edit' action in the three-dot-menu on the top right.

ejoerns, 2024-02-20:
@amorirReverse Do you pass any slot information via kernel commandline? Like `rauc.slot` or a `root=` rauc can identify as the root device? See https://rauc.readthedocs.io/en/latest/integration.html#identification-via-kernel-commandline

What kind of bootloader do you try to interface with?

Another option might be: https://github.com/rauc/rauc/pull/1215

amorirReverse, 2024-02-20:
I can't use the detection via kernel commandline, because the cmbline.txt hasn't the information : root=....