rauc/rauc #1360: Service fails to start with an assertion error in r_slot_status_load() after updating from v1.10.1 to v1.11.2

jacmet, 2024-03-06:
rauc service fails with an assertion error in r_slot_status_load() as it get called with a NULL pointer after updating to rauc v1.11.2. The same config works on v1.10.1 without issues.

v.10.1 log (booted B):
```
rauc service -d
rauc-Message: 09:00:01.613: Debug log domains: 'rauc'
rauc-Message: 09:00:01.617: Using central status file /mnt/data/rauc.status
(rauc:608): rauc-DEBUG: 09:00:01.618: Resolved bootname /dev/disk/by-partlabel/root-b to /dev/mmcblk0p6
(rauc:608): rauc-DEBUG: 09:00:01.618: Obtained system boot ID: 'd61f5b03-7a2b-47e2-b8b8-d0d831d98060'
rauc-INFO: 09:00:01.618: Failed to get machine-id: Failed to open file ?/etc/machine-id?: No such file or directory
(rauc:608): rauc-DEBUG: 09:00:01.618: Obtained system machine ID: '(null)'
(rauc:608): rauc-DEBUG: 09:00:01.618: service start
(rauc:608): rauc-DEBUG: 09:00:01.618: Found booted slot: rootfs.1 on /dev/disk/by-partlabel/root-b
(rauc:608): rauc-DEBUG: 09:00:01.624: name 'de.pengutronix.rauc' acquired
```

v1.11.2 (booted A):
```
rauc service -d
rauc-Message: 08:55:17.867: Debug log domains: 'rauc'
rauc-Message: 08:55:17.871: Using central status file /mnt/data/rauc.status
(rauc:7771): rauc-DEBUG: 08:55:17.872: Resolved bootname /dev/disk/by-partlabel/root-a to /dev/mmcblk0p4
(rauc:7771): rauc-DEBUG: 08:55:17.872: Obtained system boot ID: 'a69a02de-b0f8-47e9-826e-7be061411eb3'
rauc-INFO: 08:55:17.872: Failed to get machine-id: Failed to open file ?/etc/machine-id?: No such file or directory
(rauc:7771): rauc-DEBUG: 08:55:17.872: Obtained system machine ID: '(null)'
(rauc:7771): rauc-DEBUG: 08:55:17.872: service start
(rauc:7771): rauc-DEBUG: 08:55:17.873: Found booted slot: rootfs.0 on /dev/disk/by-partlabel/root-a

(rauc:7771): rauc-CRITICAL **: 08:55:17.873: r_slot_status_load: assertion 'dest_slot' failed
Trace/breakpoint trap
```

system.conf is:
```
cat /etc/rauc/system.conf
[system]
compatible=gaia-300d-r1
mountprefix=/run/rauc
statusfile=/mnt/data/rauc.status
bootloader=custom
bundle-formats=-plain
max-bundle-download-size=120000000

[keyring]
path=/etc/rauc/keyring.pem
use-bundle-signing-time=true

[encryption]
key=/etc/rauc/bundle-key.pem

[handlers]
bootloader-custom-backend=/usr/sbin/rauc-custom-boot-handler
post-install=/usr/sbin/rauc-post-install

[slot.boot.0]
device=/dev/disk/by-partlabel/boot
type=raw

[slot.rootfs.0]
device=/dev/disk/by-partlabel/root-a
type=raw
bootname=A

[slot.rootfs.1]
device=/dev/disk/by-partlabel/root-b
type=raw
bootname=B

[slot.kernelfs.0]
device=/dev/disk/by-partlabel/kernel-a
type=raw
parent=rootfs.0

[slot.kernelfs.1]
device=/dev/disk/by-partlabel/kernel-b
type=raw
parent=rootfs.1
```

Adding a few printf statements I see that I end up in the `r_context()->system_status` conditional in `service_start()` (as system.statusfile is set) where it tries to match `r_context()->bootslot (/dev/mmcblk0p4)` with the slot device names by `r_slot_find_by_device()`. The problem here is that those refer to the `/dev/disk/by-partlabel/*` symlinks so dumb strcmp with `/dev/mmcblk0p4` naturally fails and we end up calling `r_slot_status_load(NULL)`.

How is this supposed to work and why is trying to redo the work done in determine_slot_states() (which correctly resolved the symlink to /dev/mmcblk0p4, see the log above).

jacmet, 2024-03-06:
I see you tried to fix a similar issue in https://github.com/rauc/rauc/commit/2928ac0603a77c7e817d541f6a872fc81ad59258 but here we do have a valid boot slot

ejoerns, 2024-03-06:
@jacmet Thank you for your analysis. It's quite precise :+1: 

> How is this supposed to work and why is trying to redo the work done in determine_slot_states() (which correctly resolved the symlink to /dev/mmcblk0p4, see the log above).

It would work with adding a resolve call, but you are right, it should really not be necessary to do this work twice here.

I'll post a suggestion that introduces a more appropriate helper to just get the already determined bootslot and drop the occurrences of the old code.

jacmet, 2024-03-06:
@ejoerns I can confirm that the PR fixes the issue here:
```
rauc service -d
rauc-Message: 17:13:42.719: Debug log domains: 'rauc'
rauc-Message: 17:13:42.723: Using central status file /mnt/data/rauc.status
(rauc:535): rauc-DEBUG: 17:13:42.724: Resolved bootname /dev/disk/by-partlabel/root-a to /dev/mmcblk0p4
(rauc:535): rauc-DEBUG: 17:13:42.724: Obtained system boot ID: 'c7e93e10-bd68-4873-920d-3b0fcba0a943'
rauc-INFO: 17:13:42.724: Failed to get machine-id: Failed to open file ?/etc/machine-id?: No such file or directory
(rauc:535): rauc-DEBUG: 17:13:42.725: Obtained system machine ID: '(null)'
(rauc:535): rauc-DEBUG: 17:13:42.725: service start
(rauc:535): rauc-DEBUG: 17:13:42.725: Found booted slot: rootfs.0 on /dev/disk/by-partlabel/root-a
(rauc:535): rauc-DEBUG: 17:13:42.725: Load status for slot rootfs.1.
(rauc:535): rauc-DEBUG: 17:13:42.725: Load status for slot boot.0.
(rauc:535): rauc-DEBUG: 17:13:42.725: Load status for slot rootfs.0.
(rauc:535): rauc-DEBUG: 17:13:42.725: Load status for slot kernelfs.1.
(rauc:535): rauc-DEBUG: 17:13:42.725: Load status for slot kernelfs.0.
(rauc:535): rauc-DEBUG: 17:13:42.744: name 'de.pengutronix.rauc' acquired
```
Thanks!

ejoerns, 2024-03-06:
@jacmet Thanks for testing!

PR #1361 (2024-03-07): Simplify and Fix Bootslot Handling
* Introduces a helper `r_slot_get_booted()` to get the determined boot slot
* Replace indirect re-determination of bootslot
* remove obsolete helper `r_slot_find_by_bootname()` then

Fixes #1360
commit b8a194dc (2024-03-07): src/main: do not search for booted slot again

In 'determine_slot_states()', we already identified the bootslot
including proper path handling (calling 'resolve()' before comparing
config with actual device paths). There is no need to do something
similar in 'service_start()' again.

This fixes both the precedence of 'bootname' vs 'device path' checking
as well as the missing handling of symlinked slot paths that caused RAUC
to die with:

| rauc-CRITICAL **: 08:55:17.873: r_slot_status_load: assertion 'dest_slot' failed

when using symlinks for devices in system.conf.

Fixes #1360

Suggested-b [... cut]
commit 7554c2c3 (2024-03-15): src/main: do not search for booted slot again

In 'determine_slot_states()', we already identified the bootslot
including proper path handling (calling 'resolve()' before comparing
config with actual device paths). There is no need to do something
similar in 'service_start()' again.

This fixes both the precedence of 'bootname' vs 'device path' checking
as well as the missing handling of symlinked slot paths that caused RAUC
to die with:

| rauc-CRITICAL **: 08:55:17.873: r_slot_status_load: assertion 'dest_slot' failed

when using symlinks for devices in system.conf.

Fixes #1360

Suggested-b [... cut]
closed 2024-03-07 by jluebbe