rauc/rauc #1414: test: boot_switch: possible out of bound read and test fail

LeBlue, 2024-05-13:
**Describe the bug**
Test failure `boot_switch`, missing 0-termination in output of `sfdisk` given to `json_from_string`.

**Background information**
rauc master@cd3aa7ca3061896f2065b34826761d740599661a

Running the `qemu-test` regularly fails the boot_switch test.

**To Reproduce**
Steps to reproduce the behavior:
1. build and run `./qemu-test test='boot_switch'`

**Expected behavior**
Test should succeed.

**Logs**
Reduced to the necessary

Note the `MALLOC_PERTURB` value of 14 (== 0x0e) and the additional bytes in fdisk output below.
```
test:         rauc / boot_switch
start time:   09:56:26
duration:     1.73s
result:       killed by signal 6 SIGABRT
command:      MALLOC_PERTURB_=14 ASAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1 RAUC_SERVICE_BINARY=/home/matthiaswauer/Projekte/rauc/build/rauc UBSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 LD_LIBRARY_PATH=/home/matthiaswauer/Projekte/rauc/build/subprojects/json-glib-1.6.6/json-glib /home/matthiaswauer/Projekte/rauc/build/test/boot_switch-test
...
# size: 609
#
[Invalid UTF-8] # content: {
#    "partitiontable": {
#       "label": "dos",
#       "id": "0x8b9e754a",
#       "device": "/dev/loop0",
#       "unit": "sectors",
#       "sectorsize": 512,
#       "partitions": [
#          {
#             "node": "/dev/loop0p1",
#             "start": 8192,
#             "size": 6144,
#             "type": "ef",
#             "bootable": true
#          },{
#             "node": "/dev/loop0p2",
#             "start": 14336,
#             "size": 65536,
#             "type": "83"
#          },{
#             "node": "/dev/loop0p3",
#             "start": 79872,
#             "size": 1,
#             "type": "5"
#          }
#       ]
#    }
# }
# \x0e\x0e\x0e\x0e\x0e\x0e\x0e\xf1\x02
#
# last byte: 0xa
#
not ok /boot_switch/mbr/active-first - rauc:ERROR:../test/boot_switch.c:108:sfdisk_check: assertion failed (error == NULL): JSON data must be UTF-8 encoded (json-parser-error-quark, 6)
Bail out!
----------------------------------- stderr -----------------------------------
**
rauc:ERROR:../test/boot_switch.c:102:sfdisk_check: assertion failed (error == NULL): JSON data must be UTF-8 encoded (json-parser-error-quark, 6)
```

Diff for output printing:
```
diff --git a/test/boot_switch.c b/test/boot_switch.c
index 82cad9ea..67cba044 100644
--- a/test/boot_switch.c
+++ b/test/boot_switch.c
@@ -76,7 +76,11 @@ static gchar *sfdisk_get(const gchar *device)
        g_assert_no_error(error);
        g_assert_true(res);

-       return g_bytes_unref_to_data(stdout_buf, &size);
+       gpointer data = g_bytes_unref_to_data(stdout_buf, &size);
+       g_test_message("size: %d\n", size);
+       g_test_message("content: %s\n", (gchar*)data);
+       g_test_message("last byte: 0x%x\n", ((gchar*)data)[size-1]);
+       return data;
 }

 static void sfdisk_check(const gchar *device, const gchar *expected)
```

**Additional context**

Reading trough the docs: `g_subprocess_communicate` + `g_bytes_unref_to_data` does not produce a 0-terminated string, required by `json_from_string`.

jluebbe, 2024-05-13:
@LeBlue I've not been able to reproduce the out-of-bounds read in by setup, but you analysis is correct. I've created #1415 to fix this. Could you check if that fixes it for you?

LeBlue, 2024-05-13:
@jluebbe I am also supprised, this does not happen in the CI.

> Could you check if that fixes it for you?

I will do that.

LeBlue, 2024-05-13:
Just for reference:

Run-time dependency glib-2.0 found: YES 2.78.3
Run-time dependency gio-2.0 found: YES 2.78.3
Run-time dependency gio-unix-2.0 found: YES 2.78.3

glib2.x86_64 2.78.3-1.fc39

LeBlue, 2024-05-13:
@jluebbe Might be worth reviewing `src/bootchooser.c` that also uses `g_subprocess_communicate` to read stdout.

ejoerns, 2024-05-13:
@LeBlue Indeed, we came to that conclusion, too. Thanks for the hint! :+1:

LeBlue, 2024-05-13:
@jluebbe @I think something eles is wrong here. `MALLOC_PERTURB_` sets the value to the bit inverse on init

PR #1415 (2024-05-13): test/boot_switch: fix out of bound read
g_bytes_unref_to_data doesn't provide a null-terminated string, so json_from_string might read beyond the end of the subprocess output. Instead of adding a null-byte ourselves, switch to g_subprocess_communicate_utf8.

This is a better fit, as json_from_string expects a valid UTF-8 string anyway.

Fixes: #1414
commit b2267325 (2024-05-13): test/boot_switch: fix out of bound read

g_bytes_unref_to_data doesn't provide a null-terminated string, so
json_from_string might read beyond the end of the subprocess output.
Instead of adding a null-byte ourselves, switch to
g_subprocess_communicate_utf8.

This is a better fit, as json_from_string expects a valid UTF-8 string
anyway.

Fixes: #1414
closed 2024-05-13 by ejoerns