rauc/rauc #1416: migrate pytest tests from pydbus to dasbus

jluebbe, 2024-05-13:
The pytest test suite currently uses [pydbus](https://github.com/LEW21/pydbus) to [access the RAUC service](https://github.com/rauc/rauc/blob/master/test/conftest.py#L10). As pydbus is no longer actively maintained, it should be replaces by something else. [dasbus](https://github.com/dasbus-project/dasbus) seems a likely choice.

- [ ] add `python3-dasbus` to https://github.com/rauc/rauc/blob/master/test/Dockerfile
- [ ] update `conftest.py`

jluebbe, 2025-02-25:
It seems that dasbus isn't very active either, so perhaps https://github.com/python-sdbus/python-sdbus would be a good alternative in the longer term, but it needs explicit interface classes.

BubuOT, 2025-02-25:
You can autogenerate the interface classes from the rauc dbus xml. I'm accessing rauc's dbus interface with sdbus in our project. :)

jluebbe, 2025-02-25:
> You can autogenerate the interface classes from the rauc dbus xml. I'm accessing rauc's dbus interface with sdbus in our project. :)

Yes, I sway that, but I'm currently focused on a different part in RAUC and don't want to get sidetracked too far. If your code is public, could you link to it?

BubuOT, 2025-02-25:
Not public unfortunately, but I can  share the rauc parts here:

`python -m sdbus gen-from-file --block de.pengutronix.rauc.Installer.xml` generates the interface code (or without `--block` for the async variant, which you need for signal handling if I remember correctly)

You use it like this then:
```
class RaucDbusClient:
    def __init__(self) -> None:
        self._proxy = DePengutronixRaucInstallerInterface("de.pengutronix.rauc", "/")

    def get_rauc_operation(self) -> str:
        """Get current rauc operation via dbus"""
        return self._proxy.operation

    def get_rauc_boot_slot(self) -> str:
        """Get current boot slot (A or B)"""
        return self._proxy.boot_slot
```

Or the async variant:
```

class RaucDbusClientAsync:
    def __init__(self) -> None:
        self._proxy = DePengutronixRaucInstallerInterfaceAsync.new_proxy(
            "de.pengutronix.rauc", "/"
        )

    async def get_rauc_operation(self) -> str:
        """Get current rauc operation via dbus"""
        return await self._proxy.operation

    async def get_rauc_boot_slot(self) -> str:
        """Get current boot slot (A or B)"""
        return await self._proxy.boot_slot

    async def handle_rauc_dbus_messages(
        self,
        callback: Callable[
            [Self, tuple[str, dict[str, tuple[str, Any]], list[str]]], Awaitable[None]
        ],
    ) -> None:
        async for change in self._proxy.properties_changed:
            await callback(self, change)

    async def handle_rauc_completed_signal(
        self, callback: Callable[[Self, int], Awaitable[None]]
    ) -> None:
        async for return_code in self._proxy.completed:
            await callback(self, return_code)
```

PR #1641 (2025-02-28): test: migrate from pydbus to dasbus
Fixes: #1416
closed 2025-02-28 by ejoerns