rauc/rauc #1431: check_bundle_access() failing on BTRFS

LeBlue, 2024-05-23:
**Describe the bug**
Running e.g. `rauc info` (and others, e.g. `resign`) failes for `plain` bundles with `unable to find mounted device for bundle`, if the bundle file is on a BTRFS volume (with subvolumes) and not on the `/` filesystem.

**Background information**
Tested with v1.11.3, and master@a5d51e6d9985ed41e2ca76cdd15cea4ce027e265

The root and /home mountpoints are BTRFS subvolumes (on fedora 38).

Bundle format is `plain`. 

**To Reproduce**
```
rauc info --keyring keyring.pem bundle.raucb
...
rauc-Message: 13:24:26.973: Verifying bundle signature... 
unable to find mounted device for bundle
```

**Expected behavior**
The commands should either work or maybe give a helpful error message.

Another solution would be to provide `--trust-environment` flag similar to other commands.

**Logs**

```
/dev/mapper/luks-442a6c20-26a9-4e46-af37-b023ea33dce7 on /home type btrfs (rw,relatime,seclabel,compress=zstd:1,ssd,discard=async,space_cache=v2,subvolid=256,subvol=/home)
```

gdb output (of the device that has the `/home` subvolume):
```
(gdb) p dev_path
$2 = (const gchar *) 0x4c1a30 "/dev/mapper/luks-442a6c20-26a9-4e46-af37-b023ea33dce7"
(gdb) p/x dev_stat
$3 = {
  st_dev = 0x5,
  st_ino = 0x25e,
  st_nlink = 0x1,
  st_mode = 0x61b0,
  st_uid = 0x0,
  st_gid = 0x6,
  __pad0 = 0x0,
  st_rdev = 0xfd00,
...
}
(gdb) p/x bundle_stat
$4 = {
  st_dev = 0x2a,
  st_ino = 0x49d223e,
  st_nlink = 0x1,
  st_mode = 0x81a4,
  st_uid = 0x3e8,
  st_gid = 0x3e8,
  __pad0 = 0x0,
  st_rdev = 0x0,
...
}
```

**Additional context**
Moving the bundle to a different filesystem e.g. `/tmp` is a possible workaround.

Moving the bundle to the / subvolume (e.g. `/opt/`) also works.

The issue only affects the check that tries to match the bundle file to the device. The action/rauc command itself works (I disabled the check for testing).

jluebbe, 2024-05-23:
Can't you use verity format bundles, which don't need the access check mitigation?

LeBlue, 2024-05-23:
> Can't you use verity format bundles, which don't need the access check mitigation?

For newer developments we start using `verity` but need to use `plain` still for devices in the field (some of these for some time in the future).

I am aware that it is a low priority or won't-fix issue (in that case it could be closed, stating that it won't be fixed).

I just wanted to make anybody with the same problem aware that this issue exists (took me a bit of time to figure it out). As described, there are enough workarounds available to get the work done (mostly checking signatures and resigning).

jluebbe, 2024-05-23:
Are you encountering this issue on the target device during installation or on your development host?

LeBlue, 2024-05-27:
> Are you encountering this issue on the target device during installation or on your development host?

Only on the development host.