rauc/rauc #1441: Does the install-same slot config attribute work?

andersm, 2024-06-04:
**Describe the bug**
RAUC will re-write the same update to a slot, even though the system.conf file has "install-same=false" for that slot. This happens both on bootloader and ext4 rootfs slots.

**Background information**
RAUC v1.11.3, with a Yocto Kirkstone build.

**To Reproduce**
Steps to reproduce the behavior:
1. Set "install-same=false" for a slot in /etc/rauc/system.conf
2. Install an update
3. Restart
4. Verify that the hash in RAUC's manifest and the bundle manifest match
5. Install the same update

**Expected behavior**
RAUC should skip the slot.

**Logs**
If applicable, add console output or logs from the RAUC service to help explain your problem.

**Additional context**
It seems that inside `handle_slot_install_plan()`, `slot_state->checksum.digest` is always a NULL pointer, causing the skip check to fail. https://github.com/rauc/rauc/blob/master/src/install.c#L984
The test in the test suite for this config option do not seem to verify that the second install operation is actually skipped.

ejoerns, 2024-06-04:
@andersm Thank you for reporting!

This should be fixed in `master` by https://github.com/rauc/rauc/pull/1392.

Could you verify this?

Please also indicate if this is an important functionality for you or if waiting for the next regular release that includes the fixes would be sufficient.

andersm, 2024-06-05:
@ejoerns Thanks for the reply.

#1392 fixes the issue for non-redundant slots, but redundant slots seem broken now. The issue seems to be that RAUC checks the hash of the inactive slot, rather than the active slot.

So:
rootfs.0, booted, SHA256 = "123"
rootfs.1, inactive, SHA256 = "456"
bundle SHA256 = "123"
-> update performed, but should be skipped

rootfs.0, booted, SHA256 = "456"
rootfs.1, inactive, SHA256 = "123"
bundle SHA256 = "123"
-> update skipped, but should be performed

ejoerns, 2024-07-19:
@andersm I am not sure if I understand correctly what the issue is here.

Slot skipping only means that RAUC will skip updating the slot it would write to. Taking Scenario 1:

> rootfs.0, booted, SHA256 = "123"
> rootfs.1, inactive, SHA256 = "456"
> bundle SHA256 = "123"

Here the update would write to *rootfs.1*. Thus RAUC checks if the bundle hash to write is the one that is already present in the target slot. Here "456" is present while "123" should be installed. Thus performing an update is the intended behavior.

Note that RAUC will not skip the update itself if an active slot matches the current bundle One reason is that slot skipping works on a per-slot base and there might be multiple slots per slot group from which others cannot be skipped.

andersm, 2024-07-19:
@ejoerns That is not made clear in the documentation, but it also renders the whole feature kind of pointless, at least for a basic A-B setup.

ejoerns, 2024-07-19:
@andersm Slot skipping is a feature from the earlier days of RAUC, but there are still some potential use cases.

You'll benefit from it if you have a setup with multiple slots per slot group and one slot is updated more frequently than the other.

Another optimization we had in planning was to just sync content from the active to the inactive partition in case the active one is found identical to what's about to be installed.

See https://github.com/rauc/rauc/issues/376 for more information.

Nowadays, we have adaptive updates and enhancing the update based on these is the more generic and more efficient approach, especially for actual OTA updates.