rauc/rauc #1444: Unintended value in RAUC_SLOT_MOUNT_POINT when updating raw slot.

tunhguod, 2024-06-13:
**Describe the bug**.
Unintended value in RAUC_SLOT_MOUNT_POINT when updating raw slot

**Background information**
version 1.7

**To Reproduce**

Steps to reproduce the behavior: 

1. Configure a simple system with a vfat formatted boot partition and ext4 formatted rootfs. 
    * Define all slot types as `type=raw` in system.conf

2. Create a rauc bundle to update the boot and rootfs partitions with the rootfs slot as parent.
    * Run post-install to hook the script when updating boot partition.

3. run rauc install using rauc bundle

4. ls RAUC_SLOT_MOUNT_POINT on hook script

**Expected behavior**.

RAUC_SLOT_MOUNT_POINT is set to the mount point of boot partition, or nothing is set (because it is treated as a raw image).

**Actual behavior**

RAUC_SLOT_MOUNT_POINT is set to the mount point of rootfs partition.

**Supplement**

Looking at the following, it appears that the mount point of the ext partition is set as MOUNT_POINT in RAUC when updating the raw image

https://github.com/rauc/rauc/blob/16a1e5588eeb64d1bcc9c01b8020abc85e5311ab/src/update_handler.c#L1328-L1332

* slots->mount_point is not set when updating raw image?
* Is this why I see rootfs when I ls RAUC_SLOT_MOUNT_POINT?

Is this intentional? I've been reading through the documentation and couldn't find anything related to this...

ejoerns, 2024-07-16:
> * slots->mount_point is not set when updating raw image?

It is not set for a raw *slot* since these don't imply mountable file systems and thus they are not mounted.

> * Is this why I see rootfs when I exec `$(ls ${RAUC_SLOT_MOUNT_POINT})`?

You should only see the correct mount point or the external mount point.

> * What is an external mount point?

It is a mount detected by RAUC. I.e. a partition that is mounted on your system (and not by RAUC).

Please check with `mount` that the slot you see this for does not appear in the mounts.

Run `rauc status` and check for a `mounted: ` line to see if RAUC detected a mount point for your slot.

ejoerns, 2024-09-10:
@tunhguod Any updates on this? Otherwise, I'd like to close this for now since it is unclear to me if there is actually a bug or not.

tunhguod, 2024-09-16:
@ejoerns 
Thank you for your response, I understand most of it.
Please close the issue, especially since there is no update.

closed 2024-09-16 by ejoerns