rauc/rauc #1452: RAUC not reporting failed result on DBus when removing USB during update

Cyres300, 2024-06-26:
**Description**
When installing a update bundle from a USB, I removed the USB during the update.
At this point the RAUC service crashed and did not signal Completed (with a failed result) via DBus.

**Background information**
RAUC version: rauc 1.10.1
Build system: yocto for arm64, kirkstone

**Reproduce**
1. Install a rauc bundle located on a USB via the command line.
2. Immediately remove the USB, by pulling it out.

**Expected behavior**
Expected a DBus signal "Completed(something != 0)".

**Logs**
*Console output:*
```
root@pollux-carrier:~#` rauc install /mnt/application-debug-bundle.raucb
installing
  0% Installing
  0% Determining slot states
 10% Determining slot states done.
 10% Checking bundle
 10% Verifying signature
 20% Verifying signature done.
 20% Checking bundle done.
 20% Checking manifest contents
 30% Checking manifest contents done.
 30% Determining target install group
 40% Determining target install group done.
 40% Updating slots
 40% Checking slot boot.0
 43% Checking slot boot.0 done.
 43% Copying image to boot.0
 70% Copying image to boot.0 done.
 70% Checking slot rootfs.0
 73% Checking slot rootfs.0 done.
 73% Copying image to rootfs.0
RAUC service disappeared
D-Bus error while installing `/mnt/application-debug-bundle.raucb`
```

*Systemctl status:*
```
root@pollux-carrier:~# systemctl status rauc
× rauc.service - RAUC Update Service
     Loaded: loaded (/lib/systemd/system/rauc.service; static)
     Active: failed (Result: signal) since Tue 2024-06-25 13:16:41 UTC; 23s ago
       Docs: https://rauc.readthedocs.io
    Process: 1910 ExecStart=/usr/bin/rauc --mount=/run/rauc service (code=killed, signal=BUS)
   Main PID: 1910 (code=killed, signal=BUS)

Jun 25 13:16:40 pollux-carrier rauc[1910]: access stats for target_slot: count=6094 sum=6094.000 min=1…g=1.000
Jun 25 13:16:40 pollux-carrier rauc[1910]: access stats for active_slot: count=0
Jun 25 13:16:40 pollux-carrier rauc[1910]: access stats for source_image: count=0
Jun 25 13:16:40 pollux-carrier rauc[1910]: installing /mnt/application-debug-bundle.raucb: Updating slot boot… status
Jun 25 13:16:40 pollux-carrier rauc[1910]: installing /mnt/application-debug-bundle.raucb: Updating slot boot.0 done
Jun 25 13:16:40 pollux-carrier rauc[1910]: installing /mnt/application-debug-bundle.raucb: Checking slot rootfs.0
Jun 25 13:16:40 pollux-carrier rauc[1910]: Updating /dev/mmcblk2p3 with /run/rauc/bundle/application-rootfs-p…er.ext4
Jun 25 13:16:40 pollux-carrier rauc[1910]: installing /mnt/application-debug-bundle.raucb: Updating slot rootfs.0
Jun 25 13:16:41 pollux-carrier systemd[1]: rauc.service: Main process exited, code=killed, status=7/BUS
Jun 25 13:16:41 pollux-carrier systemd[1]: rauc.service: Failed with result 'signal'.
```

*Journal part:*
```
Jun 25 13:16:39 pollux-carrier rauc[1910]: installing /mnt/application-debug-bundle.raucb: Updating slot boot.0
Jun 25 13:16:40 pollux-carrier rauc[1910]: access stats for zero chunk: count=26637 sum=26637.000 min=1.000 max=1.000 avg=1.000 recent-avg=1.000
Jun 25 13:16:40 pollux-carrier rauc[1910]: access stats for target_slot_written (reusing source_image): count=6131 sum=37.000 min=0.000 max=1.000 avg=0.006 recent-avg=0.000
Jun 25 13:16:40 pollux-carrier rauc[1910]: access stats for target_slot: count=6094 sum=6094.000 min=1.000 max=1.000 avg=1.000 recent-avg=1.000
Jun 25 13:16:40 pollux-carrier rauc[1910]: access stats for active_slot: count=0
Jun 25 13:16:40 pollux-carrier rauc[1910]: access stats for source_image: count=0
Jun 25 13:16:40 pollux-carrier rauc[1910]: installing /mnt/application-debug-bundle.raucb: Updating slot boot.0 status
Jun 25 13:16:40 pollux-carrier rauc[1910]: installing /mnt/application-debug-bundle.raucb: Updating slot boot.0 done
Jun 25 13:16:40 pollux-carrier rauc[1910]: installing /mnt/application-debug-bundle.raucb: Checking slot rootfs.0
Jun 25 13:16:40 pollux-carrier rauc[1910]: Updating /dev/mmcblk2p3 with /run/rauc/bundle/application-rootfs-pollux-carrier.ext4
Jun 25 13:16:40 pollux-carrier rauc[1910]: installing /mnt/application-debug-bundle.raucb: Updating slot rootfs.0
Jun 25 13:16:41 pollux-carrier systemd-coredump[2154]: Process 1910 (rauc) of user 0 dumped core.
Jun 25 13:16:41 pollux-carrier systemd[1]: rauc.service: Main process exited, code=killed, status=7/BUS
Jun 25 13:16:41 pollux-carrier systemd[1]: rauc.service: Failed with result 'signal'.
```

**Additional context**
We have a GUI and we allow updating via an USB. This GUI shows the update progress. 
If sucessful the system will reboot. If not than the user needs feedback that it has failed and should try again.

I'm implementing if someone decides to remove the USB during an update. 

I also saw the expected behaviour when waiting until rauc reports 73% and than pulling out the USB. 
The expected Completed(1) signal was there.

Both situation are reproducable. The correct one waiting until 73% and the one where immediately the USB was removed.

***Possible work around***
During updating check at an interval if the rauc service is still active. If not active during update the update has failed.

ejoerns, 2024-06-26:
@Cyres300 Thanks for reporting! From the log it is quite hard to see where the service actually crashed.

Could you give it another try and start RAUC with subprocess and debug output enabled like described in https://rauc.readthedocs.io/en/latest/using.html#increasing-debug-verbosity ?

Or maybe you can extract some information from the coredump using `coredumpctl` (if you have access to debugging symbols somehow)?

Cyres300, 2024-06-26:
Hi @ejoerns, Thanks for your quick responce!

I have tried your suggestions. Hope this will help.

*coredumpctl with gdb*
```
Reading symbols from /usr/bin/rauc...
(No debugging symbols found in /usr/bin/rauc)

warning: Can't open file /run/rauc/bundle/application-rootfs-pollux-carrier.ext4.block-hash-index during file-backed mapping note processing
[New LWP 1471]
[New LWP 1354]
[New LWP 1351]
[New LWP 1352]
[New LWP 1478]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/libthread_db.so.1".
Core was generated by `/usr/bin/rauc --mount=/run/rauc service'.
Program terminated with signal SIGBUS, Bus error.
#0  memcmp () at ../sysdeps/aarch64/memcmp.S:53
53      ../sysdeps/aarch64/memcmp.S: No such file or directory.
[Current thread is 1 (Thread 0xffffacfef0c0 (LWP 1471))]
```

*Journalctl -b -f*
Interesting part is at the end I guess. But for a complete view added the logs from start to end.
```
Jun 26 12:21:33 pollux-carrier rauc[1351]: input bundle: /mnt/application-debug-bundle.raucb
Jun 26 12:21:33 pollux-carrier rauc[1351]: Active slot bootname: system1
Jun 26 12:21:33 pollux-carrier rauc[1351]: thread started for /mnt/application-debug-bundle.raucb
Jun 26 12:21:33 pollux-carrier rauc[1351]: installing /mnt/application-debug-bundle.raucb: started
Jun 26 12:21:33 pollux-carrier rauc[1351]: Installation 5fd2219b-462e-4314-adfe-b9e1af04bfd6 started
Jun 26 12:21:33 pollux-carrier rauc[1351]: Found external mountpoint for slot rootfs.1 at /
Jun 26 12:21:33 pollux-carrier rauc[1351]: installing /mnt/application-debug-bundle.raucb: Checking and mounting bundle...
Jun 26 12:21:33 pollux-carrier rauc[1351]: Reading bundle: /mnt/application-debug-bundle.raucb
Jun 26 12:21:33 pollux-carrier rauc[1351]: Found valid CMS data
Jun 26 12:21:33 pollux-carrier rauc[1351]: Verifying bundle signature...
Jun 26 12:21:33 pollux-carrier rauc[1351]: Verified inline signature by 'O = aXcent medical GmbH, CN = aXcent medical GmbH application-Rauc'
Jun 26 12:21:33 pollux-carrier rauc[1351]: Got 2 elements for trust chain
Jun 26 12:21:33 pollux-carrier rauc[1351]: Mounting bundle '/mnt/application-debug-bundle.raucb' to '/run/rauc/bundle'
Jun 26 12:21:33 pollux-carrier kernel: loop0: detected capacity change from 0 to 365356
Jun 26 12:21:33 pollux-carrier kernel: loop0: detected capacity change from 365356 to 365352
Jun 26 12:21:33 pollux-carrier rauc[1351]: Configured loop device '/dev/loop0' for 187060224 bytes
Jun 26 12:21:33 pollux-carrier kernel: device-mapper: verity: sha256 using implementation "sha256-ce"
Jun 26 12:21:33 pollux-carrier rauc[1351]: Configured dm-verity device '/dev/dm-0'
Jun 26 12:21:33 pollux-carrier rauc[1351]: Checking slot: rootfs.1
Jun 26 12:21:33 pollux-carrier rauc[1351]:         No mapping found
Jun 26 12:21:33 pollux-carrier rauc[1351]: Checking slot: boot.0
Jun 26 12:21:33 pollux-carrier rauc[1351]:         Adding mapping: boot -> boot.0
Jun 26 12:21:33 pollux-carrier rauc[1351]: Checking slot: bootloader.0
Jun 26 12:21:33 pollux-carrier rauc[1351]:         Adding mapping: bootloader -> bootloader.0
Jun 26 12:21:33 pollux-carrier rauc[1351]: Checking slot: rootfs.0
Jun 26 12:21:33 pollux-carrier rauc[1351]:         Adding mapping: rootfs -> rootfs.0
Jun 26 12:21:33 pollux-carrier rauc[1351]: Checking slot: boot.1
Jun 26 12:21:33 pollux-carrier rauc[1351]:         No mapping found
Jun 26 12:21:33 pollux-carrier rauc[1351]: Using default installation handler
Jun 26 12:21:33 pollux-carrier rauc[1351]: Found image mapping: application-boot.vfat -> boot
Jun 26 12:21:33 pollux-carrier rauc[1351]: Checking image type for slot type: vfat
Jun 26 12:21:33 pollux-carrier rauc[1351]: Image detected as type: *.vfat
Jun 26 12:21:33 pollux-carrier rauc[1351]: Found image mapping: application-rootfs-pollux-carrier.ext4 -> rootfs
Jun 26 12:21:33 pollux-carrier rauc[1351]: Checking image type for slot type: ext4
Jun 26 12:21:33 pollux-carrier rauc[1351]: Image detected as type: *.ext4
Jun 26 12:21:33 pollux-carrier rauc[1351]: Marking target slot rootfs.0 as non-bootable...
Jun 26 12:21:33 pollux-carrier rauc[1351]: launching subprocess: fw_printenv BOOT_ORDER
Jun 26 12:21:33 pollux-carrier rauc[1351]: launching subprocess: fw_setenv BOOT_ORDER system1
Jun 26 12:21:33 pollux-carrier rauc[1351]: launching subprocess: fw_setenv BOOT_system0_LEFT 0
Jun 26 12:21:33 pollux-carrier rauc[1351]: installing /mnt/application-debug-bundle.raucb: Updating slots...
Jun 26 12:21:33 pollux-carrier rauc[1351]: installing /mnt/application-debug-bundle.raucb: Checking slot boot.0
Jun 26 12:21:33 pollux-carrier rauc[1351]: Updating /dev/mmcblk2p1 with /run/rauc/bundle/application-boot.vfat
Jun 26 12:21:33 pollux-carrier rauc[1351]: installing /mnt/application-debug-bundle.raucb: Updating slot boot.0
[... 25 lines cut ...]
Jun 26 12:21:34 pollux-carrier kernel: blk_update_request: I/O error, dev sda, sector 2049 op 0x1:(WRITE) flags 0x100000 phys_seg 1 prio class 0
Jun 26 12:21:34 pollux-carrier kernel: Buffer I/O error on dev sda1, logical block 1, lost async page write
Jun 26 12:21:34 pollux-carrier kernel: blk_update_request: I/O error, dev sda, sector 34816 op 0x1:(WRITE) flags 0x100000 phys_seg 1 prio class 0
Jun 26 12:21:34 pollux-carrier kernel: Buffer I/O error on dev sda1, logical block 32768, lost async page write
[... cut ...]
Jun 26 12:21:36 pollux-carrier systemd-coredump[1482]: elfutils disabled, parsing ELF objects not supported
Jun 26 12:21:36 pollux-carrier systemd-coredump[1482]: [LNK] Process 1351 (rauc) of user 0 dumped core.
Jun 26 12:21:36 pollux-carrier systemd[1]: systemd-coredump@1-1481-0.service: Deactivated successfully.
Jun 26 12:21:36 pollux-carrier systemd[1]: rauc.service: Main process exited, code=killed, status=7/BUS
Jun 26 12:21:36 pollux-carrier systemd[1]: rauc.service: Failed with result 'signal'.
```

ejoerns, 2024-07-09:
@Cyres300 Thank you for sharing details. Unfortunately, no debugging symbols are available; thus, it is hard to guess where it crashes. If you could install debug symbols or use remote debugging techniques[1], it might be easier to trace this down.

I have attempted to reproduce it and found #1454, but I am unsure if this is the crash you see since I would have expected to see an assertion error in that case.

@Cyres300 Maybe you could check if the issue remains with latest `master` state?

[1] https://docs.yoctoproject.org/dev-manual/debugging.html#debugging-with-the-gnu-project-debugger-gdb-remotely

ejoerns, 2024-09-10:
@Cyres300 Did you already have the time to look into this again?

Jo3Zhou, 2024-09-19:
Hi @ejoerns,

Not sure if I got this correct:

In `gboolean do_install_bundle(RaucInstallArgs *args, GError **error)`, `gboolean update_external_mount_points(GError **error)` only checked once before install.

I suppose it could use `GUnixMountMonitor* g_unix_mount_monitor_get (void)` connect signal to get notified. Then cancel the install process and return the error code?

And regarding to #1454 , here is the description of  `g_assert`:

> Debugging macro to terminate the application if the assertion fails. If the assertion fails (i.e. the expression is not true), an error message is logged and the application is terminated.
> 
> The macro can be turned off in final releases of code by defining G_DISABLE_ASSERT when compiling the application, so code must not depend on any side effects from expr. Similarly, it must not be used in unit tests, otherwise the unit tests will be ineffective if compiled with G_DISABLE_ASSERT. Use g_assert_true() and related macros in unit tests instead.
> 
> This function is not directly available to language bindings.

I didn't see any `G_DISABLE_ASSERT` in source files or meson build files. I think the way you did is correct, which is return  the res from `g_output_stream_write_all`.

Cyres300, 2024-09-19:
> @Cyres300 Did you already have the time to look into this again?

No, I did not. I have implemented the work around. That works.

We have moved on to an other project. I will create a ticket in our work environment for a possible improvement. I will link this ticket. If someone picks this up, they will post the result here.

ejoerns, 2024-11-13:
> In `gboolean do_install_bundle(RaucInstallArgs *args, GError **error)`, `gboolean update_external_mount_points(GError **error)` only checked once before install.
>
> I suppose it could use [...]

Since there could be various reasons for the underlying bundle storage/source to disappear suddenly, I assume that a general robust error handling is preferable and sufficient here.
So far, the mount handling is only used for testing if slot devices are mounted (and not the bundle storage itself).

> And regarding to #1454 , here is the description of `g_assert`:
> 
> > Debugging macro to terminate the application if the assertion fails. If the assertion fails (i.e. the expression is not true), an error message is logged and the application is terminated.
> > The macro can be turned off in final releases of code by defining G_DISABLE_ASSERT when compiling the application, so code must not depend on any side effects from expr. Similarly, it must not be used in unit tests, otherwise the unit tests will be ineffective if compiled with G_DISABLE_ASSERT. Use g_assert_true() and related macros in unit tests instead.
> > This function is not directly available to language bindings.
> 
> I didn't see any `G_DISABLE_ASSERT` in source files or meson build files. I think the way you did is correct, which returns the res from `g_output_stream_write_all`.

@Jo3Zhou  Thank you for checking. Indeed, we don't use `G_DISABLE_ASSERT` currently, as the asserts are intended to catch invalid behavior that should never occur and could be critical for the update process if violated.

> No, I did not. I have implemented the work around. That works.

> We have moved on to an other project. I will create a ticket in our work environment for a possible improvement. I will link this ticket. If someone picks this up, they will post the result here.

@Cyres300 Thank you for the update! I will close this ticket now, assuming that the issue has been resolved or that someone from your original project will step in if any further problems arise.

closed 2024-11-13 by ejoerns