rauc/rauc #1461: GLib assertion fails when converting `tar.gz` slots using Casync

matteocarnelos, 2024-07-15:
**Describe the bug**
When converting a conventional RAUC bundle to a Casync bundle using `rauc convert`, the conversion of `tar.gz` slots fail with the following error:

```
(rauc:782068): GLib-CRITICAL **: 19:29:58.598: g_atomic_ref_count_dec: assertion 'old_value > 0' failed
```

With other slot types, such as `.ext4`, this error is not encountered.

**Background information**
RAUC built from source at commit cf4d63a46c6b0a7a5cba3b039d146bbd6fd59665 using Meson.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a conventional RAUC bundle containing a `tar.gz` slot
2. Convert it to a Casync bundle using `rauc convert`

**Expected behavior**
The conversion should succeed.

**Logs**
Full console log with debug enabled:

```
$ rauc -d convert --cert=certfile.cert.pem --key=keyfile.key.pem --keyring=keyring.crt.pem conventional-bundle.raucb casync-bundle.raucb
rauc-Message: 19:38:53.649: Debug log domains: 'rauc'
(rauc:782129): rauc-DEBUG: 19:38:53.649: v1.11.1-267-gcf4d63a4 create=1 emmc-boot=1 gpt=0 json=1 network=1 service=1 streaming=1
(rauc:782129): rauc-DEBUG: 19:38:53.652: convert start
(rauc:782129): rauc-DEBUG: 19:38:53.652: input bundle: conventional-bundle.raucb
(rauc:782129): rauc-DEBUG: 19:38:53.652: output bundle: casync-bundle.raucb
rauc-Message: 19:38:53.652: Reading bundle: conventional-bundle.raucb
(rauc:782129): rauc-DEBUG: 19:38:53.657: Found valid CMS data
rauc-Message: 19:38:53.659: Verifying bundle signature...
rauc-Message: 19:38:53.664: Verified inline signature by 'O = SW, CN = Development'
(rauc:782129): rauc-DEBUG: 19:38:53.666: Got 3 elements for trust chain
rauc-Message: 19:38:53.666: Verifying bundle payload...
(rauc:782129): rauc-DEBUG: 19:38:53.666: Hash verification SHA256, data blocks 25662.
(rauc:782129): rauc-DEBUG: 19:38:53.666: Using 3 hash levels.
(rauc:782129): rauc-DEBUG: 19:38:53.666: Data size: 105111552 bytes.
(rauc:782129): rauc-DEBUG: 19:38:53.667: Hashed size: 105947136 bytes.
(rauc:782129): rauc-DEBUG: 19:38:53.917: Verification of data area succeeded.
(rauc:782129): rauc-DEBUG: 19:38:53.917: Verification of root hash succeeded.

rauc-Message: 19:38:54.364: Converting rootfs.ext4 to casync blob idx rootfs.ext4.caibx
rauc-Message: 19:38:58.310: Converting archive.tar.gz to casync directory tree idx archive.tar.gz.caidx

(rauc:782129): GLib-CRITICAL **: 19:38:58.364: g_atomic_ref_count_dec: assertion 'old_value > 0' failed
Trace/breakpoint trap
```

ejoerns, 2024-07-15:
@matteocarnelos  Thank you for reporting this! I have been able to reproduce it with GDB attached and identified a sort of double-free to a GPtrArray in `casync_make_arch()`.

I have fixed this with a few other issues related to memory allocation.

@matteocarnelos Do you evaluate casync support for a new project or do you already use it? Did you have a look at RAUC's newer built-in [HTTP(S) bundle streaming](https://rauc.readthedocs.io/en/latest/advanced.html#http-streaming) with [adaptive updates](https://rauc.readthedocs.io/en/latest/advanced.html#adaptive-updates)?

matteocarnelos, 2024-07-16:
@ejoerns many thanks for the quick response and fix!

I tried both adaptive updates and casync but I wanted to proceed with casync because of the support to chunk archives and because it showed less data consumption during my tests. Moreover, I encountered a possible bug when trying adaptive updates with `resize=true` set. I still don't know if it's an issue on my end or if it's an actual bug, I'll follow-up with another issue in case.

PR #1462 (2024-07-18): Memory handling fixes for casync archive conversion
Fixes memory leakages, over-allocations, and a sort of double-free crash for 'rauc convert' calls with archives as input images.

Fixes: #1461
commit d841abdd (2024-07-18): src/bundle: fix freeing of GPtrArray in casync_make_arch()

This fixes both a memory leakage and a prevented double-free.

The 'iargs' pointer array is allocated using an auto pointer and a free
function. This works quite well until manually calling
g_ptr_array_free(). This correctly frees the pointer array structure and
leaves the memory pointed at untouched.

BUT, since 'iargs' itself is not NULL after this, the auto pointer
cleanup will later attempt to free the GPtrArray a second time and
causes an assertion error:

| g_atomic_ref_count_dec: assertion 'old_value > 0' failed

This can be fi [... cut]
closed 2024-07-18 by jluebbe