rauc/rauc #1470: Log message MESSAGE_ID_INSTALLATION_SUCCEEDED is missing the PRIORITY field in systemd journal

AndreasUfert, 2024-07-19:
**Describe the bug**

While the other log messages in install.c are logged with their respective priority, MESSAGE_ID_INSTALLATION_SUCCEEDED is missing the actual PRIORITY field in the journal.

It's treated as being "info", though, so noone will barely notice it when browsing the journal but it surfaces in log management systems.

**Background information**

version 1.11.3, build by bitbake/Yocto
https://github.com/rauc/rauc/blob/v1.11.3/src/install.c#L1149

**To Reproduce**

1. install a bundle via rauc
2. journalctl -u rauc (here, it's not distinguishable from log messages with PRIORITY info
3. journalctl -u rauc -o verbose (here, this particular message is missing the PRIORITY field)

**Expected behavior**

Log message MESSAGE_ID_INSTALLATION_SUCCEEDED should have a PRITORITY field attached, either info (6) or notice (5).

**Logs**
```
Thu 2024-07-18 21:39:51.708571 CEST [s=a8d01e952a104323a49a60f3115ba123;i=1156;b=358b8dd314394f60a6e642e84c647137;m=19a3c032b;t=61d8abe1082f9;x=739432f9a5787195]
    PRIORITY=5
    _BOOT_ID=358b8dd314394f60a6e642e84c647137
    _MACHINE_ID=2063e02c3f724434bcd7e1717d0aff62
    _RUNTIME_SCOPE=system
    _UID=0
    _GID=0
    _SYSTEMD_SLICE=system.slice
    _TRANSPORT=journal
    _CAP_EFFECTIVE=1ffffffffff
    _PID=663
    _COMM=rauc
    _EXE=/usr/bin/rauc
    _CMDLINE=/usr/bin/rauc --mount=/run/rauc service
    _SYSTEMD_CGROUP=/system.slice/rauc.service
    _SYSTEMD_UNIT=rauc.service
    _SYSTEMD_INVOCATION_ID=7c28039f337d41b683f34bc4c8f16417
    GLIB_DOMAIN=rauc-event
    RAUC_EVENT_TYPE=mark
    SLOT_NAME=platform.1
    SLOT_BOOTNAME=platform1
    _HOSTNAME=<redacted>
    MESSAGE=Marked slot platform.1 as active
    MESSAGE_ID=8b5e7435e1054d86858278e7544fe6da
    BUNDLE_HASH=5fd0b95f57310edfb481895d3134fb4965b5243d69f72ae50ecfa34ed25e0d10
    _SOURCE_REALTIME_TIMESTAMP=1721331591708571
Thu 2024-07-18 21:39:51.726253 CEST [s=a8d01e952a104323a49a60f3115ba123;i=1157;b=358b8dd314394f60a6e642e84c647137;m=19a3c45b9;t=61d8abe10c586;x=e03b2969a012b8fa]
    _BOOT_ID=358b8dd314394f60a6e642e84c647137
    _MACHINE_ID=2063e02c3f724434bcd7e1717d0aff62
    _RUNTIME_SCOPE=system
    _UID=0
    _GID=0
    _SYSTEMD_SLICE=system.slice
    _TRANSPORT=journal
    _CAP_EFFECTIVE=1ffffffffff
    _PID=663
    _COMM=rauc
    _EXE=/usr/bin/rauc
    _CMDLINE=/usr/bin/rauc --mount=/run/rauc service
    _SYSTEMD_CGROUP=/system.slice/rauc.service
    _SYSTEMD_UNIT=rauc.service
    _SYSTEMD_INVOCATION_ID=7c28039f337d41b683f34bc4c8f16417
    GLIB_DOMAIN=rauc-event
    _HOSTNAME=<redacted>
    RAUC_EVENT_TYPE=install
    TRANSACTION_ID=5bd27691-bd38-47ab-b5a2-fdc99bae71f0
    BUNDLE_HASH=5fd0b95f57310edfb481895d3134fb4965b5243d69f72ae50ecfa34ed25e0d10
    MESSAGE=Installation 5bd27691 succeeded
    MESSAGE_ID=0163db5468ac4237b090d28490c301ed
    BUNDLE_DESCRIPTION=<redacted>
    BUNDLE_VERSION=<redacted>
    _SOURCE_REALTIME_TIMESTAMP=1721331591726253
```

**Additional context**

It's a very low priority issue.

ejoerns, 2024-07-19:
Good catch!
After reading `g_log_structured_array()` again which notes:

> This assumes that log_level is already present in fields (typically as the PRIORITY field).

it is quite clear why.

I guess my confusion comes from the `log_level` argument of `g_log_strctured_array()` which just seems to serve for the fatal mask and filtering but does not make it into a `"PRIORITY"` field.
It is also interesting that glib's `log_level_to_priority()` (that `g_log_structured()` uses to generate a `"PRIORITY"` from the `log_level`) is glib-internal only. Thus we have to map it ourselves (or use hard-coded "5")

ejoerns, 2024-07-19:
@AndreasUfert I have created https://github.com/rauc/rauc/pull/1471 to address this.

PR #1471 (2024-07-19): Add missing "PRIORITY" field to structured logging arrays
In contrast to `g_log_structured()`, `g_log_structured_array()` takes a `log_level` as input argument but uses this for filtering only.

The `"PRIORITY"` field must already be present in the array fields provided.

Note that the messages are shown in systemd with default priority anyway.
A difference can only be observed when using `journalctl` with `-o verbose`.

Fixes: #1470
commit d1c4bc7d (2024-07-19): src: add missing "PRIORITY" field to structured logging arrays

In contrast to g_log_structured(), g_log_structured_array() takes a
'log_level' as input argument but uses this for filtering only.

The "PRIORITY" field must already be present in the array fields
provided.

Note that the messages are shown in systemd with default priority
anyway. A difference can only be observed when using "journalctl" with
"-o verbose".

Fixes: #1470

Reported-by: Andreas Ufert
closed 2024-07-19 by jluebbe