rauc/rauc #1477: I want to add a tutorials section to the documentation, starting with a tutorial on how to run and test RAUC locally

Erik-Sovereign, 2024-07-25:
**Is your feature request related to a problem? Please describe.**
At the moment I'm writing a Rust tool that is going to interact with the RAUC D-Bus API on an embedded device. I wanted to test the tool on my local Ubuntu PC with a working RAUC D-Bus API (I think, this is a common thing to do). I realized there was no throughout documentation on how to do this. So I wrote one, found here: https://erik-sovereign.github.io/Software-Engineering-Blog/2024/07/10/Test-the-RAUC-D-Bus-API-locally.html

**Describe the solution you'd like**
It would make even more sense to add a Tutorials section to the documentation, where maintainers or people like me can create and collaboratively update tutorials on how to achieve things with RAUC, that are commonly needed. Explainations that don't belong into other sections of the documentation, because they could confuse people or make the structure of the documentation more confusing.

**Describe alternatives you've considered**
The alternative is the blog post, which is considerably less findable than an official place in the documentation and RAUC website.

**Additional context**
I would be happy to create the tutorial section and the first tutorial. I would love to get some feedback on the tutorial, when it is ready.

jluebbe, 2024-07-25:
There is a built-in test environment:
-  https://rauc.readthedocs.io/en/latest/using.html#reproducing-issues-using-qemu-test-setup
-  https://rauc.readthedocs.io/en/latest/contributing.html#qemu-test-runner-qemu-test

Perhaps that's not easy to find or missing some details. A Tutorial section could describe how to use `qemu-test system` in more detail and the existing sections could refer to that. What do you think?

Erik-Sovereign, 2024-07-29:
Thank you for the answer :)

This is definitely not easy to find and even when found, its not obvious that this should be used to test RAUC locally.

Unfortunately `./qemu-test` also crashed early for me with 
```
qemu-test-init: 60: cannot create /etc/resolv.conf: Directory nonexistent
[    1.861537][    T1] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000200
```

jluebbe, 2024-07-29:
> This is definitely not easy to find and even when found, its not obvious that this should be used to test RAUC locally.

It's also mentioned in https://github.com/rauc/rauc?tab=readme-ov-file#running-the-test-suite, but without any background.
 
> Unfortunately `./qemu-test` also crashed early for me with
> 
> ```
> qemu-test-init: 60: cannot create /etc/resolv.conf: Directory nonexistent
> [    1.861537][    T1] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000200
> ```

Hmm, could you attach the full output? I don't see how that could happen from the source. Which host distro do you use? Do you use RAUC master?

Zylatis, 2024-08-01:
Hey I am having the same problem, here is the full log
```
 ~/Desktop/rauc  master  sudo ./qemu-test passthrough                                                                                                                                          ✘ 1  10:51:41 
ninja: Entering directory `build/'
[1/1] Generating version.h with a custom command
found qboot, enabling
[    0.000000][    T0] Linux version 6.6.36-rauc-test-g1a7cf977edae (runner@fv-az775-981) (gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP Fri Jun 28 09:59:08 UTC 2024
[    0.004862][    T0] Kernel command line: loglevel=6 console=ttyS0 nandsim.id_bytes=0x20,0xa2,0x00,0x15 nandsim.parts=0x100 mtdram.total_size=32768 root=/dev/root rootfstype=9p rootflags=msize=16777216 rw init=/home/user/Desktop/rauc/qemu-test-init rauc.slot=system0 passthrough
[    0.004883][    T0] Unknown kernel command line parameters "passthrough", will be passed to user space.
[    0.004890][    T0] random: crng init done
[    0.027968][    T1] SCSI subsystem initialized
[    0.031604][    T1] FS-Cache: Loaded
[    0.040647][    T1] Initialise system trusted keyrings
[    0.041026][    T1] Key type blacklist registered
[    0.044943][    T1] Key type asymmetric registered
[    0.045291][    T1] Asymmetric key parser 'x509' registered
[    0.045720][    T1] Asymmetric key parser 'pkcs8' registered
[    0.046115][    T1] Key type pkcs7_test registered
[    0.046481][    T1] Warning: Processor Platform Limit event detected, but not handled.
[    0.046950][    T1] Consider compiling CPUfreq support into your kernel.
[    0.055437][    T1] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.056055][    T1] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.056638][    T1] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.057222][    T1] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.057816][    T1] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.058407][    T1] flash size: 64 MiB
[    0.058627][    T1] page size: 2048 bytes
[    0.058870][    T1] OOB area size: 64 bytes
[    0.059133][    T1] sector size: 128 KiB
[    0.059461][    T1] pages number: 32768
[    0.059783][    T1] pages per sector: 64
[    0.060092][    T1] bus width: 8
[    0.060349][    T1] bits in sector size: 17
[    0.060663][    T1] bits in page size: 11
[    0.060963][    T1] bits in OOB size: 6
[    0.061253][    T1] flash size with OOB: 67584 KiB
[    0.061639][    T1] page address bytes: 4
[    0.061960][    T1] sector address bytes: 2
[    0.062285][    T1] options: 0x8
[    0.062838][    T1] Creating 2 MTD partitions on "NAND 64MiB 1,8V 8-bit":
[    0.063366][    T1] 0x000000000000-0x000002000000 : "NAND simulator partition 0"
[    0.064173][    T1] 0x000002000000-0x000004000000 : "NAND simulator partition 1"
[    0.069475][    T1] Loading compiled-in X.509 certificates
[    0.070998][    T1] Key type .fscrypt registered
[    0.071368][    T1] Key type fscrypt-provisioning registered
[    0.072152][    T1] Key type encrypted registered
+ export PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
+ mount -t proc none /proc
+ mount -t sysfs none /sys
+ mount -t debugfs debugfs /sys/kernel/debug/
+ mount -t tracefs tracefs /sys/kernel/tracing/
+ mount -t tmpfs none /mnt
+ mount -t tmpfs none /root
+ mount -t tmpfs none /run
+ mount -t tmpfs none /tmp
+ mkdir /dev/shm
+ mount -t tmpfs none /dev/shm
+ mkdir /tmp/etc-overlay /tmp/etc-work
+ mount -t overlay overlay -o lowerdir=/etc,upperdir=/tmp/etc-overlay,workdir=/tmp/etc-work /etc
+ dirname /home/user/Desktop/rauc/qemu-test-init
+ cd /home/user/Desktop/rauc
+ realpath build
+ BUILD_DIR=/home/user/Desktop/rauc/build/
+ cat /proc/cmdline
+ [ loglevel=6 = shell ]
+ [ loglevel=6 = system ]
+ [ loglevel=6 = asan ]
+ [ loglevel=6 = service-backtrace ]
+ [ console=ttyS0 = shell ]
+ [ console=ttyS0 = system ]
[... 51 lines cut ...]
[    0.175632][    T1] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000200
[    0.175991][    T1] CPU: 0 PID: 1 Comm: qemu-test-init Not tainted 6.6.36-rauc-test-g1a7cf977edae #1
[    0.176359][    T1] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 0.1 11/11/2019
[    0.176694][    T1] Call Trace:
[    0.176827][    T1]  <TASK>
[    0.176945][    T1]  dump_stack_lvl+0x33/0x50
[    0.177128][    T1]  panic+0x288/0x2d0
[    0.177285][    T1]  do_exit+0x843/0x9b0
[    0.177450][    T1]  ? handle_mm_fault+0x95/0x170
[    0.177647][    T1]  do_group_exit+0x28/0x80
[    0.177826][    T1]  __x64_sys_exit_group+0xf/0x10
[    0.178023][    T1]  do_syscall_64+0x35/0x80
[    0.178201][    T1]  entry_SYSCALL_64_after_hwframe+0x4b/0xb5
[    0.178435][    T1] RIP: 0033:0x7fd659a6bc31
[    0.178610][    T1] Code: 05 f2 12 00 be e7 00 00 00 ba 3c 00 00 00 eb 16 66 0f 1f 84 00 00 00 00 00 89 d0 0f 05 48 3d 00 f0 ff ff 77 1c f4 89 f0 0f 05 <48> 3d 00 f0 ff ff 76 e7 f7 d8 64 41 89 00 eb df 0f 1f 80 00 00 00
[    0.179591][    T1] RSP: 002b:00007ffd0f4d9208 EFLAGS: 00000246 ORIG_RAX: 00000000000000e7
[    0.180135][    T1] RAX: ffffffffffffffda RBX: 0000000000000003 RCX: 00007fd659a6bc31
[    0.180652][    T1] RDX: 000000000000003c RSI: 00000000000000e7 RDI: 0000000000000002
[    0.181168][    T1] RBP: 0000000000000000 R08: ffffffffffffff80 R09: 0000000000000003
[    0.181683][    T1] R10: 0000000000000000 R11: 0000000000000246 R12: 00007ffd0f4d9478
[    0.182197][    T1] R13: 0000000000000000 R14: 0000555cf9814ef8 R15: 00007fd659bfd040
[    0.182712][    T1]  </TASK>
[    0.182966][    T1] Kernel Offset: disabled
[    0.183153][    T1] Rebooting in 1 seconds..
```

ejoerns, 2024-09-26:
@Zylatis I also have no real clue what is going wrong here.

The entire `/etc` directory created earlier in the script:

> mount -t overlay overlay -o lowerdir=/etc,upperdir=/tmp/etc-overlay,workdir=/tmp/etc-work /etc

does not seem to exist anymore later in the script:

> /home/user/Desktop/rauc/qemu-test-init: 61: cannot create /etc/resolv.conf: Directory nonexistent

Removing the line  `echo "nameserver 10.0.2.3" > /etc/resolv.conf` from `qemu-test-init` and doing `./qemu-test shell` and listing content of `/` or `/etc` could help to debug the issue.

JSydll, 2024-09-27:
@Erik-Sovereign and @Zylatis, the issues you observed with `qemu-test` should be fixed on latest master after !1526 now. Maybe you could confirm whether the original issue persists?

Erik-Sovereign, 2025-07-23:
Sorry for answering late. I have no plan to do that at this moment.

closed 2025-07-23 by Erik-Sovereign