rauc/rauc #1479: Failed to obtain name on system bus despite dbus files installed

Zylatis, 2024-08-01:
**Describe the bug**
After following the install steps to get RAUC on a ubuntu target (virtualized), the dbus service fails to start and i cannot install the bundle generated on host.

**Background information**
I am testing out RAUC in a VM (which as far as i can tell should not be an issue?). It compiles and installs with no problems, and puts the dbus service/config files in the correct place. However when trying to install it times out due do the dbus service failing (logs below)

**To Reproduce**
Steps to reproduce the behavior:
As described above

**Expected behavior**
I expect the service to start and to be able to install a bundle

**Logs**
Install log
[image]
Dbus file loc
[image]
Service failure
[image]
Conf location
[image]
Conf contents
[image]

**Additional context**
I have seen this issue crop up in other contexts but the cause in those cases seemed to be incorrect install/lack of dbus config but i can't see what i'm missing in that regard here. I also saw some discussion of building without the dbus service but i'm unsure of how to do that/if it's appropriate. 

Thanks in advance, happy to provide more info. Looking forward to getting RAUC up and running!

Zylatis, 2024-08-08:
Ah, i see it is meant to copy files to `/etc/dbus-1/system/` instead maybe, at least moving from `/usr/share` to there has fixed it. Not sure if this is correct or if i understand what's going on very well though...

ejoerns, 2024-08-08:
@Zylatis Thanks for reporting. I guess this is caused by dbuspolicydir being `/usr/local/share/dbus-1/system.d` in your install log above. I have checked how this directory is constructed and proposed a fix in #1483.

Would be great if you could test it.

PR #1483 (2024-08-08): meson.build: derive dbuspolicydir from libdbus datadir
All other dbus directory locations are derived from libdbus pkg-config variables.
The dbuspolicydir instead only used the 'datadir' which has a fixed default and might thus not be valid on some distributions.

If the value cannot be obtained from pkg-config, the fallback remains the same as before.

Fixes #1479
commit 0327a8f3 (2024-08-08): meson.build: derive dbuspolicydir from libdbus datadir

All other dbus directory locations are derived from libdbus pkg-config
variables.
The dbuspolicydir instead only used the 'datadir' which has a fixed
default and might thus not be valid on some distributions.

If the value cannot be obtained from pkg-config, the fallback remains
the same as before.

Fixes #1479
closed 2024-08-08 by jluebbe