rauc/rauc #1488: Rauc issue with encrption keys from optee Pkcs#11 background

Muhammadhassan1234, 2024-08-10:
**Describe the bug**
A clear and concise description of what the bug is.

**Background information**
Describe which RAUC version and build system you are using.

**To Reproduce**
Steps to reproduce the behavior:
1. …

**Expected behavior**
A clear and concise description of what you expected to happen.

**Logs**
If applicable, add console output or logs from the RAUC service to help explain your problem.

**Additional context**
Add any other context about the problem here.

I have a rauc encrypted bundle and I am trying to load keys and certs from optee using pkcs11 object. When I try to do this, I got the following error

`LastError: Failed to decrypt bundle: Failed to decrypt CMS EnvelopedData: error:170000BA:CMS routines::kdf parameter error
`
Here are the keys and certs value in system.conf

[image]

I have stored the objects in optee using pkcs#11 using the below API'S

```

pkcs11-tool --module /usr/lib/libckteec.so.0 --init-token --label rauc --so-pin 12341234

pkcs11-tool --module /usr/lib/libckteec.so.0 --label rauc --login --so-pin 12341234 --init-pin --pin 12341234

pkcs11-tool --module /usr/lib/libckteec.so.0 --login --write-object crypt.key.pem --type privkey --label crypt-key

pkcs11-tool --module /usr/lib/libckteec.so.0 --login --write-object crypt.cert.pem --type cert --label pub-cert
```

The key is loaded from optee successfully but then this function gave the above error in rauc

```
	if (!CMS_decrypt(icms, privkey, NULL, NULL, outdecrypt, 0)) {
		res = NULL;
		g_set_error(
				error,
				R_SIGNATURE_ERROR,
				R_SIGNATURE_ERROR_INVALID,
				"Failed to decrypt CMS EnvelopedData: %s", get_openssl_err_string());
		goto out;
	}

```

If I try to do the same with loading keys from file, it works fine. Can anyone help me in analyzing the root cause of this issue?

ejoerns, 2024-08-10:
@Muhammadhassan1234 Which version of RAUC is that?

The cited code in RAUC looks different from your code:

``` diff
- if (!CMS_decrypt(icms, privkey, NULL, NULL, outdecrypt, 0)) {
+ if (!CMS_decrypt(icms, privkey, decrypt_cert, NULL, outdecrypt, 0)) {

Muhammadhassan1234, 2024-08-10:
@ejoerns I was just playing with the code but the error persists with the original code too. What I think is that the API's used inside rauc to fetch privkey from pkcs11 background return success but actually don't fetch the keys. 

I made a test where I tried to print the key. When I fetched the key from file in userspace and printed it from rauc code, it printed fine. However, when I fetched the same key from pkcs11 API , it returned success but no key was available in privkey variable. Here is a simple function I used for printing ( chatgpt made it for me so pardon if my testing is wrong)

```

	BIO *bio_out = BIO_new(BIO_s_mem());
	if (PEM_write_bio_PrivateKey(bio_out, privkey, NULL, NULL, 0, NULL, NULL)) {
		char *key_data;
		long key_len = BIO_get_mem_data(bio_out, &key_data);
		g_message("Private Key:\n%.*s", (int)key_len, key_data);
	} else {
		g_message("Failed to print private key");
	}
	BIO_free(bio_out);
```

So, here the few observations that I want to discuss. First, When I fetched the key from file and printed it, it printed the key in the plain text form as in my original file. But when I fetched the key from the PKCS11 and tried to print it, it returned `Failed to print private key`. Which clearly means that API used for fetching keys via pkcs11 are not working perfectly.

Second approach was that I tried to read the private key from optee using pkcs11-tool commands and it says that it doesn't support reading private objects. But rauc tries to fetch the keys from the optee and then use it in the decrypt function. is it possible? 

Thirdly, if rauc with pkcs11 support in any Hardware is tested earlier? Is there any guideline about how we should store keys in optee with pkcs11 so that they can work well with rauc. I saw just 1-2 documents of optee which just tells about how to make pkcs11 based url in system.conf and 2 environment variable for setting the pin and module_path.

Muhammadhassan1234, 2024-08-29:
@ejoerns or community! have anyone ever tested rauc with pkcs11 and optee?

jluebbe, 2024-08-29:
> I made a test where I tried to print the key. When I fetched the key from file in userspace and printed it from rauc code, it printed fine. However, when I fetched the same key from pkcs11 API , it returned success but no key was available in privkey variable.

OP-TEE (or any other HSM) doesn't allow read-out of the private key material, as that would be against the whole purpose of a HSM, which is to protect the key material. Instead, in OpenSSL, you get a handle which allows you to sign or decrypt with the key. So it's expected that `PEM_write_bio_PrivateKey` would fail.

> Thirdly, if rauc with pkcs11 support in any Hardware is tested earlier? Is there any guideline about how we should store keys in optee with pkcs11 so that they can work well with rauc. I saw just [...]

The current test-suite only uses PKCS#11 for signing and decryption is only tested with files. It should be relatively easy to extend that do decyption.

jluebbe, 2024-08-29:
> The current test-suite only uses PKCS#11 for signing, while decryption is only tested with files. It should be relatively easy to extend that to decyption.

I've added a test case for decryption using PKCS#11 in https://github.com/rauc/rauc/pull/1508, which runs successfully.

Commit https://github.com/rauc/rauc/pull/1508/commits/f4f5393952a61473573e8a15913eb16d6b77feb2 might also be useful to you as an example, as it calls the `openssl` CLI to use the keys via PKCS#11 without RAUC.

Which algorithm (RSA, ECC, ...) do you use?

ejoerns, 2024-09-26:
@Muhammadhassan1234 Did the working test case help you with your PKCS#11 encryption setup? Or did it help to find out where RAUC is probably incompatible with your setup?

ejoerns, 2024-11-13:
Since the test cases work and there were no more details about the conditions when encryption with keys from PKCS#11 does not work, I removed the bug label and close the issue now. If there are more findings, feel free to re-open.

closed 2024-11-13 by ejoerns