rauc/rauc #1517: Streaming installation aborts on bad network conditions.

BubuOT, 2024-09-11:
**Describe the bug**
When doing a streaming installation over an unreliable mobile network connection we sometimes get a fatal error "Installation error: Failed updating slot rootfs.1: Failed splicing data: Error reading from file: Input/output error" and the installation aborts.

**Background information**
buildroot with rauc v1.11.3 (recently upgraded to 1.12 but the logs are still from the old version)

**To Reproduce**
Very unreliable to reproduce. 
Start a streaming installation over a fleet of 2k devices connected via mobile network
~10-50 of these will fail and need to be retried, sometimes >5 times, increasing the traffic consumption of the update drastically. Some also never manage to install the update until mobile connectivity improves.

**Expected behavior**
Rauc keeps continuing the installation over the unreliable network and eventually succeeds.

**Logs**
kernel:
```
Jun 03 16:47:57 othermo-8E56A652 kernel: nbd0: detected capacity change from 0 to 209416
Jun 03 16:47:57 othermo-8E56A652 kernel: device-mapper: ioctl: 4.45.0-ioctl (2021-03-22) initialised: dm-devel@redhat.com
Jun 03 16:47:58 othermo-8E56A652 kernel: device-mapper: verity: sha256 using implementation "sha256-generic"
Jun 03 16:50:26 othermo-8E56A652 kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
Jun 03 16:57:21 othermo-8E56A652 kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
Jun 03 17:03:06 othermo-8E56A652 kernel: INFO: task kcompactd0:43 blocked for more than 120 seconds.
Jun 03 17:03:06 othermo-8E56A652 kernel:       Tainted: G         C        5.15.84-v8 #1
Jun 03 17:03:06 othermo-8E56A652 kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Jun 03 17:03:06 othermo-8E56A652 kernel: task:kcompactd0      state:D stack:    0 pid:   43 ppid:     2 flags:0x00000008
Jun 03 17:03:06 othermo-8E56A652 kernel: Call trace:
Jun 03 17:03:06 othermo-8E56A652 kernel:  __switch_to+0x110/0x170
Jun 03 17:03:06 othermo-8E56A652 kernel:  __schedule+0x314/0x870
Jun 03 17:03:06 othermo-8E56A652 kernel:  schedule+0x6c/0x130
Jun 03 17:03:06 othermo-8E56A652 kernel:  io_schedule+0x44/0x70
Jun 03 17:03:06 othermo-8E56A652 kernel:  wait_on_page_bit_common+0x114/0x310
Jun 03 17:03:06 othermo-8E56A652 kernel:  __lock_page+0x5c/0x80
Jun 03 17:03:06 othermo-8E56A652 kernel:  migrate_pages+0x150/0xa70
Jun 03 17:03:06 othermo-8E56A652 kernel:  compact_zone+0x59c/0xe40
Jun 03 17:03:06 othermo-8E56A652 kernel:  proactive_compact_node+0x78/0xb0
Jun 03 17:03:06 othermo-8E56A652 kernel:  kcompactd+0x1a0/0x3f0
Jun 03 17:03:06 othermo-8E56A652 kernel:  kthread+0x140/0x160
Jun 03 17:03:06 othermo-8E56A652 kernel:  ret_from_fork+0x10/0x20
Jun 03 17:04:16 othermo-8E56A652 kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
Jun 03 17:05:07 othermo-8E56A652 kernel: INFO: task kcompactd0:43 blocked for more than 241 seconds.
Jun 03 17:05:07 othermo-8E56A652 kernel:       Tainted: G         C        5.15.84-v8 #1
Jun 03 17:05:07 othermo-8E56A652 kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Jun 03 17:05:07 othermo-8E56A652 kernel: task:kcompactd0      state:D stack:    0 pid:   43 ppid:     2 flags:0x00000008
Jun 03 17:05:07 othermo-8E56A652 kernel: Call trace:
Jun 03 17:05:07 othermo-8E56A652 kernel:  __switch_to+0x110/0x170
Jun 03 17:05:07 othermo-8E56A652 kernel:  __schedule+0x314/0x870
Jun 03 17:05:07 othermo-8E56A652 kernel:  schedule+0x6c/0x130
Jun 03 17:05:07 othermo-8E56A652 kernel:  io_schedule+0x44/0x70
Jun 03 17:05:07 othermo-8E56A652 kernel:  wait_on_page_bit_common+0x114/0x310
Jun 03 17:05:07 othermo-8E56A652 kernel:  __lock_page+0x5c/0x80
Jun 03 17:05:07 othermo-8E56A652 kernel:  migrate_pages+0x150/0xa70
Jun 03 17:05:07 othermo-8E56A652 kernel:  compact_zone+0x59c/0xe40
Jun 03 17:05:07 othermo-8E56A652 kernel:  proactive_compact_node+0x78/0xb0
Jun 03 17:05:07 othermo-8E56A652 kernel:  kcompactd+0x1a0/0x3f0
Jun 03 17:05:07 othermo-8E56A652 kernel:  kthread+0x140/0x160
Jun 03 17:05:07 othermo-8E56A652 kernel:  ret_from_fork+0x10/0x20
Jun 03 17:05:52 othermo-8E56A652 kernel: block nbd0: Connection timed out, retrying (1/1 alive)
Jun 03 17:05:52 othermo-8E56A652 kernel: block nbd0: Receive control failed (result -32)
Jun 03 17:05:52 othermo-8E56A652 kernel: block nbd0: Dead connection, failed to find a fallback
Jun 03 17:05:52 othermo-8E56A652 kernel: block nbd0: shutting down sockets
Jun 03 17:05:52 othermo-8E56A652 kernel: blk_update_request: I/O error, dev nbd0, sector 73824 op 0x0:(READ) flags 0x800 phys_seg 8 prio class 0
Jun 03 17:05:52 othermo-8E56A652 kernel: SQUASHFS error: Failed to read block 0x240c439: -5
Jun 03 17:05:52 othermo-8E56A652 kernel: SQUASHFS error: Unable to read data cache entry [240c439]
Jun 03 17:05:52 othermo-8E56A652 kernel: SQUASHFS error: Unable to read page, block 240c439, size 7263
Jun 03 17:05:52 othermo-8E56A652 kernel: SQUASHFS error: Unable to read data cache entry [240c439]
Jun 03 17:05:52 othermo-8E56A652 kernel: SQUASHFS error: Unable to read page, block 240c439, size 7263
Jun 03 17:05:52 othermo-8E56A652 kernel: SQUASHFS error: Unable to read data cache entry [240c439]
Jun 03 17:05:52 othermo-8E56A652 kernel: SQUASHFS error: Unable to read page, block 240c439, size 7263
Jun 03 17:05:52 othermo-8E56A652 kernel: blk_update_request: I/O error, dev nbd0, sector 208016 op 0x0:(READ) flags 0x0 phys_seg 2 prio class 0
[... 56 lines cut ...]
Jun 03 16:48:12 othermo-8E56A652 rauc[245]: installing <REDACTED>: Updating slots...
Jun 03 16:48:12 othermo-8E56A652 rauc[245]: installing <REDACTED>: Checking slot bootloader.0
Jun 03 16:48:12 othermo-8E56A652 rauc[245]: Updating /dev/mmcblk0 with /run/rauc/bundle/boot.vfat
Jun 03 16:48:12 othermo-8E56A652 rauc[245]: installing <REDACTED>: Updating slot bootloader.0
Jun 03 16:48:12 othermo-8E56A652 rauc[245]: Found inactive (first) half of boot partition region (pos. 4194304B, size 33554432B)
Jun 03 16:48:12 othermo-8E56A652 rauc[245]: Clearing inactive (first) half of boot partition region on /dev/mmcblk0
Jun 03 16:48:14 othermo-8E56A652 rauc[245]: Write image to inactive (first) half of boot partition region on /dev/mmcblk0
Jun 03 16:49:02 othermo-8E56A652 rauc[245]: Setting first half of boot partition region active in MBR
Jun 03 16:49:02 othermo-8E56A652 rauc[245]: installing <REDACTED>: Updating slot bootloader.0 status
Jun 03 16:49:02 othermo-8E56A652 rauc[245]: installing <REDACTED>: Updating slot bootloader.0 done
Jun 03 16:49:02 othermo-8E56A652 rauc[245]: installing <REDACTED>: Checking slot rootfs.1
Jun 03 16:49:02 othermo-8E56A652 rauc[245]: Updating /dev/mmcblk0p6 with /run/rauc/bundle/rootfs.ext4
Jun 03 16:49:02 othermo-8E56A652 rauc[245]: installing <REDACTED>: Updating slot rootfs.1
Jun 03 17:05:52 othermo-8E56A652 rauc-nbd[460941]: exiting nbd server
Jun 03 17:05:52 othermo-8E56A652 rauc-nbd[460941]: nbd server failed with: failed to read request from client: Unexpected end of file
Jun 03 17:05:52 othermo-8E56A652 rauc[245]: Continuing after adaptive mode error: no chunk with required hash [3483234d4d01c46adfb1142f2bb6e0d65162a7caa2f370a2bfcac324f2651ad4] found
Jun 03 17:05:52 othermo-8E56A652 rauc[245]: opening slot device /dev/mmcblk0p6
Jun 03 17:05:52 othermo-8E56A652 rauc[245]: writing data to device /dev/mmcblk0p6
Jun 03 17:05:52 othermo-8E56A652 rauc[245]: r_nbd_remove_device
Jun 03 17:05:52 othermo-8E56A652 rauc[245]: nbd server stopping
Jun 03 17:05:52 othermo-8E56A652 rauc[245]: Installation error: Failed updating slot rootfs.1: Failed splicing data: Error reading from file: Input/output error
Jun 03 17:05:52 othermo-8E56A652 rauc[245]: installing <REDACTED>: Installation error: Failed updating slot rootfs.1: Failed splicing data: Error reading from file: Input/output error
Jun 03 17:05:52 othermo-8E56A652 rauc[245]: installing <REDACTED>: finished
Jun 03 17:05:52 othermo-8E56A652 rauc[245]: installing `<REDACTED>` failed: 1
```

jluebbe, 2024-09-11:
It seems that the NBD client reaches it's timeout (should be 5 minutes) on a request before we see any errors from curl. This means that a single get request is stuck for a similar amount of time.

Should RAUC continue to wait for the HTTP TCP connection to eventually complete in cases like these? I'm not sure that would be useful as less than a few KiB in 5 minutes is basically no network at all.

I've created https://github.com/rauc/rauc/pull/1518 to print the curl statistics even on errors like these. That might help to debug here.

BubuOT, 2024-09-11:
> Should RAUC continue to wait for the HTTP TCP connection to eventually complete in cases like these?

I don't know what best to do on that level tbh. The problem we are facing from a user level is that this might be running for 1-2 hours, managing to install something like 80% of a 120 MB update bundle but than failing without a retry/recovery path. (I need to schedule a new update action from hawkbit, and afaik the data already downloaded isn't re-used)

BubuOT, 2024-09-11:
> If you're using adaptive (block-hash-index) updates and it's re-downloading blocks that have already been written, at least that part could (and should :) be fixed.

We are using adaptive bundles, yes.

> Otherwise, the NBD timeout could be made configurable to just let it wait longer, which would only help if the network connection actually recovers in a reasonable timeframe.

If I remember correctly the non-streaming mode (download via rauc-hawkbit-updater) retries the download even on timeout errors? (`retry_wait` from https://rauc-hawkbit-updater.readthedocs.io/en/latest/reference.html#configuration-file, but maybe I'm misremembering/misunderstanding?)

So my problem (besides increased data consumption, which you mentioned should be fixed either way) is really that timeout=update-failure is a bad experience for us. It would be much nicer if the update (or single chunk or whatever makes sense) is just being retried on timeout errors. Connection quality varies wildly and things eventually completing (even if it takes a day or two) is much better than having to baby sit the update for those 5% of the fleet that have a not great internet connection.

jluebbe, 2024-09-12:
> If I remember correctly the non-streaming mode (download via rauc-hawkbit-updater) retries the download even on timeout errors? (`retry_wait` from [...]

Yes, but with streaming mode, the installation itself will fail. Currently, it's not really clean from the returned error that this was a network error. That could probably be improved by checking the state of the NBD device. In that case, retrying from rauc-hawkbit-updater might be useful.

> So my problem (besides increased data consumption, which you mentioned should be fixed either way) is really that timeout=update-failure is a bad experience for us. It would be much nicer if the [...]

The simplest change/workaround for you would probably be to make the [NBD timeout ](https://github.com/rauc/rauc/blob/master/src/nbd.c#L181) configurable. If you increase it to hours, the [request level retries](https://github.com/rauc/rauc/blob/master/src/nbd.c#L954) would be able to trigger. Currently, they are hard-coded to 5 tries, so that might need to be made configurable as well.

With that, the installation would just become very slow until network conditions improve.