rauc/rauc #1522: Rauc install command fails with PKCS11 URL

RoboAsystent, 2024-09-13:
**Describe the bug**
rauc install from https server fails when pkcs11 url for private key is used,  local ssl certificate issue is reported in service logs

**Background information**
Yocto kirkstone with rauc 1.11.1
Atecc608 HSM module with libcryptoauth as provider

**To Reproduce**
Steps to reproduce the behavior:
1. Add [streaming] section into rauc system.conf :

[streaming]
sandbox-user=<user-name>
tls-cert=client.crt.pem
tls-key=pkcs11:token=00ABC;object=device
tls-ca=CA.pem

2. export RAUC_PKCS11_PIN=<pin>
3. systemctl restart rauc
4. rauc install https://localhost/bundle.raucb -d

**Expected behavior**
Bundle installation should finish with success

**Logs**
journal logs:

Sep 12 22:54:47  rauc-nbd[18047]: running as UID 0, GID 0
Sep 12 22:54:47 rauc-nbd[18047]: received: {'url': <'https://localhost/bundle.raucb'>, 'key': <'pkcs11:token=00ABC;object=device'>, 'info-headers': <@as []>, 'ca': <'CA.pem'>, 'cert': <'client.crt.pem'>}
Sep 12 22:54:47 rauc-nbd[18047]: configuring for URL: https://localhost/bundle.raucb
Sep 12 22:54:47 rauc-nbd[18047]: request failed: Problem with the local SSL certificate (retrying)
Sep 12 22:54:48 rauc-nbd[18047]: configuring for URL: https://localhost/bundle.raucb
Sep 12 22:54:48 rauc-nbd[18047]: request failed: Problem with the local SSL certificate (retrying)
Sep 12 23:34:44 rauc-nbd[21424]: request failed (no more retries)
Sep 12 23:34:44 rauc[21408]: received http error 0
Sep 12 23:34:44 rauc-nbd[21424]: exiting nbd server
Sep 12 23:34:44 rauc-nbd[21424]: nbd server failed with: finish_request failed, shutting down
Sep 12 23:34:44 rauc[21408]: nbd server stopping

rauc logs:

rauc-Message: 23:30:30.762: Debug log domains: 'rauc'
(rauc:21040): rauc-DEBUG: 23:30:30.765: install started
(rauc:21040): rauc-DEBUG: 23:30:30.765: input bundle: https://localhost/bundle.raucb
(rauc:21040): rauc-DEBUG: 23:30:30.776: Trying to contact rauc service
installing
  0% Installing
  0% Determining slot states
 10% Determining slot states done.
 10% Checking bundle
 20% Checking bundle failed.
100% Installing failed.
LastError: Failed to stream bundle https://localhost/bundle.raucb: failed to configure streaming: server not responding
Installing `https://localhost/bundle.raucb` failed

**Additional context**
Add any other context about the problem here.

I have tried to increase debugging verbosity by setting G_MESSAGES_DEBUG="all", but there were no more logs.
Is there any way to track mentioned problem with certificate?

When I use private key stored as file then install command does not fail, so I assume that there is issue with using pkcs11 object
I saw in sources that pkcs11 is used in signature.c and there are error logs, but in my case no errors related to pkcs11 or crypto operations were reported

jluebbe, 2024-09-13:
> **Background information** Yocto kirkstone with rauc 1.11.1 Atecc608 HSM module with libcryptoauth as provider

Please update to RAUC 1.12, which is available in meta-rauc for kirkstone:
https://github.com/rauc/meta-rauc/commit/9b55c6e1cbab71b633ecccbdd3a1f39c80e2f18c

> ```
> [streaming]
> sandbox-user=user-name
> tls-cert=client.crt.pem
> tls-key=pkcs11:token=00ABC;object=device
> tls-ca=CA.pem
> ```

Try using absolute paths. We currently don't resolve these to absolute paths automatically.

> `Sep 12 22:54:47 rauc-nbd[18047]: running as UID 0, GID 0`

Hmm, is you sandbox user somehow mapped to root?

> `Sep 12 22:54:47 rauc-nbd[18047]: received: {'url': 'https://localhost/bundle.raucb'), 'key': <'pkcs11:token=00ABC;object=device'>, 'info-headers': <@as []>, 'ca': <'CA.pem'>, 'cert': <'client.crt.pem'>}`

As above, these aren't absolute paths.

> `Sep 12 22:54:47 rauc-nbd[18047]: configuring for URL: https://localhost/bundle.raucb`
> `Sep 12 22:54:47 rauc-nbd[18047]: request failed: Problem with the local SSL certificate (retrying)`

You can enable more debug output: https://rauc.readthedocs.io/en/latest/using.html#enabling-verbose-curl-output

> **Additional context** Add any other context about the problem here.
> 
> I have tried to increase debugging verbosity by setting G_MESSAGES_DEBUG="all", but there were no more logs. Is there any way to track mentioned problem with certificate?

You also need to pass `-d` when starting the service.

> When I use private key stored as file then install command does not fail, so I assume that there is issue with using pkcs11 object I saw in sources that pkcs11 is used in signature.c and there are error logs, but in my case no errors related to pkcs11 or crypto operations were reported

Hmm. Perhaps try using strace in addition to the curl logs to better see where it's failing? Does the curl CLI work fine?

RoboAsystent, 2024-09-17:
Thanks for your fast response

My bad I haven't checked for updates. I saw in release info that there was issue with pkcs11 URL resolved in rauc 1.12

> Hmm, is you sandbox user somehow mapped to root?

Yes, I've development environment where the user is actually root. I was unaware that install command is starting sub processes as not root user, so I had to add sandbox-user into config

> Try using absolute paths. We currently don't resolve these to absolute paths automatically.

Those are absolute paths, I just replaced them here with aliases such as "CA.pem"

Unfortunately problem with pkcs11 private key persists, now rauc-nbd is returning following error:

`rauc-nbd[1916]: nbd server configuring for URL: https://localhost/bundle.raucb
rauc-nbd[1916]: request failed: unable to set private key file: 'pkcs11:token=00ABC;object=device' type PEM (retrying 1/5)
rauc-nbd[1916]: nbd server configuring for URL: https://localhost/bundle.raucb
rauc-nbd[1916]: request failed: unable to set private key file: 'pkcs11:token=00ABC;object=device' type PEM (retrying 2/5)
rauc-nbd[1916]: nbd server configuring for URL: https://localhost/bundle.raucb
rauc-nbd[1916]: request failed: unable to set private key file: 'pkcs11:token=00ABC;object=device' type PEM (retrying 3/5)
rauc-nbd[1916]: nbd server configuring for URL: https://localhost/bundle.raucb
rauc-nbd[1916]: request failed: unable to set private key file: 'pkcs11:token=00ABC;object=device' type PEM (retrying 4/5)
rauc-nbd[1916]: nbd server configuring for URL: https://localhost/bundle.raucb
rauc-nbd[1916]: request failed: unable to set private key file: 'pkcs11:token=00ABC;object=device' type PEM (retrying 5/5)
rauc-nbd[1916]: nbd server configuring for URL: https://localhost/bundle.raucb
rauc-nbd[1916]: request failed (no more retries)
`

Same config for streaming section was used. For some reason --tls-key is automatically assumed to be a .pem file, where actually it supposed to be handled as pkcs11 URL.

I've looked into documentation first if I haven't forgotten setting any relevant variable or flag for rauc to enable pkcs11, but I haven't found anything.

ejoerns, 2024-09-26:
@RoboAsystent I'm not that deep into the topic but had a short look. What we do in RAUC is to set `CURLOPT_SSLKEY` only.

If your above example works and all options are necessary, then it might be necessary for using the PKCS#11 URI as `tls-key` to set `CURLOPT_SSLENGING` and `CURLOPT_SSLKEYTYPE` in RAUC, too. Like

``` c
code |= curl_easy_setopt(xfer->easy, CURLOPT_SSLKEY, xfer->ctx->tls_key);
code |= curl_easy_setopt(xfer->easy, CURLOPT_SSLKEYTYPE, "ENG");
code |= curl_easy_setopt(xfer->easy, CURLOPT_SSLENGINE, "pkcs11");
```

Could you test if that works for you (by modifying `prepare_curl()` in `src/nbd.c` accordingly)?

ejoerns, 2024-11-13:
@RoboAsystent Have you been able to test this?