rauc/rauc #1524: Log messages to from openssl are not diverted to stderr

frdve, 2024-09-19:
**Describe the bug**
When I request a bundle info in JSON format, there are error messages in the response

`sudo rauc info --output-format json /tmp/bundle.raucb
rauc-Message: 15:38:12.903: Using central status file /data/rauc/central.raucs
rauc-Message: 15:38:12.903: valid /etc/rauc/system.conf found, using it
rauc-Message: 15:38:12.903: Reading bundle: /tmp/bundle.raucb
rauc-Message: 15:38:12.909: Verifying bundle signature...
rauc-Message: 15:38:12.911: Verified inline signature by 'O = rauc Inc., CN = rauc-demo'
409A8171817F0000:error:1700006B:CMS routines:cms_get_enveloped_type:content type not enveloped data:../crypto/cms/cms_env.c:41:
409A8171817F0000:error:1700006B:CMS routines:cms_get_enveloped_type:content type not enveloped data:../crypto/cms/cms_env.c:41:
409A8171817F0000:error:1700006B:CMS routines:cms_get_enveloped_type:content type not enveloped data:../crypto/cms/cms_env.c:41:
{"compatible":"EdgeServer","version":"2024.09-6","description":null,"build":null,"format":"verity","hooks":[],"hash":"9165df4a7cf2738f8e60cd9455293dd3471617aa5657c9978a173c801a23ac85","images":[{"rootfs":{"variant":null,"filename":"diskimage.img","checksum":"793c65ffb9be0d47b0aa1574e88e50ca52740d5fe30a7be363cfe30386b7bc04","size":26843545600,"hooks":[],"adaptive":[]}}]}`

This makes it difficult to handle JSON parsing.  I noticed this should be fixed in version 1.7 (https://rauc.readthedocs.io/en/v1.11.3/changes.html#release-1-7-released-jun-3-2022), but it looks like its not handled correcly here.

Also (and this should maybe be another ticket), I dont know why this error message is there... 

**Background information**
Describe which RAUC version and build system you are using.
rauc: 1.12
libglib2.0-dev: 2.72.4-0ubuntu2.3
OpenSSL: 3.0.2 15 Mar 2022
Platform: Ubuntu 22.04.5 LTS

**To Reproduce**
Steps to reproduce the behavior:
1. sudo rauc info --output-format json /tmp/bundle.raucb

**Expected behavior**
Only the JSON should be returned.  The lines with the error should go to stderr

**Logs**
sudo rauc info --output-format json /tmp/bundle.raucb
rauc-Message: 15:38:12.903: Using central status file /data/rauc/central.raucs
rauc-Message: 15:38:12.903: valid /etc/rauc/system.conf found, using it
rauc-Message: 15:38:12.903: Reading bundle: /tmp/bundle.raucb
rauc-Message: 15:38:12.909: Verifying bundle signature...
rauc-Message: 15:38:12.911: Verified inline signature by 'O = rauc Inc., CN = rauc-demo'
409A8171817F0000:error:1700006B:CMS routines:cms_get_enveloped_type:content type not enveloped data:../crypto/cms/cms_env.c:41:
409A8171817F0000:error:1700006B:CMS routines:cms_get_enveloped_type:content type not enveloped data:../crypto/cms/cms_env.c:41:
409A8171817F0000:error:1700006B:CMS routines:cms_get_enveloped_type:content type not enveloped data:../crypto/cms/cms_env.c:41:
{"compatible":"EdgeServer","version":"2024.09-6","description":null,"build":null,"format":"verity","hooks":[],"hash":"9165df4a7cf2738f8e60cd9455293dd3471617aa5657c9978a173c801a23ac85","images":[{"rootfs":{"variant":null,"filename":"diskimage.img","checksum":"793c65ffb9be0d47b0aa1574e88e50ca52740d5fe30a7be363cfe30386b7bc04","size":26843545600,"hooks":[],"adaptive":[]}}]}

**Additional context**
Add any other context about the problem here.

ejoerns, 2024-09-19:
> sudo rauc info --output-format json /tmp/bundle.raucb

@frdve Why is 'sudo' for `rauc info` required in your use case?

> This makes it difficult to handle JSON parsing. I noticed this should be fixed in version 1.7 (https://rauc.readthedocs.io/en/v1.11.3/changes.html#release-1-7-released-jun-3-2022), but it looks like its not handled correcly here.

This refers to log messages emitted by RAUC itself (and that go through glib). The openssl messages are emitted by openssl and we have no direct control here.

Openssl 3.0.2 is relatively old. Please try updating to a more recent version and check if the issue remains. I remember having seen these log messages, too, mainly back in 2022.

frdve, 2024-09-20:
Thanks for the response!

> Why is 'sudo' for rauc info required in your use case?
Valid question, normall I dont use sudo on rauc, but the bundle.raucb was downloaded by another application (qbee) and I didn't had permissions to it.  I changed the permissions and tried without the 'sudo', but its the same.

> Openssl 3.0.2 is relatively old. Please try updating to a more recent version and check if the issue remains. I remember having seen these log messages, too, mainly back in 2022.
Its true, but its the latest version from apt for ubuntu 22, but I will check to download the latest from github and try again.

ejoerns, 2024-11-13:
@frdve Have you been able to check this with a newer OpenSSL?

frdve, 2024-11-13:
Yes, I updated my ubuntu to 24.04 and then could get a newer version of OpenSSL from apt. That fixed it and the error lines are gone :⁠-⁠) 
Thanks!

closed 2024-11-14 by ejoerns