rauc/rauc #1529: Resign an already signed bundle of bundle format crypt fails

gambels, 2024-09-26:
**Describe the bug**
Currently it is not possible to resign an already signed bundle, if the bundle has been created with bundle format _crypt_. Resign a bundle of bundle format _verity_ works as expected.

**Background information**
Used RAUC version: v1.10.1
Platform: Ubuntu 22.04.5 LTS

**To Reproduce**
Resign bundle of bundle format crypt.

```sh
$ rauc resign \
    --conf system.conf \
    --keyring root-ca.cert.pem \
    --intermediate rauc-ca.cert.pem
    --cert rauc-sign.cert.pem \
    --key rauc-sign.key.pem \
    --no-verify \
    crypt-bundle.raucb crypt-bundle-resigned.raucb
```

**Expected behavior**
Resign an already signed bundle of bundle format _crypt_ is possible. Otherwise the bundle has to be created from scratch with new signing cert and key.

**Logs**
Resign bundle of bundle format crypt.

```
rauc-Message: 15:51:30.732: Using central status file /srv/rauc/central.raucs
rauc-Message: 15:51:30.732: Reading bundle: crypt-bundle.raucb
FATAL ERROR: Can't find a valid SQUASHFS superblock on /proc/3861076/fd/3
Failed to resign bundle: Failed to extract manifest from bundle: Failed to run unsquashfs: Child process exited with code 

```

**Additional context**
The _rauc-sign.cert.pem_ has been created with x509 extendedKeyUsage _codeSigning_. In this case, the RAUC _system.conf_ includes the following section.

```
[keyring]
check-purpose=codesign
```
The crypt bundle is still unencrypted.

```
Bundle Format:  crypt [unencrypted CMS]
```

ejoerns, 2024-09-26:
@gambels resign support for the newer 'crypt' format was added in RAUC v1.11 (with https://github.com/rauc/rauc/pull/1276).
If possible, update to the latest RAUC version.

gambels, 2024-09-27:
Many thanks for your feedback!

closed 2024-09-27 by gambels