rauc/rauc #1566: RAUC with efi boots from wrong slot and some strange entry is created

martin-petzold, 2024-11-24:
**Describe the bug**

After installing an image to slot B, it changes to this slot B and boots from B. But on a consecutive boot is falls back to slot A. Both slots are marked as good. There is a change in the boot entries which is not explainable to me.

Also some strange boot entry was added (by ???):

`Boot0002  UEFI OS	HD(2,GPT,d56615a6-227e-4d5e-ab31-f9b974c962ed,0x40000,0x40000)/File(\EFI\BOOT\BOOTX64.EFI)..BO`

Also on a boot (some time earlier, not directly related) I also had some warning (on a later call this was gone):

`(rauc:721): rauc-WARNING **: 21:55:56.567: Failed getting primary slot: efi backend: Regex matching failed!`

**Background information**

Debian 12
Linux 6.1.36
RAUC 1.8

/etc/rauc/system.conf

```
[system]
compatible=xyz
bootloader=efi
data-directory=/var/lib/rauc
bundle-formats=verity

[keyring]
path=/to/keyring.pem

[slot.efi.0]
device=/dev/sda1
type=vfat
parent=rootfs.0

[slot.efi.1]
device=/dev/sda2
type=vfat
parent=rootfs.1

[slot.rootfs.0]
device=/dev/sda3
type=ext4
bootname=A

[slot.rootfs.1]
device=/dev/sda4
type=ext4
bootname=B
```

**To Reproduce**

1. Install A/B setup with an efi partition each (like in the documentation [1])
2. Initial boot ->boots from A
3. Reboot -> boots from A
4. Reboot -> boots from A
5. Install RAUC image -> installs to B
6. Reboot -> boots from B
7. Reboot -> boots from A (this is wrong!)

[1] https://rauc.readthedocs.io/en/latest/integration.html#efi

The following shows how we add the boot entries initially after first flashing of the system. Initial flashing is done using a USB boot device. The first steps remove the existing entries. However, we do not add "--unicode ..." to our boot entries, because it did not seem to make sense. According to Debian documentation the boot entries should point to the efi partitions and there within the generated efi stub (bootx64.efi) there is the cmdline wrapped containing the root entry ("root=${BOOT_DEVICE} ..."). 

Maybe this could be clarified from your side? Are there "--unicode ..." parameters with "root=..." because you assume only vmlinuz and initrd.img files on the efi partitions and not a unified kernel image (bootx64.efi)? But why do you then use "...BZIMAGE.EFI" in the parameters, which is actually a unified kernel image then?

```
efibootmgr --verbose | sed -n 's/Boot\([0-9A-Z]\{4\}\).*/\1/gp' | sudo xargs -i efibootmgr --delete-bootnum --bootnum {} > /dev/null 2>&1

efibootmgr --create --disk /dev/sda --part 2 --label "B" --loader "\EFI\BOOT\bootx64.efi" > /dev/null 2>&1
efibootmgr --create --disk /dev/sda --part 1 --label "A" --loader "\EFI\BOOT\bootx64.efi" > /dev/null 2>&1

efibootmgr --timeout=0 > /dev/null 2>&1
```

**Expected behavior**

System should remain booting from slot B.

**Logs**

```
~$ rauc status
=== System Info ===
Compatible:  xyz
Variant:     
Booted from: rootfs.0 (/dev/sda3)

=== Bootloader ===
Activated: rootfs.0 (A)

=== Slot States ===
○ [rootfs.1] (/dev/sda4, ext4, inactive)
	bootname: B
	boot status: good
    [efi.1] (/dev/sda2, vfat, inactive)
	mounted: /boot/efi1

⏺ [rootfs.0] (/dev/sda3, ext4, booted)
	bootname: A
	mounted: /
	boot status: good
    [efi.0] (/dev/sda1, vfat, active)
	mounted: /boot/efi0

~$ efibootmgr --verbose
Timeout: 0 seconds
BootOrder: 0001,0000
Boot0000  B	HD(2,GPT,d56615a6-227e-4d5e-ab31-f9b974c962ed,0x40000,0x40000)/File(\EFI\BOOT\BOOTX64.EFI)
Boot0001  A	HD(1,GPT,d6b22123-3da0-4378-b95b-013696af46d9,0x22,0x3ffde)/File(\EFI\BOOT\BOOTX64.EFI)

~$ sudo reboot

~$ efibootmgr --verbose
Timeout: 0 seconds
BootOrder: 0001,0000
Boot0000  B	HD(2,GPT,d56615a6-227e-4d5e-ab31-f9b974c962ed,0x40000,0x40000)/File(\EFI\BOOT\BOOTX64.EFI)
Boot0001  A	HD(1,GPT,d6b22123-3da0-4378-b95b-013696af46d9,0x22,0x3ffde)/File(\EFI\BOOT\BOOTX64.EFI)

~$ rauc status
=== System Info ===
Compatible:  xyz
Variant:     
Booted from: rootfs.0 (/dev/sda3)

=== Bootloader ===
Activated: rootfs.0 (A)

=== Slot States ===
○ [rootfs.1] (/dev/sda4, ext4, inactive)
	bootname: B
	boot status: good
    [efi.1] (/dev/sda2, vfat, inactive)
	mounted: /boot/efi1

⏺ [rootfs.0] (/dev/sda3, ext4, booted)
	bootname: A
	mounted: /
	boot status: good
    [efi.0] (/dev/sda1, vfat, active)
	mounted: /boot/efi0

~$ rauc install xyz.raucb 
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
 40% Checking manifest contents
 60% Checking manifest contents done.
 60% Determining target install group
 80% Determining target install group done.
 80% Updating slots
 80% Checking slot rootfs.1
 90% Checking slot rootfs.1 done.
 90% Copying image to rootfs.1
 99% Copying image to rootfs.1 done.
 99% Updating slots done.
100% Installing done.
idle
Installing `/home/to/xyz.raucb` succeeded

~$ sudo reboot

~$ rauc status
=== System Info ===
Compatible:  xyz
Variant:     
Booted from: rootfs.1 (/dev/sda4)

=== Bootloader ===
Activated: rootfs.1 (B)

=== Slot States ===
⏺ [rootfs.1] (/dev/sda4, ext4, booted)
	bootname: B
	mounted: /
	boot status: good
    [efi.1] (/dev/sda2, vfat, active)
	mounted: /boot/efi1

○ [rootfs.0] (/dev/sda3, ext4, inactive)
	bootname: A
	boot status: good
    [efi.0] (/dev/sda1, vfat, inactive)
	mounted: /boot/efi0

~$ efibootmgr --verbose
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0001,0002
Boot0000  B	HD(2,GPT,d56615a6-227e-4d5e-ab31-f9b974c962ed,0x40000,0x40000)/File(\EFI\BOOT\BOOTX64.EFI)
Boot0001  A	HD(1,GPT,d6b22123-3da0-4378-b95b-013696af46d9,0x22,0x3ffde)/File(\EFI\BOOT\BOOTX64.EFI)
Boot0002  UEFI OS	HD(2,GPT,d56615a6-227e-4d5e-ab31-f9b974c962ed,0x40000,0x40000)/File(\EFI\BOOT\BOOTX64.EFI)..BO

~$ sudo reboot

~$ rauc status
=== System Info ===
Compatible:  xyz
Variant:     
Booted from: rootfs.0 (/dev/sda3)

=== Bootloader ===
Activated: rootfs.0 (A)

=== Slot States ===
○ [rootfs.1] (/dev/sda4, ext4, inactive)
	bootname: B
	boot status: good
    [efi.1] (/dev/sda2, vfat, inactive)
	mounted: /boot/efi1

⏺ [rootfs.0] (/dev/sda3, ext4, booted)
	bootname: A
	mounted: /
	boot status: good
    [efi.0] (/dev/sda1, vfat, active)
	mounted: /boot/efi0

~$ efibootmgr --verbose
Timeout: 0 seconds
BootOrder: 0001,0000
Boot0000  B	HD(2,GPT,d56615a6-227e-4d5e-ab31-f9b974c962ed,0x40000,0x40000)/File(\EFI\BOOT\BOOTX64.EFI)
Boot0001  A	HD(1,GPT,d6b22123-3da0-4378-b95b-013696af46d9,0x22,0x3ffde)/File(\EFI\BOOT\BOOTX64.EFI)
```

**Additional context**

I know how to use RAUC quite well. We have it in production with u-boot (arm64 processor, non-efi) without any problem. Now we have another platform with efi. Both setups are fully reproducible builds and the rootfs is identical (there may be some difference in kernel configuration).

martin-petzold, 2024-11-26:
I found some problem: After RAUC install, for some reason the labels of the partitions are changed. There are now also two partitions with same UUID. Maybe it is required to set the UUID via --unicode in order for RAUC to work as it should.

jluebbe, 2024-11-26:
For normal slot updates, RAUC will not touch the GPT, so the GPT UUIDs should not change. As the complete partition contents are replaced, you should not use filesystem UUIDs or labels.

martin-petzold, 2024-11-26:
Before RAUC upgrade (after initial write of whole image to device):

```
~$ lsblk --raw --noheadings --output PATH,LABEL,UUID
/dev/sda  
/dev/sda1 efi0 A9C5-FB67
/dev/sda2 efi1 A9D0-101F
/dev/sda3 rootfs0 0c7b121f-2b30-4714-830f-5accd98e5bfc
/dev/sda4 rootfs1 7c520d2d-c495-44b2-940b-8242fdf2ca90
/dev/sda5 system c33f98a9-3946-42ba-b7ac-96127f8f8e9a
/dev/sda6 user df2358f7-a290-4bf4-81bf-2564f9285fb6
/dev/sdb  
/dev/sdb1 efi C60A-B48D
/dev/sdb2 rootfs 89903d5a-265c-4a9a-b2f0-a5f5482ff8f1
```

After RAUC upgrade:

```
~$ lsblk --raw --noheadings --output PATH,LABEL,UUID
/dev/sda  
/dev/sda1 efi0 A9C5-FB67
/dev/sda2 efi1 A9D0-101F
/dev/sda3 rootfs0 0c7b121f-2b30-4714-830f-5accd98e5bfc
/dev/sda4 rootfs0 0c7b121f-2b30-4714-830f-5accd98e5bfc
/dev/sda5 system c33f98a9-3946-42ba-b7ac-96127f8f8e9a
/dev/sda6 user df2358f7-a290-4bf4-81bf-2564f9285fb6
/dev/sdb  
/dev/sdb1 efi C60A-B48D
/dev/sdb2 rootfs 89903d5a-265c-4a9a-b2f0-a5f5482ff8f1
```

martin-petzold, 2024-11-26:
Does RAUC update EFI boot manager entry with some UUID root entry?
Does RAUC actually touch the boot entries? Or only switch BootOrder and Boot Next?

martin-petzold, 2024-11-26:
Okay, maybe the problem is that I install the same RAUC image (RAUC update image to be installed in rootfs1 is the same as the originally installed image in rootfs0). However, this still does not explain why the label is changed.

jluebbe, 2024-11-26:
> After RAUC upgrade:
> 
> ```
> $ lsblk --raw --noheadings --output PATH,LABEL,UUID
> …
> /dev/sda3 rootfs0 0c7b121f-2b30-4714-830f-5accd98e5bfc
> /dev/sda4 rootfs0 0c7b121f-2b30-4714-830f-5accd98e5bfc
> …
> ```

This is expected, as `LABEL` and `UUID` are store *in the partition*. Accordingly they will change as you install new filesystem images. Instead, use labels and UUIDs stored *in the partition table* (`PARTLABEL`/`PARTUUID` in lsblk) to refer to partitions.

For example, on the LXA TAC:
```
root@lxatac-00007:~# lsblk -o PATH,LABEL,UUID,PARTLABEL,PARTUUID /dev/mmcblk1
PATH           LABEL UUID                                 PARTLABEL PARTUUID
/dev/mmcblk1                                                        
/dev/mmcblk1p1                                            reserved  4b8ad790-51c7-499e-a6fa-2b95c8507551
/dev/mmcblk1p2       12c42e2a-471d-4039-9046-31a9ecced0fa root-a    e82e6873-62cc-46fb-90f0-3e936743fa62
/dev/mmcblk1p3       12c42e2a-471d-4039-9046-31a9ecced0fa root-b    8eb3e87e-2b4e-45e3-888b-2f678662862d
/dev/mmcblk1p4 data  8ac00a5a-3255-4613-9119-dc58209566e8 data      491e0930-78b8-4da1-ba18-fb1fdc8b6515
```
The `system.conf` uses the partition UUIDs:
```
…
[slot.rootfs.0]
device=/dev/disk/by-partuuid/e82e6873-62cc-46fb-90f0-3e936743fa62
type=ext4
bootname=system0
resize=true

[slot.rootfs.1]
device=/dev/disk/by-partuuid/8eb3e87e-2b4e-45e3-888b-2f678662862d
type=ext4
bootname=system1
resize=true
…
```

Also see [this note](https://rauc.readthedocs.io/en/latest/integration.html#partitioning-your-device) for some suggestions.

The additional boot entries may be generated by our UEFI.

> Does RAUC update EFI boot manager entry with some UUID root entry?

No. It finds the entry using the boot entry label, which is set in the RAUC bootname. See the last paragraph in https://rauc.readthedocs.io/en/latest/integration.html#efi.

> Does RAUC actually touch the boot entries? Or only switch BootOrder and Boot Next?

It only sets BootOrder and BootNext (using `efibootmgr --bootorder` and `--bootnext`).

ejoerns, 2024-11-26:
> Also on a boot (some time earlier, not directly related) I also had some warning (on a later call this was gone):
> 
> `(rauc:721): rauc-WARNING **: 21:55:56.567: Failed getting primary slot: efi backend: Regex matching failed!`

As your RAUC version is quite old, this might be fixed by https://github.com/rauc/rauc/pull/1197 already. Please consider using a recent RAUC version for development.

martin-petzold, 2024-11-26:
Okay, I understand. This was my fault with the UUIDs. However, as I don't use the UUIDs for the cmdline within the unified efi image, this should not be the problem.

Still some open questions:
1. Do I have to use --unicode with related root partition for the bootmgr entries for RAUC to work? Or should it also work with root entry within cmdline in the unified image?
2. For some reason the boot entries are behaving strange. I did not post, but even in between there was only one boot entry (A) left and I did not touch it. Is it possible that the efi boot manager behaves strange?
3. Maybe would you strongly advise to use (partition) UUIDs for the boot manager entries, as stated in the documentation?