rauc/rauc #1612: g_close(fd:9) failed with EBADF. The tracking of file descriptors got messed up

capiman, 2025-01-28:
I have RAUC 1.12 in a yocto environment. Board is a Phytec Pollux with an i.MX8MP.
Having system A/B on eMMC. No SD card.

Before:

```
# rauc status
=== System Info ===
Compatible:  test-phyboard-pollux-imx8mp-3
Variant:
Booted from: rootfs.0 (system0)
 
=== Bootloader ===
Activated: rootfs.0 (system0)
 
=== Slot States ===
  [bootloader.0] (/dev/mmcblk2, boot-emmc, inactive)
 
o [rootfs.1] (/dev/mmcblk2p6, ext4, inactive)
        bootname: system1
        boot status: good
    [boot.1] (/dev/mmcblk2p2, vfat, inactive)
 
x [rootfs.0] (/dev/mmcblk2p5, ext4, booted)
        bootname: system0
        mounted: /
        boot status: good
    [boot.0] (/dev/mmcblk2p1, vfat, active)
 
 
#
```

```
# rauc info /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb
rauc-Message: 21:23:25.895: Using central status file /mnt/config/rauc/central.raucs
rauc-Message: 21:23:25.895: valid /etc/rauc/system.conf found, using it
rauc-Message: 21:23:25.895: Reading bundle: /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb
rauc-Message: 21:23:25.901: Verifying bundle signature...
rauc-Message: 21:23:25.906: Verified inline signature by 'O = SW, CN = PHYTEC Messtechnik GmbH Development-1'
Compatible:     'test-phyboard-pollux-imx8mp-3'
Version:        'r0'
Description:    'PHYTEC rauc bundle based on BSP-Yocto-NXP-i.MX8MP-PD24.1.0-devel'
Build:          '20250128183516'
Hooks:          ''
Bundle Format:  verity
  Verity Salt:  '741186010c29ad7407f3176312d6f5e99ff663ff920ed76932e71909e8cbd743'
  Verity Hash:  '869b452375835ebca3e79e1eef6ec2345636018514aefa588977c239172e3bc3'
  Verity Size:  724992
Manifest Hash:  '7db3c167bd0091277450f2aa0ec9ab3c3be492ff26d13f987b08f37f4355d546'
 
 
3 Images:
  [rootfs]
        Filename:  test-headless-image-test-phyboard-pollux-imx8mp-3.rootfs.ext4
        Checksum:  3f236a3288c97b8274f3f156910043b14087830f71e2240a6624f87455ebe3d1
        Size:      358.0?MB (357998592 bytes)
        Hooks:
        Adaptive:  block-hash-index
  [boot]
        Filename:  boot.tar.gz
        Checksum:  4dada7d751066fd68702fe430b59165dbae19a6a44b46f21c2ddb447d6a6414a
        Size:      13.4?MB (13380826 bytes)
        Hooks:
  [bootloader]
        Filename:  emmc-boot.img
        Checksum:  d8eeb871272c0317d0cff5b274e9627bd94ec5f597c4a46fcbd35d62f76e7105
        Size:      1.7?MB (1685360 bytes)
        Hooks:
 
Certificate Chain:
...if needed/important for bug report, please let me know...
#
```

I call 

```
rauc status -d &
rauc install -d <filename.raucb>

# rauc install -d /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb
rauc-Message: 21:28:21.336: Debug log domains: 'rauc'
(rauc:915): rauc-DEBUG: 21:28:21.336: 1.12 create=0 emmc-boot=1 gpt=1 json=1 network=1 service=1 streaming=1
(rauc:915): rauc-DEBUG: 21:28:21.339: install started
(rauc:915): rauc-DEBUG: 21:28:21.339: input bundle: /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb
(rauc:915): rauc-DEBUG: 21:28:21.348: Trying to contact rauc service
installing
  0% Installing
  0% Determining slot states
10% Determining slot states done.
10% Checking bundle
10% Verifying signature
20% Verifying signature done.
20% Checking bundle done.
20% Checking manifest contents
30% Checking manifest contents done.
30% Determining target install group
40% Determining target install group done.
40% Updating slots
40% Checking slot rootfs.1
42% Checking slot rootfs.1 done.
42% Copying image to rootfs.1
RAUC service disappeared
D-Bus error while installing `/media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb`
root@test-phyboard-pollux-imx8mp-3:~#
```

journalctl -xe

tells

```
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: input bundle: /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Active slot bootname: system0
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: installing /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb: started
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: [🡕] Installation 6f7b89da started
░░ Subject: RAUC Installation started
░░ Defined-By: rauc
░░ Support: https://rauc.io/
░░ Documentation: https://rauc.readthedocs.io/
░░
░░ The installation with transaction ID 6f7b89da-0d13-47ea-b26a-db65a40488f1 started.
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: installing /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb: Checking and mounting bundle...
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Reading bundle: /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Verifying bundle signature...
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Verified inline signature by 'O = SW, CN = PHYTEC Messtechnik GmbH Development-1'
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Mounting bundle '/media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb' to '/run/rauc/bundle'
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 kernel: loop0: detected capacity change from 0 to 179583
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 kernel: loop0: detected capacity change from 179583 to 179576
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Configured loop device '/dev/loop0' for 91942912 bytes
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Configured dm-verity device '/dev/dm-0'
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 kernel: device-mapper: verity: sha256 using implementation "sha256-caam"
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Starting pre install handler: /usr/lib/rauc/rauc-pre-install.sh
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Checking image type for slot type: ext4
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Image detected as type: *.ext4
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Checking image type for slot type: vfat
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Image detected as type: *.tar*
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Checking image type for slot type: boot-emmc
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Image detected as type: *.img
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Marking target slot rootfs.1 as non-bootable...
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: [🡕] Marked slot rootfs.1 as bad
░░ Subject: RAUC slot rootfs.1 was marked as 'bad'
░░ Defined-By: rauc
░░ Support: https://rauc.io/
░░ Documentation: https://rauc.readthedocs.io/en/latest/using.html#react-to-a-successfully-booted-system-failed-boot
░░
░░ The RAUC slot 'rootfs.1' was marked as 'bad'.
░░
░░ This informs the bootloader to not boot the entry for 'system1' anymore.
░░ Where this information is persisted, mainly depends on the bootloader used
░░ (configured in your RAUC system config).
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: installing /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb: Updating slots...
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: installing /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb: Checking slot rootfs.1
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Updating slot rootfs.1
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: installing /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb: Updating slot rootfs.1
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: Updating /dev/mmcblk2p6 with /run/rauc/bundle/test-headless-image-test-phyboard-pollux-imx8mp-3.rootfs.ext4
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: g_close(fd:9) failed with EBADF. The tracking of file descriptors got messed up
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 systemd[1]: rauc.service: Main process exited, code=dumped, status=5/TRAP
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░
░░ An ExecStart= process belonging to unit rauc.service has exited.
░░
░░ The process' exit code is 'dumped' and its exit status is 5.
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 systemd[1]: rauc.service: Failed with result 'core-dump'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░
░░ The unit rauc.service has entered the 'failed' state with result 'core-dump'.
```

It is 100% reproduceable on 2 boards with same software.

I assume this is the relevant error message from journalctl -xe:

```
Jan 28 21:28:21 test-phyboard-pollux-imx8mp-3 rauc[813]: g_close(fd:9) failed with EBADF. The tracking of file descriptors got messed up
```

ejoerns, 2025-01-28:
@capiman Thanks for reporting! If you have the chance to restart the `rauc service` with `-d`, too, this could provide some more debug output for the installation flow.

capiman, 2025-01-28:
Sorry, I thought I had also added it. Here a complete run:

```
root@test-phyboard-pollux-imx8mp-3:~# rauc service -d &
[1] 2139
root@test-phyboard-pollux-imx8mp-3:~# rauc-Message: 22:15:46.927: Debug log domains: 'rauc'
(rauc:2139): rauc-DEBUG: 22:15:46.928: 1.12 create=0 emmc-boot=1 gpt=1 json=1 network=1 service=1 streaming=1
(rauc:2139): rauc-DEBUG: 22:15:46.931: No value for key "max-bundle-download-size" in [system] defined - using default value of 8388608 bytes.
rauc-Message: 22:15:46.931: Using central status file /mnt/config/rauc/central.raucs
(rauc:2139): rauc-DEBUG: 22:15:46.931: Obtained system boot ID: 'f7544ba9-0a3d-4e50-8a43-a05eb70a6497'
(rauc:2139): rauc-DEBUG: 22:15:46.931: Obtained system machine ID: '5f3566ef18e84d989869e241bb092677'
(rauc:2139): rauc-DEBUG: 22:15:46.931: service start
(rauc:2139): rauc-DEBUG: 22:15:46.931: Found booted slot: rootfs.0 on /dev/mmcblk2p5
rauc-event-Message: 22:15:46.932: Service restarted
(rauc:2139): rauc-DEBUG: 22:15:46.938: name 'de.pengutronix.rauc' acquired
 
root@test-phyboard-pollux-imx8mp-3:~# rauc install -d /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb
rauc-Message: 22:16:40.420: Debug log domains: 'rauc'
(rauc:2164): rauc-DEBUG: 22:16:40.421: 1.12 create=0 emmc-boot=1 gpt=1 json=1 network=1 service=1 streaming=1
(rauc:2164): rauc-DEBUG: 22:16:40.423: install started
(rauc:2164): rauc-DEBUG: 22:16:40.423: input bundle: /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb
(rauc:2164): rauc-DEBUG: 22:16:40.432: Trying to contact rauc service
input bundle: /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb
rauc-Message: 22:16:40.434: Active slot bootname: system0
(rauc:2139): rauc-DEBUG: 22:16:40.434: thread started for /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb
rauc-Message: 22:16:40.435: installing /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb: started
rauc-event-Message: 22:16:40.435: Installation 7dd9c9a7 started
installing
(rauc:2139): rauc-DEBUG: 22:16:40.436: Found external mountpoint for slot rootfs.0 at /
  0% Installing
  0% Determining slot states
rauc-Message: 22:16:40.437: installing /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb: Checking and mounting bundle...
rauc-Message: 22:16:40.437: Reading bundle: /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb
10% Determining slot states done.
10% Checking bundle
(rauc:2139): rauc-DEBUG: 22:16:40.443: Found valid CMS data
rauc-Message: 22:16:40.444: Verifying bundle signature...
10% Verifying signature
rauc-Message: 22:16:40.449: Verified inline signature by 'O = SW, CN = PHYTEC Messtechnik GmbH Development-1'
20% Verifying signature done.
(rauc:2139): rauc-DEBUG: 22:16:40.451: Got 3 elements for trust chain
rauc-Message: 22:16:40.451: Mounting bundle '/media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb' to '/mnt/rauc/bundle'
20% Checking bundle done.
rauc-Message: 22:16:40.501: Configured loop device '/dev/loop0' for 91942912 bytes
20% Checking manifest contents
30% Checking manifest contents done.
rauc-Message: 22:16:40.512: Configured dm-verity device '/dev/dm-0'
(rauc:2139): rauc-DEBUG: 22:16:40.516: Checking slot: rootfs.1
(rauc:2139): rauc-DEBUG: 22:16:40.516:  Adding mapping: rootfs -> rootfs.1
(rauc:2139): rauc-DEBUG: 22:16:40.516: Checking slot: boot.0
(rauc:2139): rauc-DEBUG: 22:16:40.516:  No mapping found
(rauc:2139): rauc-DEBUG: 22:16:40.516: Checking slot: bootloader.0
(rauc:2139): rauc-DEBUG: 22:16:40.516:  Adding mapping: bootloader -> bootloader.0
(rauc:2139): rauc-DEBUG: 22:16:40.516: Checking slot: rootfs.0
(rauc:2139): rauc-DEBUG: 22:16:40.516:  No mapping found
(rauc:2139): rauc-DEBUG: 22:16:40.516: Checking slot: boot.1
(rauc:2139): rauc-DEBUG: 22:16:40.516:  Adding mapping: boot -> boot.1
rauc-Message: 22:16:40.516: Starting pre install handler: /usr/lib/rauc/rauc-pre-install.sh
30% Determining target install group
40% Determining target install group done.
(rauc:2139): rauc-DEBUG: 22:16:40.591: No min-bundle-version configured in the system.conf. Version-limit check does nothing.
(rauc:2139): rauc-DEBUG: 22:16:40.591: Using default installation handler
(rauc:2139): rauc-DEBUG: 22:16:40.591: Found image mapping: test-headless-image-test-phyboard-pollux-imx8mp-3.rootfs.ext4 -> rootfs
rauc-Message: 22:16:40.591: Checking image type for slot type: ext4
rauc-Message: 22:16:40.591: Image detected as type: *.ext4
(rauc:2139): rauc-DEBUG: 22:16:40.591: Found image mapping: boot.tar.gz -> boot
rauc-Message: 22:16:40.591: Checking image type for slot type: vfat
rauc-Message: 22:16:40.591: Image detected as type: *.tar*
(rauc:2139): rauc-DEBUG: 22:16:40.591: Found image mapping: emmc-boot.img -> bootloader
rauc-Message: 22:16:40.591: Checking image type for slot type: boot-emmc
[... 1 lines cut ...]
rauc-Message: 22:16:40.592: Marking target slot rootfs.1 as non-bootable...
rauc-event-Message: 22:16:40.608: Marked slot rootfs.1 as bad
rauc-Message: 22:16:40.609: installing /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb: Updating slots...
rauc-Message: 22:16:40.609: installing /media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb: Checking slot rootfs.1
(rauc:2139): rauc-DEBUG: 22:16:40.609: Load status for slot rootfs.1.
[... cut ...]
rauc-INFO: 22:16:40.615: Selected adaptive update method 'block-hash-index'
 
(rauc:2139): GLib-CRITICAL **: 22:16:40.615: g_close(fd:8) failed with EBADF. The tracking of file descriptors got messed up
42% Checking slot rootfs.1 done.
42% Copying image to rootfs.1
RAUC service disappeared
D-Bus error while installing `/media/usb/test-headless-bundle-test-phyboard-pollux-imx8mp-3-20250128183516.raucb`
[1]+  Trace/breakpoint trap   (core dumped) rauc service -d
root@test-phyboard-pollux-imx8mp-3:~#
```

capiman, 2025-01-28:
glib seems to be glib-2.0_2.78.6.bb (Yocto Recipe name)

ejoerns, 2025-01-28:
@capiman Thank you! I've actually overlooked the `Adaptive: ` in the rauc info output. Together with your hint of the glib version I'd say this is what was fixed by https://github.com/rauc/rauc/pull/1509.

Could you test if RAUC 1.13 is working for you?

If I find time I should also do a meta-rauc backport for scarthgap soon.

capiman, 2025-01-28:
I think it is possible to use RAUC 1.13 instead. I will try tomorrow and let you know if this fixes the problem.

ejoerns, 2025-02-04:
@capiman Any news on this? Otherwise, I would close this issue since I'd assume it's fixed.

capiman, 2025-02-04:
@ejoerns Thanks for your help! We meanwhile updated to RAUC 1.13 and the error disappeared.
I think we can close this entry.

ejoerns, 2025-02-04:
@capiman Thank you for the update 👍

closed 2025-02-04 by capiman