rauc/rauc #1617: Bundle post-install hook wrong argument given when calling hook script

arapaill, 2025-02-04:
**Describe the bug**
Hello,

I want to execute some commands after installing a bundle on the slot it has been installed to. However, when I install the bundle, the expected argument "slot-post-install" is not sent to the script. Instead, the script receives two unexpected paths as arguments:

-  /home/ubuntu/buildroot/output/images
-  /home/ubuntu/buildroot/output/images/imx8mn-var-som-symphony-uboot.dtb

**Background information**
RAUC version: 1.11.3
Build system: Buildroot 2024.02.6

**To Reproduce**
Steps to reproduce the behavior:
1. Use the following manifest:
```
[update]
compatible=red-03
version=3.0.0a

[bundle]
format=verity

[hooks]
filename=hook

[image.rootfs]
sha256=363a147ded71d0f7bd40128cd11ba4708f271de1b79958cdfef612796d620581
size=1073741824
filename=rootfs.ext4
hooks=post-install;
adaptive=block-hash-index;
```

2. hook script (hook):
```
#!/bin/sh

echo "Hook start" >> /var/log/firmware-update.log
echo "TIME: Tue Feb  4 08:14:12 UTC 2025" >> /var/log/firmware-update.log
echo "Arguments: /home/ubuntu/red-buildroot/output/images /home/ubuntu/red-buildroot/output/images/imx8mn-var-som-symphony-uboot.dtb" >> /var/log/firmware-update.log

case "/home/ubuntu/red-buildroot/output/images" in
    slot-post-install)
        echo "Slot post-install hook" >> /var/log/firmware-update.log
        ;;
    *)
        echo "Unknown hook argument: /home/ubuntu/red-buildroot/output/images" >> /var/log/firmware-update.log
        exit 1
        ;;
esac

echo "Hook done" >> /var/log/firmware-update.log
exit 0
```

3. We use Buildroot on an EC2 instance, and once built, the bundle is uploaded to AWS S3, where our devices download it via URL.

**Expected behavior**
The hook script should receive the expected argument slot-post-install, but instead, it is receiving paths from the EC2 instance where the bundle was built.

**Logs**
Here is the output of /var/log/firmware-update.log:
```
Hook start
TIME: Tue Feb  4 08:14:12 UTC 2025
Arguments: /home/ubuntu/buildroot/output/images /home/ubuntu/buildroot/output/images/imx8mn-var-som-symphony-uboot.dtb
Unknown hook argument: /home/ubuntu/buildroot/output/images
````
**Additional context**
If you need any additional information, please let me know.

Thank you for your time,
Alexandre Rapaille @ Insens

ejoerns, 2025-02-04:
@arapaill Can you `rauc mount` the bundle and check the files contained for these paths?

arapaill, 2025-02-04:
Thank you for replying, those files are not in the bundle:
```
Mounted bundle at /run/rauc/bundle. Use 'umount /run/rauc/bundle' to unmount.
$ cd /run/rauc/
$ ls
bundle  slots
$ cd bundle/
$ ls
hook                          manifest.raucm                rootfs.ext4                   rootfs.ext4.block-hash-index
$
```

ejoerns, 2025-02-04:
@arapaill I might have been imprecise here. I meant to check the file contents. Somehow the host paths must find their way into the target. I cannot reproduce this behavior in my test setup. Neither with 1.12.3 nor with a more recent rauc version (1.13).

ejoerns, 2025-02-04:
@arapaill Glad I could help! (And that it's not a RAUC issue 😉)

closed 2025-02-04 by ejoerns