rauc/rauc #1649: Intermediate CA CRL cannot be passed with an update bundle.

Emmmery, 2025-03-03:
This issue is related to the scenario where we have a standalone 2-tier RAUC PKI :

- An offline-managed Root CodeSigning CA serves as the trust anchor for product instances.
- An online-managed Sub CodeSigning CA issues and revokes developer code signing certificates (leafs).

Enabling CRL verification (check-crl=true in config) requires, at the time of installing a bundle, the CRLs of the Root CA and the Sub CA.

However, for CRLs, RAUC does not seem to look in the PEM file passed with --intermediate at the time of bundle generation. The only way seems to be to have the CRL of the Sub CA already present in the target's keyring; otherwise, we get an error "unable to get CRL...".

This behavior seems very limiting because, in the case where the Sub CA is lost (loss of keys), we find ourselves unable to install an update signed with a key (leaf) certified with a new Sub CA. The latter will not be accepted by the target, which does not possess a CRL for the new Sub CA in its keyring.

**Workaround**
As a workaround, we proceed as follows:

Offline backup of the Sub CA keys.
In the event that the Sub CA is compromised, publish and install an update signed with the compromised CA that:
updates the CRL of the root CA in the keyring (CRL that revokes the compromised Sub CA)
adds the CRL of a new Sub CA to the keyring

The expected behavior would be for RAUC to fetch the CRL of the intermediate CA passed with --intermediate in the same PEM file provided with --intermediate. This would allow directly passing an update signed with keys certified by a new, intact Sub CA if the Sub CA has been compromised. The offline-managed Root CA could then properly play its role as the sole trust anchor.

Does this issue seem relevant to you?"

jluebbe, 2025-03-03:
Our original expectation regarding the CRL support was that a separate mechanism would regularly fetch a fresh CRL (e.g. via the network) and make it available to RAUC. As the actual process for that is very project specific, it was not implemented in RAUC itself.

I understand that you'd like to include the CRLs in the bundles' CMS structure itself. While this might require re-signing older bundles if the included CRLs have expired, this would be a nice way to avoid the need for a separate CRL refresh mechanism. Until I searched for it in the context of this issue, I wasn't ware that including CRLs in CMS was possible. It seems to be supported in [OpenSSL via the `CMS_add1_crl()` function](https://docs.openssl.org/master/man3/CMS_add0_cert/).

I've not looked deeper how this could be used from RAUC. As there has not been community or commercial demand for this feature so far, so it's not on our roadmap yet.

If you want to take a shot at implementing it, here a some things to keep in mind:

- A user might want to include a CRL for the Root CA, so loading CRLs only from `--intermediate` wouldn't be enough. Perhaps adding a new `--crl` option would be better. It looks like the CMS structure doesn't have a fixed association between additional certificates and CRLs, so that should be fine.
- RAUC uses OpenSSL's [`CMS_sign()`](https://docs.openssl.org/master/man3/CMS_sign/) in [`src/signature.c`](https://github.com/rauc/rauc/blob/master/src/signature.c#L565). `CMS_sign()` supports additional certificates via the `certs` argument, but there is no obvious way to pass CRLs. Perhaps they can just be added to the `CMS_ContentInfo` after signing?
- It's not clear to me how a CRL for the Root CA in the CMS and another one in the local keyring would interact in OpenSSL during verification. Which one would be preferred?

With your current workaround, which validity periods do you use for your CRLs?

Emmmery, 2025-03-04:
Thank you for your quick response and all these judicious elements that we will take into account.

_"Our original expectation regarding the CRL support was that a separate mechanism would regularly fetch a fresh CRL (e.g. via the network)"_
In our somewhat unique use case, we need to meet a requirement that our devices must be able to be maintained offline. Hence the need to pass the CRLs with the bundles.

_With your current workaround, which validity periods do you use for your CRLs?_
Indeed, this is also our problem here. We are forced to have an (overly) long validity period for our CRLs (several months for the Sub CA CRL).
We are therefore obliged to leave too large a margin compared to the update cycle to avoid bricking the device if the operator does not apply the update in time. This creates too large a margin for the possibility of rollback and compromise if there is a key leak.

jluebbe, 2025-03-04:
In many of the use-cases I know about, using a single (self-signed) long-lived signing key and storing it in a HSM for offline singing (e.g. via ``rauc resign``) was deemed secure enough. In the (unlikely) case of a compromise, that key would be replaced by an update. The change discussed in https://github.com/openssl/openssl/issues/26382 is intended avoid the need for multiple bundles in that case.

It seems that wouldn't be a solution for you though. If you want to implement it yourself and would like to discuss implementation details, I'd suggest joining our [Matrix](https://matrix.to/#/#rauc:matrix.org) channel.