rauc/rauc #1675: Improvement suggestion to RAUC's detection of the booted slot

oli-ben, 2025-03-27:
**Is your feature request related to a problem? Please describe.**
@jluebbe @a3f I just ran into a small issue with RAUC, where I was unable to mark the booted slot as `good`.

Looking at the logs, RAUC is actually throwing on an assumption, that the booted slot has a `bootname`.

The reason is d, but noticed that RAUC still disagreed as to which was the `booted` slot.oesn't is, that RAUC actually thinks that my rootFS is the booted slot, even though:
* the rootFS has a `parent` in `system.conf`
* I have implemented `get-current` in my custom bootloader backend for the RPi firmware which points to the parent

Why does RAUC disagree? By reading the doc, I concluded that it was because of the `root=` argument on the kernel command line.
My problem here is, that I cannot get rid of that argument, or the kernel won't know where to find my rootFS, but that should not be considered the booted slot.

**Describe the solution you'd like**
Here is the behaviour I would like, because that's the one I thought RAUC would implement initially:

* If `rauc,slot=` is not specified on the kernel command line and a custom backend is in use, that should be the next source of truth, not `root=`

**Describe alternatives you've considered**

* Alternatively, RAUC might consider walking *up* the tree of parents  when the `booted` slot does not have a `bootname`. This would also have solved my issue.

This seems like a more adventurous feature, and I can see that this slightly shifts the definition of what a "booted" slot is

ejoerns, 2025-03-28:
> * If rauc,slot= is not specified on the kernel command line and a custom backend is in use, that should be the next source of truth, not root=

Isn't this what #1669 addresses?

oli-ben, 2025-03-28:
It is indeed, Gael is just faster than me every time. Sorry, I should have checked open issues before opening a new one!

closed 2025-03-28 by oli-ben