rauc/rauc #1684: Crash when mounting a corrupt bundle

graemi, 2025-04-02:
An attempt to mount a corrupt verity bundle results in a crash of rauc:

Steps to reproduce:

1. Create a bundle in verity format. 
2. Manually delete a few bytes in the bundle payload with an editor, so that payload size is no longer a multiple of 4KiB.
3. Attempt to mount the bundle.

Expected: Mount fails with an error message, rauc exits with error.
Observed: Crash in rauc
```
$  rauc mount /path/to/bundle.raucb
(rauc:976): rauc-DEBUG: 11:43:40.053: No value for key "max-bundle-download-size" in [system] defined - using default value of 8388608 bytes.
(rauc:976): rauc-DEBUG: 11:43:40.055: No mount prefix provided, using /mnt/rauc/ as default
rauc-Message: 11:43:40.056: No data directory or status file set, falling back to per-slot status.
Consider setting 'data-directory=<path>' or 'statusfile=<path>/per-slot' explicitly.
rauc-Message: 11:43:40.056: Using per-slot statusfile. System status information not supported!
rauc-Message: 11:43:40.058: Using system config file /etc/rauc/system.conf
(rauc:976): rauc-DEBUG: 11:43:40.059: Obtained system boot ID: 'f7a11aeb-e8e0-446d-b39f-7c22c1840440'
(rauc:976): rauc-DEBUG: 11:43:40.059: Obtained system machine ID: '8e0859162b3b434ab6bcee067e7c1f23'
(rauc:976): rauc-DEBUG: 11:43:40.059: input bundle: /path/to/bundle.raucb
rauc-Message: 11:43:40.060: Reading bundle: /path/to/bundle.raucb
rauc-Message: 11:43:40.064: Payload size (259505939) is not a multiple of 4KiB. See https://rauc.readthedocs.io/en/latest/faq.html#what-causes-a-payload-size-that-is-not-a-multiple-of-4kib
(rauc:976): rauc-DEBUG: 11:43:40.079: Found valid CMS data
rauc-Message: 11:43:40.083: Verifying bundle signature...
rauc-Message: 11:43:40.093: Verified inline signature by '[manually removed]'
(rauc:976): rauc-DEBUG: 11:43:40.096: Got 2 elements for trust chain
(rauc:976): rauc-DEBUG: 11:43:40.097: bundle payload size: 259505939
rauc-Message: 11:43:40.098: Mounting bundle '/path/to/bundle.raucb'
rauc-Message: 11:43:40.174: Configured loop device '/dev/loop0' for 259505939 bytes
(rauc:976): rauc-CRITICAL **: 11:43:40.176: r_dm_setup: assertion 'dm->data_size > 0 && dm->data_size % 4096 == 0' failed
Trace/breakpoint trap
$
```

Observed with rauc 1.13 on Debian trixie (package version 1.13-2)

It looks like a bug in the error handling when the assertion error is propagated up to the main function.

ejoerns, 2025-04-02:
@graemi I guess this corner case is not checked at the moment.

Normally, one has a modified bundle (this would cause verification issues) or a bundle that lacks the last bytes (e.g. due to incomplete transmissions).

But removing some bytes in the middle of the bundle does not happen that often. Did you face this issue during explicit tests or did that happen in normal test scenarios?

graemi, 2025-04-02:
@ejoerns Thanks for quick feedback.

This happened during explicit test, not under normal test scenario.

I admit that it is a corner case, but it is caught by an assertion in function r_dm_setup():

`g_return_val_if_fail(dm->data_size > 0 && dm->data_size % 4096 == 0, FALSE);
`

The problem is not that the case is not checked, but that the error from the assertion is not properly propagated.

ejoerns, 2025-04-02:
> This happened during explicit test, not under normal test scenario.

Okay, thank you for the clarification!
I agree it should be handled cleanly anyway.

> I admit that it is a corner case, but it is caught by an assertion in function r_dm_setup():
> 
> `g_return_val_if_fail(dm->data_size > 0 && dm->data_size % 4096 == 0, FALSE); `
> 
> The problem is not that the case is not checked, but that the error from the assertion is not properly propagated.

These `g_return_val_*()` checks are not really meant to be propagated. They are more like sanity checks.
The issue here is that the input to `r_dm_setup()` is not checked before but the method assumes they must be valid.

We could either make it a GError-based check in `r_dm_setup()` (and replace the `g_return_val_if_fail()` or actually check it before calling `r_dm_setup()`.

graemi, 2025-04-02:
> These g_return_val_*() checks are not really meant to be propagated. They are more like sanity checks.
The issue here is that the input to r_dm_setup() is not checked before but the method assumes they must be valid.

Yes, makes sense.

> We could either make it a GError-based check in r_dm_setup() (and replace the g_return_val_if_fail() or actually check it before calling r_dm_setup().

Which solution would you prefer? Maybe I can prepare a PR.

ejoerns, 2025-04-02:
@graemi What I did not have in mind is that we actually check the size already, but due to compatibility issues with order casync+plain bundles, we only emit a message here. This is the one you can see in your logs, too:

> rauc-Message: 11:43:40.064: Payload size (259505939) is not a multiple of 4KiB. See https://rauc.readthedocs.io/en/latest/faq.html#what-causes-a-payload-size-that-is-not-a-multiple-of-4kib

For verity bundles, we should however handle this as suggested. Our favorite approach would be to just make the explicit GError-check within `r_dm_setup()` since this is the single code path for all cases.

commit c966aefd (2025-06-06): src/dm: replace data size assertion by gerror failure

When using a modified raucb bundle with payload that is
not a multiple of 4096, the following assertion happens:

"rauc-CRITICAL **: r_dm_setup: assertion 'dm->data_size > 0 && dm->data_size % 4096 == 0' failed"

Replace the assertion with a proper gerror failure.

Fixes: #1684
Reported-by: Tobias Graemer <tobias.graemer@mt.com>
closed 2025-06-06 by jluebbe