rauc/rauc #1688: OpenSSL 3.0.0 has deprecated the ENGINE API

baxeno, 2025-04-03:
OpenSSL 3.0.0 was released Sep 21, 2024 and it has deprecated the engine API.
`#include <openssl/engine.h>` is used in `src/signature.c`

Release note snippets:

- Deprecated the ENGINE API.
- Added providers, a new pluggability concept that will replace the ENGINE API and ENGINE implementations.

Suggested updates:

- Make get_pkcs11_engine optional behind a compile flag.
- Migrate from engine to providers.

References:

- https://github.com/openssl/openssl/releases/tag/openssl-3.0.0
- https://docs.openssl.org/3.0/man1/openssl-engine/
- https://docs.openssl.org/3.0/man7/migration_guide/#providers-are-a-replacement-for-engines-and-low-level-method-overrides
- https://www.redhat.com/en/blog/openssl-3-providers-rhel-10
- https://blog.hansenpartnership.com/converting-engines-to-openssl-3-providers/

jluebbe, 2025-04-04:
OpenSSL 3.0.0 was actually released Sep 7, 2021 (the GitHub release page was created much later). For us the main blocker was [support for PKCS#11](https://github.com/latchset/pkcs11-provider) via the providers API, which has only recently reached 1.0. That version is not yet part of stable releases for widely used distributions, so we can't rely on it yet.

As I understand it, the engine API will only be removed with OpenSSL 4.0, which is not under development yet.

So, for now, we've preferred to stay on the engine API to keep the testing effort low. We'd probably want remove engine support and switch the test suite to providers for a release later this year, though, when pkcs11-provider is more widely available. If you have some time to work on this, a PR  would be very appreciated (either making it optional at compile-time if that can be done cleanly or otherwise removing it). :)

Do you have a specific reason to prefer the providers API (beyond the deprecation notices)?

baxeno, 2025-04-05:
I have been working on adding rauc package to Fedora, since I use rauc (outside of ptxdist) in a firmware signing setup where the developer signature is replaced with an official signature before field deployment. In F41 (release 2024H2) a change proposal was accepted that would _prevent creating new packages dependent on OpenSSL ENGINE API and remove ENGINE dependencies from the existing packages_.

In the latest rauc RPM spec I have added a quick-and-dirty patch that `#ifdef` remove the engine support:
https://github.com/baxeno/fedora-copr-spec/blob/main/rauc_no_openssl_engine.patch

I can try and improve it with proper Meson integration and send a PR.

Fedora 41 change proposal: OpenSSL deprecate engine (system wide):
https://discussion.fedoraproject.org/t/f41-change-proposal-openssl-deprecate-engine-system-wide/111344

Fedora Copr build system:
https://copr.fedorainfracloud.org/coprs/bax/rauc/

Fedora package review request:
https://bugzilla.redhat.com/show_bug.cgi?id=2356980

On the roadmap OpenSSL 4.0 is expected around April 2026:
https://openssl-library.org/roadmap/index.html

PR #1690 (2025-04-16): meson: add support for building without OpenSSL engine support
Adding -DDISABLE_OPENSSL_ENGINE will disable engine support and remove openssl-devel-engine dependency on Fedora.
OpenSSL engine support will also be removed in 4.0 in favour of providers API.
I have tested an earlier version of the patch in Fedora Copr RPM build.
https://copr.fedorainfracloud.org/coprs/bax/rauc/build/8859137/

Fixes: #1688
closed 2025-04-16 by jluebbe