rauc/rauc #1701: Support (re)creating UEFI boot entries

jluebbe, 2025-04-17:
**Is your feature request related to a problem? Please describe.**
When using a setup like the one suggested by @hnez in https://github.com/rauc/rauc/pull/1696#issuecomment-2809369764, some UEFI
implementations tend to remove boot entries if they point to unmountable
filesystems (which can happen due to a power cycle while installing).

**Describe the solution you'd like**
We should add the missing information to the slot sections in the `system.conf` so that RAUC can recreate entries if they are missing.

```
[slot.esp.0]
device=/dev/sda1
bootname=system0
efi-loader=\\EFI\\BOOT\\BOOTX64.EFI
efi-unicode="@1" # UKI profile 1

[slot.esp.1]
device=/dev/sda2
bootname=system1
efi-loader=\\EFI\\BOOT\\BOOTX64.EFI
efi-unicode="@2" # UKI profile 2
```

**Describe alternatives you've considered**
The current workaround is to use `efibootmgr` from a post-install hook.
As probably many UEFI implementations are affected, this should be moved to common code in RAUC.

loblik, 2025-04-17:
Seems UEFI (EDK2 with NVIDIA patches) I currently work with, is not doing this. 

But it could also simplify initial installation. Default profile cmdline could contain "rauc.external" and then each entry would be created automatically when slot is flashed for the first time.

loblik, 2025-04-17:
> * But how would we even know which of the slots contain the ESP if none have a `esp-*` entry?

Not sure I got this. Entry would be created for the disk specified in `device` parameter I guess? That is passing it to `efibootmgr --device`?

Also what is the difference between scenario 1 and 3?

hnez, 2025-04-17:
> > * But how would we even know which of the slots contain the ESP if none have a `esp-*` entry?
> 
> Not sure I got this. Entry would be created for the disk specified in `device` parameter I guess? That is passing it to `efibootmgr --device`?

> Also what is the difference between scenario 1 and 3?

I've re-read what I wrote and I did indeed not get that point across very well. It comes down to the question: "what happens if no slot in the `system.conf` has any `efi-*` parameters specified?".
If `efi-cmdline` and `efi-loader` were both optional and had sensible defaults, a user may not want to specify any of them explicitly.

What would happen in the three scenarios if no `efi-*` parameter is specified for any slot:

1) A boot entry is created anyways using some defaults, because if it weren't the next step of the installation fails (setting the boot entry as BootNext).
2) No boot entry is created. If none was present already the installation will fail in the "Set BootNext" phase.
3) No boot entry is created. If none was present already the installation will fail in the "Set BootNext" phase.

---

I think we should just add a `efi-create-boot-entry=` parameter with options `always` and `if-missing` and make both `efi-loader=` and `efi-cmdline=` optional. We can then just no pass those parameters to `efibootmgr` if they are not specified and rely on it having sensible defaults.

loblik, 2025-04-17:
We care only about slots which contains `bootname` parameter right? So if we are flashing a slot which has `bootname` parametr, we check whether there is a UEFI bootentry with label matching `bootname`. If not we create it (with efibootmgr default values or with slot defined `efi-*` properties).

So the only question I guess is, whether we should do some migration, and recreate bootentries, which has `bootname` as label, but does not match what is specified with current `efi-*` properties. Or I'm missing something?

jluebbe, 2025-04-17:
I'd say this can be done in separate steps: 

1. create if entries are found missing while we write bootnext/bootorder
2. define mechanism if/when to update existing entries (probably if they have different values for efi-loader/efi-cmdline)

loblik, 2025-04-17:
> * Only if the boot entry does not exist? This would reduce the number of EFI variable writes. I think I remember hearing some concerns over the number of write cycles on the EFI variable backing store.

I also wanted to second this. On security concerned devices, variable storage can be protected by monotonic HW counter to prevent roll-back attacks. Frequent writes could make the counter run out.