rauc/rauc #1713: Change license for de.pengutronix.rauc.Installer.xml

baxeno, 2025-05-19:
**Is your feature request related to a problem? Please describe.**

`de.pengutronix.rauc.Installer.xml` has no license as it claims to be data.
This is incompatible with Fedora Packaging Guidelines.

`License note: This file is considered data and thus no license is needed for any usage.`

https://github.com/rauc/rauc/blob/master/src/de.pengutronix.rauc.Installer.xml#L2

**Describe the solution you'd like**

Would it be possible to change license to something like `CC0-1.0` or similar?

**Describe alternatives you've considered**

Tried to package as is.

**Additional context**

N/A

jluebbe, 2025-05-20:
Could you point to the part of the Fedora Packaging Guidelines which requires a license for pure data?

baxeno, 2025-05-20:
I am not familiar with the term pure data maybe it's a Debian category.

The XML file will fall into the **content** category and must have a license on the **allowed-content** list.

> Fedora applies different license approval criteria for different categories of material: **code**, **documentation**, **content**, **fonts**, and **firmware**. The corresponding status names for approved licenses are: **allowed**, **allowed-documentation**, **allowed-content**, **allowed-fonts**, and **allowed-firmware**.

Quote source: https://docs.fedoraproject.org/en-US/legal/license-approval/

List of allowed-content licenses: https://docs.fedoraproject.org/en-US/legal/allowed-licenses/#_allowed_content_licenses

In my understanding stating that the XML file has no license result in users do not have permission to use the content as they have not been granted any rights to use, modify, or distribute the work.

PR #1720 (2025-06-03): de.pengutronix.rauc.Installer.xml: add and use CC0-1.0 license
While we initially intended this file to be freely reusable, as we
considered it not to be copyrightable, it seems that opinions differ and
not having an explicit licenses causes friction.

To clarify and simplify the situation, we use the CC0-1.0 to declare
that it can be used without restrictions.

Fixes: #1713
Reported-by: Bruno Thomsen <bruno.thomsen@gmail.com>