rauc/rauc #1714: Extended attributes are ignored when installing a bundle from a tar archive

hacel, 2025-05-20:
**Describe the bug**
When installing a bundle containing a tar archive, extended file attributes like `security.capability` are not preserved.

**Background information**
I'm using RAUC version v1.13 and the bundle is a simple tar gzipped archive of the filesystem installed on a ext4 slot on Debian 12.

**To Reproduce**
Steps to reproduce the behavior:
0. Assume `mountpoint` is where the desired filesystem for the bundle is mounted at.
2. Choose a file that has extended attributes and ensure that they are there with the command `getcap mountpoint/bin/ping`. On Debian 12, you should see `cap_net_raw=ep` in the output of the previous command.
3. Create a tar archive of the desired filesystem using the command `tar --xattrs -cf image.tar.gz -C mountpoint .`
4. Create a bundle from the tar archive
5. Install the bundle on a machine
6. Boot into or simply just mount the freshly installed bundle
7. Check if extended file attributes were preserved by running for example `getcap /bin/ping` inside the newly installed bundle.

**Expected behavior**
I expect the extended attributes to be preserved on installation of the bundle.

**Logs**
N/A

**Additional context**
N/A

hacel, 2025-05-20:
Just adding this line made it work as expected
```
diff --git a/src/update_handler.c b/src/update_handler.c
index a90e4095..f1d6f879 100644
--- a/src/update_handler.c
+++ b/src/update_handler.c
@@ -1250,4 +1250,5 @@ static gboolean untar_image(WfosImage *image, gchar *dest, GError **error)
 	g_ptr_array_add(args, g_strdup(dest));
 	g_ptr_array_add(args, g_strdup("--numeric-owner"));
+	g_ptr_array_add(args, g_strdup("--xattrs-include=*"));
 	g_ptr_array_add(args, g_strdup(suffix_to_tar_flag(image->filename)));
 	g_ptr_array_add(args, NULL);
```

jluebbe, 2025-05-20:
From the manpage it seems that `--xattrs` should be enough. Do you know it this is supported by busybox tar as well?

hacel, 2025-05-20:
> From the manpage it seems that `--xattrs` should be enough. Do you know if this is supported by busybox tar as well?

From my testing, while extracting, simply doing `--xattrs` was not enough I had to add `--xattrs-include=*` as well. Which led me to trying only `--xattrs-include=*` which worked fine without `--xattrs`. Not sure what's happening there but that's what works.

I have no idea if it works with busybox, sorry. I only use GNU tar.

jluebbe, 2025-05-20:
So far, RAUC didn't have support for extended attributes with tar images, so adding support for them would be a new features. Accordingly we need to understand the implications and the behavior of the different tar implementations enough to avoid backwards compatibility issues.

hacel, 2025-05-20:
I see, it's your call of course. I'm going to be using my patched version in the mean time and will report any problems.