rauc/rauc #1770: casync extraction fails during delta BSP install with RAUC, but works manually

st-navajith, 2025-08-13:
We are experiencing an issue where casync extraction fails when performing a delta BSP install using RAUC in our Yocto-based embedded project. The failure occurs during the bundle extraction stage when RAUC invokes casync. However, extracting the same bundle manually with casync extract works fine, which suggests the bundle itself is valid and the problem may be in the way RAUC calls casync or in its runtime environment.

**Failed command - rauc install my-update.raucb**

RAUC version: [1.8]

Yocto/Distro version: [3.1.14]

Kernel version: [5.10.83-cip1-yocto-standard]

Host/target architecture: [smarc-rzg2lc]

Logs
```
Aug 13 15:24:21 smarc-rzg2lc rauc[503]: input bundle: /home/root/my-update-casync.raucb
Aug 13 15:24:21 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39mActive slot bootname: /dev/mmcblk0p2[[0m
Aug 13 15:24:21 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39minstalling /home/root/my-update-casync.raucb: started[[0m
Aug 13 15:24:21 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39minstalling /home/root/my-update-casync.raucb: Checking and mounting bundle...[[0m
Aug 13 15:24:21 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39mReading bundle: /home/root/my-update-casync.raucb[[0m
Aug 13 15:24:21 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39mVerifying bundle signature... [[0m
Aug 13 15:24:21 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39mVerified detached signature by 'O = Test Org, CN = Test Org Development-1'[[0m
Aug 13 15:24:21 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39mMounting bundle '/home/root/my-update-casync.raucb' to '/run/rauc/bundle'[[0m
Aug 13 15:24:21 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39mConfigured loop device '/dev/loop0' for 417792 bytes[[0m
Aug 13 15:24:21 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39mMarking target slot rootfs.0 as non-bootable...[[0m
Aug 13 15:24:22 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39mChecking image type for slot type: ext4[[0m
Aug 13 15:24:22 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39minstalling /home/root/my-update-casync.raucb: Updating slots...[[0m
Aug 13 15:24:22 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39mImage detected as type: *.caidx[[0m
Aug 13 15:24:22 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39minstalling /home/root/my-update-casync.raucb: Checking slot rootfs.0[[0m
Aug 13 15:24:22 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39mUpdating /dev/mmcblk0p1 with /run/rauc/bundle/core-image-bsp-smarc-rzg2lc.tar.gz.caidx[[0m
Aug 13 15:24:22 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39mFormatting ext4 slot /dev/mmcblk0p1[[0m
Aug 13 15:24:22 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39minstalling /home/root/my-update-casync.raucb: Updating slot rootfs.0[[0m
Aug 13 15:24:22 smarc-rzg2lc rauc[1084]: mke2fs 1.45.7 (28-Jan-2021)
Aug 13 15:24:23 smarc-rzg2lc rauc[1084]: [120B blob data]
Aug 13 15:24:23 smarc-rzg2lc rauc[1084]: Creating filesystem with 3932160 4k blocks and 983040 inodes
Aug 13 15:24:23 smarc-rzg2lc rauc[1084]: Filesystem UUID: 73b55213-a287-43fb-b9d3-c361909973be
Aug 13 15:24:23 smarc-rzg2lc rauc[1084]: Superblock backups stored on blocks:
Aug 13 15:24:23 smarc-rzg2lc rauc[1084]:         32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208
Aug 13 15:24:23 smarc-rzg2lc rauc[1084]: [57B blob data]
Aug 13 15:24:23 smarc-rzg2lc rauc[1084]: [54B blob data]
Aug 13 15:24:25 smarc-rzg2lc rauc[1084]: Creating journal (16384 blocks): done
Aug 13 15:24:25 smarc-rzg2lc rauc[1084]: [91B blob data]
Aug 13 15:24:25 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39mMounting ext4 slot /dev/mmcblk0p1[[0m
Aug 13 15:24:25 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39mExtracting /run/rauc/bundle/core-image-bsp-smarc-rzg2lc.tar.gz.caidx to /run/rauc/rootfs.0[[0m
Aug 13 15:25:59 smarc-rzg2lc rauc[1092]: Failed to run synchronizer: Input/output error
Aug 13 15:26:00 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39mUnmounting seed slot /dev/mmcblk0p2[[0m
Aug 13 15:26:00 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39mUnmounting ext4 slot /dev/mmcblk0p1[[0m
Aug 13 15:26:00 smarc-rzg2lc rauc[503]: [[0;1;38;5;185m[[0;1;39m[[0;1;38;5;185mInstallation error: Failed updating slot rootfs.0: failed to run casync extract: Child process exited with code 1[[0m
Aug 13 15:26:00 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39minstalling /home/root/my-update-casync.raucb: Installation error: Failed updating slot rootfs.0: failed to run casync extract: Child process exited with code 1[[0m
Aug 13 15:26:00 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39minstalling /home/root/my-update-casync.raucb: finished[[0m
Aug 13 15:26:00 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39minstalling `/home/root/my-update-casync.raucb` failed: 1[[0m
Aug 13 15:28:24 smarc-rzg2lc rauc[503]: [[0;1;39m[[0;1;31m[[0;1;39mstopping service[[0m
````

**casync extract --store=/path/to/castr /path/to/caidx /mnt/target**
This command succeeded.

Please advise on what differences in environment or invocation between RAUC and manual casync execution could cause extraction to fail only in RAUC's process. Are there known issues or additional options/configuration needed for RAUC to handle casync extraction reliably for delta updates?

Regards,
Navajith Jayakumar

ejoerns, 2025-08-25:
@st-navajith If manual extraction works, you could set

```
G_MESSAGES_DEBUG="rauc rauc-subprocess" rauc service
```

as described in https://rauc.readthedocs.io/en/latest/using.html#increasing-debug-verbosity, to see the actual `casync` command executed and test if running it manually still works.

ejoerns, 2025-08-26:
> [@ejoerns](https://github.com/ejoerns), this was resolved after updating the RAUC version.

Okay 👍 . Thank you for the update!

> However, am I able to configure the reconnect timeout when fetching the bundle via https://server.example.com/deploy/bundle-YYYYMMDD.raucb? I want to control the retry behavior for fetching.

Since, if using casync (in contrast to RAUC's built-in streaming with adaptive updates), the chunk fetching is handled by `casync`, this is something to be asked or resolved for casync. Maybe around https://github.com/systemd/casync/blob/e6817a79d89b48e1c6083fb1868a28f1afb32505/src/casync-http.c#L37